adafruit/adafruit-beaglebone-io-python #274: Threading motors on beaglebone

gahigg1, 2018-08-30:
Hi, My goal is to start four motors simultaneously at different speeds which are based on previous conditions. I am threading the motors in hope to do so. The motors do not start at the same time, but they do have different speeds and they stop at the same time. I need them to start at the same time, I copied the code below. Please take a look and help, much appreciated.
#!/usr/bin/python

import Adafruit_BBIO.GPIO as GPIO
import Adafruit_BBIO.PWM as PWM
import time
import threading
from threading import Thread

GPIO.setup("P9_23",GPIO.OUT)
GPIO.setup("P9_25",GPIO.OUT)
GPIO.setup("P9_27",GPIO.OUT)
GPIO.setup("P9_15",GPIO.OUT)
GPIO.setup("P9_41",GPIO.OUT)
GPIO.setup("P8_15",GPIO.OUT)
GPIO.setup("P8_17",GPIO.OUT)
GPIO.setup("P8_26",GPIO.OUT)
GPIO.setup("P8_7",GPIO.OUT)
GPIO.setup("P8_9",GPIO.OUT)

GPIO.output("P9_27",GPIO.HIGH)
GPIO.output("P8_26",GPIO.HIGH)

def Motor1(duty,sec):
        GPIO.output("P9_27",GPIO.HIGH)
        GPIO.output("P9_15",GPIO.HIGH)
        GPIO.output("P9_41",GPIO.LOW)
        PWM.start("P9_22",duty,1000)

def Motor2(duty,sec):
        GPIO.output("P9_27",GPIO.HIGH)
        GPIO.output("P9_23",GPIO.HIGH)
        GPIO.output("P9_25",GPIO.LOW)
        PWM.start("P9_16",duty,1000)

def Motor3(duty,sec):
        GPIO.output("P8_26",GPIO.HIGH)
        GPIO.output("P8_9",GPIO.HIGH)
        GPIO.output("P8_7",GPIO.LOW)
        PWM.start("P8_13",duty,1000)

def Motor4(duty,sec):
        GPIO.output("P8_26",GPIO.HIGH)
        GPIO.output("P8_15",GPIO.HIGH)
        GPIO.output("P8_17",GPIO.LOW)
        PWM.start("P9_42",duty,1000)

def stop(pin,channel):
        PWM.stop(channel)
        GPIO.output(pin,GPIO.LOW)

m1=threading.Thread(target=Motor1(100,3))
m2=threading.Thread(target=Motor2(100,3))
m3=threading.Thread(target=Motor3(100,3))
m4=threading.Thread(target=Motor4(100,3))

time.sleep(2)
stop("P9_15","P9_22")
stop("P9_23","P9_16")
stop("P8_9","P8_13")
stop("P8_15","P9_42")

pdp7, 2018-08-31:
> Hi, My goal is to start four motors simultaneously at different speeds which are based on previous conditions. I am threading the motors in hope to do so. The motors do not start at the same time, [...]

@gahigg1  I am uncertain why you would need to use threading.

I would recommend trying this:
```
Motor1(100,3)
Motor2(100,3)
Motor3(100,3)
Motor4(100,3)
```

gahigg1, 2018-08-31:
The motors are still starting at slightly different times, is there a way to enable them all at the same exact time?

pdp7, 2018-09-01:
@gahigg1 You could try doing all the calls to `GPIO.setup()` and `GPIO.output` methods first, and then run `PWM.start()` for each motor.

pdp7, 2018-09-06:
please reopen if this is still an issue

closed 2018-09-06 by pdp7