adafruit/adafruit-beaglebone-io-python #275: PCA9685 8 Channel PWM driver with beaglebone?

gahigg1, 2018-09-04:
I'm trying to use the Featherwing PCA9685 8-Channel PWM driver with a beaglebone black. Using 'i2cdetect -r 2', I can see which address the driver is on. I tried to install libraries for the Adafruit-pca9685, but the packages were not located? Can anyone help me with trying to control this driver/ use the libraries?

pdp7, 2018-09-06:
please run:
```sudo /opt/scripts/tools/version.sh```
and copy the output to this issue.

this will help me know if what versions of system software you have installed.

pdp7, 2018-09-06:
Please paste the Python code that you are trying to run.

Please copy and paste exactly any error messages.

Also, it would be helpful if you could run strace so that I can better understand what is happening.

Install strace (if it doesn't already exist):
```sudo apt-get install strace```

Then run strace with i2cdetect:
```strace -f -o strace.log python i2cdetect -r 2```
Paste the contents of strace.log into a GitHub Gist and post the link in a reply here.

Then run strace with python, replace test.py with the name of your test program:
```strace -f -o strace.log python test.py```

Paste the contents of strace.log into a seperate GitHub Gist and post the link in a reply here.

pdp7, 2018-09-11:
@gahigg1 have you been able to run the above commands?

pdp7, 2018-09-13:
@gahigg1 please re-open if this is still an issue

closed 2018-09-13 by pdp7