adafruit/adafruit-beaglebone-io-python #276: Enabled CircuitPython integration

pdp7, 2018-09-13:
@ladyada has been discussing with me on Adafruit Discord how to add support for BeagleBone Black in CircuitPython.

FYI - this is initial support to enable a basic CircuitPython test on BeagleBone to blink LED connected to P9_12:
https://github.com/adafruit/Adafruit_Blinka/pull/36

I will update this issue with further developments.

pdp7, 2018-09-13:
For example, this is CircuitPython code that blinks an LED connected to P9_12:
```python
import time
import board
import digitalio

print("hello blinky!")

led = digitalio.DigitalInOut(board.P9_12)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

pdp7, 2019-02-13:
Support for BeagleBone Black has been added to Blinka

closed 2019-02-13 by pdp7

ladyada, 2019-02-13:
@pdp7 we've rearch'd/refactored blinka a bit in the last couple months, wanna test it with BBB again to make sure it still works? we've split pin definitions into `chip` and `board` since many boards have the same chip, but w/ different pin configs

pdp7, 2019-02-19:
@ladyada thanks for letting me know.  I'll test and open up issue on Blinka if I encounter any regressions.

hulkco, 2019-03-21:
Hi @pdp7  and @ladyada, I have a BBGW from Seeedstudio, dusting around here, and I think making it agreeable with blinka is a good project, so I'll be seeing the post for the BBB and make the modifications, Regards !!

ladyada, 2019-03-21:
@hulkco yes please try - the chip is at least somewhat supported, hopefully not too much work :)

hulkco, 2019-03-21:
Hello @ladyada & @pdp7 i made fresh install of Debian 9.5 2018-10-07 4GB SD LXQT (with graphical desktop) for BeagleBone in a BBGW, and then install the Adafruit Beaglebone I/O Python API, i made all test and run like a charm, I would like to share my experience, adding to the guide https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/gpio, as I did or create a new one for this version of Beaglebone , is this possible?

pdp7, 2019-03-22:
Hi @hulkco,

The BeagleBone Green is identical to the BeagleBone Black from the perspective of Adafruit_BBIO.  The Blinka support should work fine as well.

I think you'll need to email support@adafruit.com if you would like access to edit the learning system.

thanks,
drew

ladyada, 2019-03-22:
@pdp7 hiya please work with @hulkco since you have learn system editing privileges, that would be the best way to update a single page

pdp7, 2019-03-23:
@hulkco please email me drew@pdp7.com and we can figure out the edits to make to the learning system tutorial.

in addition, I would appreciate any pull requests to update the examples in this repo's README:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/README.md