adafruit/adafruit-beaglebone-io-python #277: GPIO.IN setup fails in Debian Stretch Error: Set gpio mode failed, missing file or invalid permissions.

origtech, 2018-09-20:
Hi I'm having the same problem with Adafruit_BBIO.GPIO setup when configuring a port as an input pin. This same code runs OK using Debian Jessie but fails on Stretch. I am currently using the new uboot to for device table overlays and have tried both mapping the output pin in a device table and not mapping a the output pin in a device table. The setup for the IN mode fails in both cases. The setup for the OUT mode works.

Here is a sample of my test program:
#!/user/bin/python
import Adafruit_BBIO.GPIO as GPIO
GPIO.setup('P8_11', GPIO.IN)
if(GPIO.input('P8_11')):
    print 'high'
else:
    print 'low'

When I run the program I get this error message:
File "ioTest.py", line 4, in 
GPIO.setup('P8_11', GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.

sudo /opt/scripts/tools/version.sh produces the following:

git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT000C1826BBBG0624]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.9.82-ti-r102]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0stretch+20180517]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-4D7R-01, coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[ 1.344481] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[ 1.345071] gpio-of-helper ocp:gpio_helper: ready
[ 1.345687] gpio-of-helper ocp:cape-universal: ready
END

cat /boot/uEnv.txt produces the following:
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.9.82-ti-r102
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1

###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/.dtbo
#uboot_overlay_addr1=/lib/firmware/.dtbo
#uboot_overlay_addr2=/lib/firmware/.dtbo
#uboot_overlay_addr3=/lib/firmware/.dtbo

###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/.dtbo
#uboot_overlay_addr5=/lib/firmware/.dtbo
#uboot_overlay_addr6=/lib/firmware/.dtbo
#uboot_overlay_addr7=/lib/firmware/.dtbo

###Custom Cape
dtb_overlay=/lib/firmware/BBB-UMC-TESTER-00A0.dtbo

###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1

###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo

###Cape Universal Enable
enable_uboot_cape_universal=1

###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1

###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

To try and correct the problem I have run:
sudo apt-get update
sudo apt-get upgrade

But the problem remains. I am running a fresh install of Debian Version 9.3 IoT Stretch.
Any suggestions on what to try?

RobertCNelson, 2018-09-20:
So setting this:
```bash
###Custom Cape
dtb_overlay=/lib/firmware/BBB-UMC-TESTER-00A0.dtbo
```

with:
```bash
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
``` 
will disable this:

```bash
###Cape Universal Enable
enable_uboot_cape_universal=1
```

So what's in BBB-UMC-TESTER-00A0.dtbo?

Regards,

origtech, 2018-09-20:
BBB-UMC-TESTER-00A0.dtbo is a custom dtbo file that maps the GPIO that I am using.

I've modified /boot/uEnv.txt file to eliminate the custom cape and I still get the error. I have also verified that the error also appears when I do a fresh install using the Debian 9.3 LXQT image 

With custom cape dtbo file removed.
root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[e307a944e0be0610ff5296e0abe4ad31a6e70daa]
eeprom:[A335BNLT000C1826BBBG0624]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-03-05]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-ge9ff418fb8]:[location: dd MBR]
kernel:[4.9.82-ti-r102]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0~stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-4D7R-01, coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.340551] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.341574] gpio-of-helper ocp:cape-universal: ready
END

RobertCNelson, 2018-09-20:
```bash
debian@beaglebone:~$ cat test.py 
#!/usr/bin/python

import Adafruit_BBIO.GPIO as GPIO

GPIO.setup('P8_11', GPIO.IN)
if(GPIO.input('P8_11')):
	print 'high'
else:
	print 'low'
```
```bash
debian@beaglebone:~$ ./test.py 
low
```
```bash
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[73593ebe3b7d3cc381eeb502d45ccb33a6ec5e78]
eeprom:[A335BNLT000C1823BBBG1273]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.67-ti-r73]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.2-git20180829.0-0rcnee0~stretch+20180830]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.099542] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.111609] gpio-of-helper ocp:cape-universal: ready
END
```

origtech, 2018-09-21:
Downloaded latest version of BBB image (Debian-9.5 2018-09-16) and re-flashed BB.  Issue resolved.

closed 2018-09-21 by origtech

RobertCNelson, 2018-09-21:
and i completely missed this note from version.sh too...

```bash
uboot_detected_capes=BB-BONE-4D7R-01, 
```
So you have a an lcd, yeah, that would also require v4.14.x, to properly allow access to the unused pins... Doh..

Regards,