adafruit/adafruit-beaglebone-io-python #279: PWM.start generates runtime error: RuntimeError: Problem with the cape manager

suraj051, 2018-09-28:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  BeagleBone black, 
Linux beaglebone 4.14.67-ti-r73 #1 SMP PREEMPT Thu Aug 30 00:08:52 UTC 2018 armv7l GNU/Linux
Image used:bone-debian-9.5-iot-armhf-2018-08-30-4gb.img
- Python version (run `python -version` or `python3 -version`):  Python 2.7.13 

- Error message you are receiving, including any Python exception traces:  
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Problem with the cape manager
>>> RuntimeError: Problem with the cape manager
  File "<stdin>", line 1
    RuntimeError: Problem with the cape manager

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 
>>> import Adafruit_BBIO.PWM as PWM
>>> PWM.start("P8_13", 25, 1000)

pdp7, 2018-10-02:
Please run:
```
sudo /opt/scripts/tools/version.sh
```
and copy the output to this issue.

pdp7, 2018-10-07:
@suraj051 
Please run:
`sudo /opt/scripts/tools/version.sh`
and copy the output to this issue.

pdp7, 2018-10-09:
please re-open if still an issue.

closed 2018-10-09 by pdp7

suraj051, 2018-10-16:
Sorry for the delay in response, below is the outcome of the "sudo /opt/scripts/tools/version.sh"
/*******************************************************************************************************/

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[73593ebe3b7d3cc381eeb502d45ccb33a6ec5e78]
eeprom:[A335BNLT00A52414BBBK4307]
model:[SanCloud_BeagleBone_Enhanced]:WiFi AP Broken on Mainline
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
kernel:[4.14.67-ti-r73]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0~stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.2-git20180926.0-0rcnee0~stretch+20180927]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.057871] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.059112] gpio-of-helper ocp:cape-universal: ready
END

pdp7, 2018-10-16:
There is an older version of u-boot bootloader on the emmc.  This will prevent device tree overlays from being handled correctly by the bootloader.

Delete the old version of the u-boot bootloader on the eMMC with this command:
`sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k`

The BeagleBone will then boot from the newer version of u-boot bootloader on your SD card.