adafruit/adafruit-beaglebone-io-python #281: GPIO library leaves LED is a bad state on latest BeagleBone image on PocketBeagle

erikwelsh, 2018-09-30:
When running the LED blink code for a USR led on the PocketBeagle with the latest image (bone-debian-9.5-iot-armhf-2018-08-30-4gb.img.xz), the LED is left in a bad state upon exit (i.e. LED looks partially lit).  At that point, you are unable to use that LED until you perform a hard reboot of the PocketBeagle.  Here is the information to recreate the issue on the PocketBeagle:

debian@beaglebone:/var/lib/cloud9/file_io$ uname -a
Linux beaglebone 4.14.67-ti-r73 #1 SMP PREEMPT Thu Aug 30 00:08:52 UTC 2018 armv7l GNU/Linux

debian@beaglebone:/var/lib/cloud9/file_io$ python
Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.__file__
'/usr/local/lib/python2.7/dist-packages/Adafruit_BBIO-1.0.10-py2.7-linux-armv7l.egg/Adafruit_BBIO/GPIO.so'
>>> exit()

debian@beaglebone:/var/lib/cloud9/file_io$ cat bbio_led.py
import Adafruit_BBIO.GPIO as GPIO
import time

LED = "USR3"

GPIO.setup(LED, GPIO.OUT)

while True:
    GPIO.output(LED, GPIO.HIGH)
    time.sleep(0.5)
    GPIO.output(LED, GPIO.LOW)
    time.sleep(0.5)

The issue is in the gpio_export() function of source/event_gpio.c.  If you check that the GPIO is an LED before executing the export commands, like what is done in gpio_set_direction, then this fixes the issue.

I will send a pull request with the fix.  This is a quick update that copies the LED detection code from gpio_set_direction.  Eventually, the code should be refactored to consolidate the LED detection into a separate function or macro.

Keep up the good work and thanks for all the support that this library provides.

pdp7, 2018-10-02:
thanks for creating the PR.  I will test this out and merge if no issues.

pdp7, 2018-10-07:
I have merged #282 
thanks @erikwelsh

closed 2018-10-07 by pdp7