adafruit/adafruit-beaglebone-io-python #286: PWM Fails on PocketBeagle - "RuntimeError: Problem with the cape manager"

ansarid, 2018-10-12:
I have been trying to get PWM working on pins **P2_01** and **P2_03** on the **PocketBeagle** but have been unsuccessful. I am using **Debian 9.5** and have installed the **Adafruit-BBIO** library. I do  have digital outputs working but no PWM. I wrote a short program to test PWM as seen here:

```
import Adafruit_BBIO.PWM as PWM

pin_l = "P2_1"		# GPIO050 (A PWM1) - Left Motors Pin
pin_r = "P2_3"		# GPIO023 (B PWM2) - Right Motors Pin

DC_l = 100		# Duty Cycle in % from (0-100) - Left Motors Pin
DC_r = 100		# Duty Cycle in % from (0-100) - Right Motors Pin

PWM.start(pin_l, 0, 1000)	# Start PWM on Left Motors Pin
PWM.start(pin_r, 0, 1000)	# Start PWM on Right Motors Pin

while 1:

    PWM.set_duty_cycle(pin_l, DC_l)	# Set PWM on Left Motors Pin
    PWM.set_duty_cycle(pin_r, DC_r)	# Set PWM on Right Motors Pin

PWM.stop(pin_l)
PWM.stop(pin_r)
PWM.cleanup()
```

When I run this program I get the below error:

```
Traceback (most recent call last):
  File "/var/lib/cloud9/pwm.py", line 9, in <module>
    PWM.start(pin_l, 0, 1000)   # Start PWM on Left Motors Pin
RuntimeError: Problem with the cape manager
```

As you can see, the program fails when starting PWM on the first pin and I have been unable to solve this issue. Also, this code does work perfectly fine on the BeagleBone Black after changing the code to use its PWM pins.

pdp7, 2018-10-12:
@ansarid Please run:
`sudo /opt/scripts/tools/version.sh`
and copy the output to this issue.

ansarid, 2018-10-12:
> @ansarid Please run:
> `sudo /opt/scripts/tools/version.sh`
> and copy the output to this issue.

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335PBGL00A21748EPB00801]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.099274] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.107530] gpio-of-helper ocp:cape-universal: ready
END
```

pdp7, 2018-10-12:
@ansarid It looks like you are a running an update to image.  The next step is for me to try to reproduce this issue.   I will attempt to reproduce and followup here.

ansarid, 2018-10-12:
> @ansarid It looks like you are a running an update to image. The next step is for me to try to reproduce this issue. I will attempt to reproduce and followup here.

Thank you for your help! As far as I remember all I have done is flashed [Debian 9.5](https://debian.beagleboard.org/images/bone-debian-9.5-iot-armhf-2018-10-07-4gb.img.xz) to a microSD card and followed [Adafruit's Adafruit_BBIO install guide for Debian](https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu).

pdp7, 2018-10-16:
Ok thanks.  I just tried running this on a BeagleBone Black and got no exception.  I should have access to a PocketBeagle later today to try running on that.

ansarid, 2018-10-16:
> Ok thanks. I just tried running this on a BeagleBone Black and got no exception. I should have access to a PocketBeagle later today to try running on that.

Yes, the code works fine on the BeagleBone Black, but does not work on the PocketBeagle.

pdp7, 2018-10-17:
Related code in `./source/py_pwm.c`:
```c
    err = pwm_start(key, duty_cycle, frequency, polarity);
    switch (err) {
        case BBIO_OK:
            break;

        case BBIO_ACCESS:
            PyErr_SetString(PyExc_IOError, "could not access a necessary file");
            return NULL;

        case BBIO_SYSFS:
            PyErr_SetString(PyExc_RuntimeError, "Problem with a sysfs file");
            return NULL;

        case BBIO_CAPE:
            PyErr_SetString(PyExc_RuntimeError, "Problem with the cape manager");
            return NULL;

```

pdp7, 2018-10-17:
This might do the trick:
```
diff --git a/source/c_pwm.c b/source/c_pwm.c
index f983592..ec4183a 100644
--- a/source/c_pwm.c
+++ b/source/c_pwm.c
@@ -339,6 +339,10 @@ BBIO_err pwm_setup(const char *key, __attribute__ ((unused)) float duty, __attri
     // Make sure that one of the universal capes is loaded
     if( !uboot_overlay_enabled() // only check kernel overlays if u-boot overlays are not being used
         &&
+        !beaglebone_blue() // beaglebone blue has complete dtb file and does not need overlays 
+        &&
+        !pocketbeagle() // pocketbeagle has complete dtb file and does not need overlays
+        &&
         !( device_tree_loaded("cape-univ-audio") // from cdsteinkuehler/beaglebone-universal-io
         || device_tree_loaded("cape-univ-emmc")  // ""
         || device_tree_loaded("cape-univ-hdmi")  // ""
```
Testing it out

pdp7, 2018-10-17:
looks that did resolve the issue:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo strace -f -o /tmp/a python3 ~/pwm-issue286.py 
Adafruit_BBIO: pwm_start: P2_1, 0.000000, 1000.000000, 0
Adafruit_BBIO: pwm_start: P2_3, 0.000000, 1000.000000, 0
```

I'll push a commit.

pdp7, 2018-10-17:
@ansarid please try master branch which includes this fix https://github.com/adafruit/adafruit-beaglebone-io-python/commit/c09ab8ee43fe211a38523a96e7039aee0df33084

ansarid, 2018-10-18:
> @ansarid please try master branch which includes this fix [c09ab8e](https://github.com/adafruit/adafruit-beaglebone-io-python/commit/c09ab8ee43fe211a38523a96e7039aee0df33084)

It works now! Thank you!

pdp7, 2018-10-18:
I've created a new release:
[Release 1.1.0](https://github.com/adafruit/adafruit-beaglebone-io-python/releases/tag/1.1.0)
I published it to PyPI and is available to install via pip.

ansarid, 2018-10-18:
I'll test that out today to make sure the pip package works now.

Thank you for your help!

lifton, 2018-10-23:
I just ran into the same problem and verified that @pdp7's fix works. Thanks!

closed 2018-10-23 by pdp7