adafruit/adafruit-beaglebone-io-python #287: ValueError: Set gpio mode failed, missing file or invalid permissions.

mmk7070, 2018-10-12:
hi,
I am new the beaglebone black. I have written a code. when i run it, i am getting this error that is below . ı looked for this issue and, try the other titles but i didn't get it . Could you help this issue? 

Best regards

File "main2.py", line 8, in <module>
    GPIO.setup("P9_25", GPIO.IN)
ValueError: Set gpio mode failed, missing file or invalid permissions.
 

sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[73593ebe3b7d3cc381eeb502d45ccb33a6ec5e78]
eeprom:[A335BNLT000C1828BBBG0971]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7                                                                                                             c6a]:[location: dd MBR]
kernel:[4.18.12-bone13]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                                                                                             00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.2-git20180829.0-0rcnee0~stretch+20180830]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                                                                                             ers systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk                                                                                                              weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/m                                                                                                             mcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.913768] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.923616] gpio-of-helper ocp:cape-universal: ready
END

mmk7070, 2018-10-12:
Additional information

debian@beaglebone:~/3MES$ cat /etc/dogtag
BeagleBoard.org Debian Image 2018-08-30
debian@beaglebone:~/3MES$ dmesg | grep bone
[    0.000000] Linux version 4.18.12-bone13 (root@a3-imx6q-wandboard-2gb) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)) #1 PREEMPT Tue Oct 9 11:35:35 UTC 2018
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    1.179258] usb usb1: Manufacturer: Linux 4.18.12-bone13 musb-hcd
[   22.079009] systemd[1]: Set hostname to <beaglebone>.

mmk7070, 2018-10-12:
hi @pdp7 
our code is :+1: 

import web
import Adafruit_BBIO.GPIO as GPIO

GPIO.setup("P9_27", GPIO.OUT)
GPIO.setup("P9_30", GPIO.OUT)
GPIO.setup("P9_21", GPIO.IN)
GPIO.setup("P9_23", GPIO.IN)
GPIO.setup("P9_25", GPIO.IN) 

urls = (
	'/output1', 'output1', 
	'/output2', 'output2',
	'/input1', 'input1',
	'/input2', 'input2',
	'/input3', 'input3',
	'/control', 'control'
)

class output1:
    def GET(self):
        i = web.input(status=None)
	if (i.status == None):
		if GPIO.input("P9_27"):
			GPIO.output("P9_27", GPIO.LOW)
		else:
			GPIO.output("P9_27", GPIO.HIGH)
	else:
		GPIO.output("P9_27", int(i.status))

class output2:
    def GET(self):
        i = web.input(status=None)
	if (i.status == None):
		if GPIO.input("P9_30"):
			GPIO.output("P9_30", GPIO.LOW)
		else:
			GPIO.output("P9_30", GPIO.HIGH)
	else:
		GPIO.output("P9_30", int(i.status))

class input1:
    def GET(self):
	return web.websafe(GPIO.input("P9_21"))
	
class input2:
    def GET(self):
	return web.websafe(GPIO.input("P9_23"))

class input3:
    def GET(self):
	return web.websafe(GPIO.input("P9_25"))	
	
class control:
    def GET(self):
	web = """<!DOCTYPE html>
<html>
<head>
    <meta charset="utf-8" />
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <title>3MES Controller</title>
    <meta name="viewport" content="width=device-width, initial-scale=1">
    <script type="text/javascript" src = "static/jquery-3.3.1.min.js"></script>
    <script type="text/javascript">
        var url = "http://192.168.7.2:1234/"
        $(document).ready(function(){
            $("#output1").click(function(){
                $.get(url + "output1")
            });

            $("#output2").click(function(){
                $.get(url + "output2")
            });
        });   
    
    </script>
</head>
<body>
    <h2>3MES Controller</h2>
    <table>
        <tr>
          <td>Input1</td>
          <td>"""
	web += str(GPIO.input("P9_21"))
	web += """</td>
        </tr>
        <tr>
          <td>Input2</td>
          <td>"""
	web += str(GPIO.input("P9_23"))
	web += """</td>
        </tr>
        <tr>
          <td>Input3</td>
          <td>"""
	web += str(GPIO.input("P9_25"))
	web += """</td>
        </tr>
        <tr>
          <td>Output1</td>
          <td><button type="button" id="output1">Switch</button></td>
        </tr>
        <tr>
          <td>Output2</td>
          <td><button type="button" id="output2">Switch</button></td>
        </tr>
      </table>
</body>
</html>"""
	return web		

if __name__ == "__main__":
    app = web.application(urls, globals())
    app.run()

mmk7070, 2018-10-12:
Also i wrote your suggestion. I got this

debian@beaglebone:~$ sudo apt-get install strace
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  strace
0 upgraded, 1 newly installed, 0 to remove and 45 not upgraded.
Need to get 479 kB of archives.
After this operation, 793 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian stretch/main armhf strace armhf 4.15-2 [479 kB]
Fetched 479 kB in 3s (155 kB/s)
Selecting previously unselected package strace.
(Reading database ... 41727 files and directories currently installed.)
Preparing to unpack .../strace_4.15-2_armhf.deb ...
Unpacking strace (4.15-2) ...
Setting up strace (4.15-2) ...
debian@beaglebone:~$ strace -f -o strace.log python test.py
python: can't open file 'test.py': [Errno 2] No such file or directory

mmk7070, 2018-10-12:
debian@beaglebone:~$ sudo apt-get install strace
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  strace
0 upgraded, 1 newly installed, 0 to remove and 45 not upgraded.
Need to get 479 kB of archives.
After this operation, 793 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian stretch/main armhf strace armhf 4.15-2 [479 k                                                                                                             B]
Fetched 479 kB in 3s (155 kB/s)
Selecting previously unselected package strace.
(Reading database ... 41727 files and directories currently installed.)
Preparing to unpack .../strace_4.15-2_armhf.deb ...
Unpacking strace (4.15-2) ...
Setting up strace (4.15-2) ...
debian@beaglebone:~$ strace -f -o strace.log python test.py
python: can't open file 'test.py': [Errno 2] No such file or directory
d
debian@beaglebone:~$ debian@beaglebone:~$  strace -f -o strace.log python main2.                                                                                                           py                   python: can't open file 'main2.py': [Errno 2] No such file                                                                                                              or directory

pdp7, 2018-10-16:
@mmk7070 what is the name of your program?
```
debian@beaglebone:$ strace -f -o strace.log python test.py
python: can't open file 'test.py': [Errno 2] No such file or directory
d
debian@beaglebone:$ debian@beaglebone:$ strace -f -o strace.log python main2. py python: can't open file 'main2.py': [Errno 2] No such file or directory
```
It the last argument should be the path to your python file.

mmk7070, 2018-10-17:
it is main2.py

mmk7070, 2018-10-17:
it is main2.py. I sent it that was ;
debian@beaglebone:$ strace -f -o strace.log python main2. py
python: can't open file 'main2.py': [Errno 2] No such file or directory

mmk7070, 2018-10-17:
dear @pdp7 i have changed gpio pin and problem was solved. i changed from "P9_25" pin to "P9_12". thank you for help. However, i don't understand why it does not run .

pdp7, 2018-10-17:
@mmk7070 It could be due to the pin mode (e.g. pinmux) that pin is configured for.

Please run this and paste the output:
```
#print the current pin mode
config-pin -q p9.25 

#list the possible pin modes
config-pin -l p9.25  

#print the current pin mode
config-pin -q p9.12

#list the possible pin modes
config-pin -l p9.12
```

pdp7, 2018-10-24:
@mmk7070 are you still having this issue?

mmk7070, 2018-10-24:
No. Thanks for help. I have changed another I/O pin and i solved it.

closed 2018-10-24 by pdp7