adafruit/adafruit-beaglebone-io-python #288: PWM.start generates runtime error: RuntimeError: Problem with the cape manager

suraj051, 2018-10-16:
Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): BeagleBone black,
Linux beaglebone 4.14.67-ti-r73 #1 SMP PREEMPT Thu Aug 30 00:08:52 UTC 2018 armv7l GNU/Linux
Image used:bone-debian-9.5-iot-armhf-2018-08-30-4gb.img

Python version (run python -version or python3 -version): Python 2.7.13

Error message you are receiving, including any Python exception traces:
Traceback (most recent call last):
File "", line 1, in 
RuntimeError: Problem with the cape manager

RuntimeError: Problem with the cape manager
File "", line 1
RuntimeError: Problem with the cape manager

List the steps to reproduce the problem below (if possible attach code or commands
to run):
import Adafruit_BBIO.PWM as PWM
PWM.start("P8_13", 25, 1000)

/********************************************************************************************/
Sorry for the delay in response, below is the outcome of the "sudo /opt/scripts/tools/version.sh"
/*******************************************************************************************************/

debian@beaglebone:$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[73593ebe3b7d3cc381eeb502d45ccb33a6ec5e78]
eeprom:[A335BNLT00A52414BBBK4307]
model:[SanCloud_BeagleBone_Enhanced]:WiFi AP Broken on Mainline
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
kernel:[4.14.67-ti-r73]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade ]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0stretch+20180517]
pkg:[kmod]:[23-2rcnee1stretch+20171005]
pkg:[librobotcontrol]:[1.0.2-git20180926.0-0rcnee0stretch+20180927]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[ 1.057871] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[ 1.059112] gpio-of-helper ocp:cape-universal: ready
END

pdp7, 2018-10-16:
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
```

There is an older version of u-boot bootloader on the emmc.  This will prevent device tree overlays from being handled correctly by the bootloader.

Delete the old version of the u-boot bootloader on the eMMC with this command:
`sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k`

The BeagleBone will then boot from the newer version of u-boot bootloader on your SD card.

suraj051, 2018-10-16:
Thanks for the quick reply but i have below error again and board never boots again from sdcard:
-1:
/*****************************************************************/
debian@beaglebone:~$ sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k                                   [sudo] password for debian:
1+0 records in
1+0 records out
131072 bytes (131 kB, 128 KiB) copied, 0.0753454 s, 1.7 MB/s
/****************************************************************/

2.
Tried to use PWM after that then below error:
debian@beaglebone:~$ python pwm.py
Traceback (most recent call last):
  File "pwm.py", line 2, in <module>
    PWM.start("P8_13",25,1000)
RuntimeError: Unknown error

3.After i restart it never boots again...!

suraj051, 2018-10-16:
please suggest me a flashable image version which you are uisng with this lib and have no issues...

thanks,
Suraj

pdp7, 2018-10-16:
@robertCnelson any suggestions?

pdp7, 2018-10-16:
The current image for SD card by @robertCnelson is here http://beagleboard.org/latest-images

suraj051, 2018-10-16:
I am looking only for the adafruit python lib to wrok with my BBB as my project involves UART and AD.

I hope the link which you have shared satisfies my request.

thanks :)

suraj051, 2018-10-16:
Hello Pdp7,

i tried to boot the image "bone-debian-9.5-lxqt-armhf-2018-10-07-4gb"this is with GUI(http://debian.beagleboard.org/images/bone-debian-9.5-lxqt-armhf-2018-10-07-4gb.img.xz) from SDCARD.

1.when i hold the boot button and power the BBB it dosent boot(only power LED glows).

2. Then with sdcard instered without holding the boot button power is applied along with power LED user LED's 0,1,2 glows contineiously and 3rd LED is off(i think it did not boot).

Please let me know what went wrong??

thanks,

pdp7, 2018-10-18:
@RobertCNelson any suggestions for @suraj051 to try?

pdp7, 2018-10-24:
@suraj051 are you still having this issue?

suraj051, 2018-10-24:
Yes I still face this issue. If I am also using the image which u suggested which ideally should not have any issues I think this is a major issue.

Which prevents even basic usage. Please provide a working solution asap.

Thanks.

pdp7, 2018-10-24:
@suraj051 
> 1.when i hold the boot button and power the BBB it dosent boot(only power LED glows).

Is the SD card inserted when you apply power?

The procedure should be:
1) insert SD card
2) press and hold the boot button
3) turn on power to the BeagleBone
4) release boot button

@jadon @RobertCNelson any other suggestions?

suraj051, 2018-10-25:
That's exactly what I am dng I think the command suggested by you screwed the boot loader and thus never booting again.
I reflaged the imgage u suggested no need to waste time on booting now.please concentrate on the library issue.
Why do we have soo many issues when just trying to use the library?? If it is due to capes why can't they release a seperate image without this Cape support as I think 70% of the users sides bb without any cape.

Please try to reproduce issue and provide a working solution asap.

suraj051, 2018-10-28:
Hello Pdp7,

Could you please help me with this??

I am just asking to give inputs so that the library will work.

I am even ready to flash different image.

If your are responsible for this link you might be having a test setup why can't u provide the same image config to me please.

After having BBB if I cannot use basic peripherals like UART adc I am worried.

Looking here I don't think anyone is using this lib actively.

If you have any idea about some other library which works fine please point me in that direction.

Thanks

pdp7, 2018-10-29:
@suraj051 

The PWM example from the first post in this issue does run without error on my BeagleBone Black:
```
debian@beaglebone:~$ cat pwm-test.py
import Adafruit_BBIO.PWM as PWM
PWM.start("P8_13", 25, 1000)
debian@beaglebone:~$ python pwm-test.py
```

The term 'cape' is often used as a synonym for Device Tree Overlay.  Device Tree Overlays are now handled in the bootloader, instead of the cape manager driver in the Linux kernel.  This is why the version of bootloader on the eMMC is very important. 

Please run the version script again:
```
sudo /opt/scripts/tools/version.sh
```

I am in particularly interested in the lines that start with `bootloader`.

For reference, here is the output for the BeagleBone Black that I am testing on:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT000C3014BBBK1316]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-07-15]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-g254339602c]:[location: dd MBR]
kernel:[4.14.54-ti-r63]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180803.0-0rcnee0~stretch+20180804]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.0-git20180710.0-0rcnee0~stretch+20180712]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.040666] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.047813] gpio-of-helper ocp:cape-universal: ready
END
```

suraj051, 2018-10-29:
Thanks for the reply and suggesstions.

Let me run the test and steps u suggested I will get back to you once done.

As I am busy with some other activity it may take some time to reply. Please don't close this issue.
Thanks

suraj051, 2018-11-20:
Yes still I have did u check at your side ??

pdp7, 2018-11-20:
@suraj051 
> Let me run the test and steps u suggested I will get back to you once done.
Please run the version script again:
```
sudo /opt/scripts/tools/version.sh
```
I am in particularly interested in the lines that start with bootloader.

pdp7, 2019-02-13:
@suraj051 please reopen if still an issue

closed 2019-02-13 by pdp7