adafruit/adafruit-beaglebone-io-python #289: ADC.setup causes segmentation fault

suraj051, 2018-10-16:
Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.): BeagleBone black,
Linux beaglebone 4.14.67-ti-r73 #1 SMP PREEMPT Thu Aug 30 00:08:52 UTC 2018 armv7l GNU/Linux
Image used:bone-debian-9.5-iot-armhf-2018-08-30-4gb.img

Python version (run python -version or python3 -version): Python 2.7.13

Error message you are receiving, including any Python exception traces:

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3614.917896] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP ARM

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.037436] Process python (pid: 2270, stack limit = 0xdadde218)

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.043469] Stack: (0xdaddfce8 to 0xdade0000)

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.047850] fce0: c0b7f290 c0b798b0 df95a498 b674d2 76 dae55c40 dae5a5a0

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.056067] fd00: dae55c40 00000000 dae55a00 dae5a5a0 dae55a00 dae5a5 80 c11391e8 00000000

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.064285] fd20: daddfd54 daddfd30 c0386c3c c03864f8 df95dcf0 dae5a5 80 00000044 00000001

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.072503] fd40: dae5a580 c11391e8 daddfdec daddfd58 c0b85330 c0386b f4 00000000 014000c0

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.080719] fd60: 00000000 c0d9ea88 daddfd94 daddfd78 c016d8d8 dae5a5 84 c15d8eb0 dae5a5a0

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.088937] fd80: c1504dc8 df95dcf0 dae5a598 00000000 daddfdac 000000 00 c0dab8c8 db210000

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.097153] fda0: c11be3a8 daddfdb0 00000008 c0b8408c da97c810 daddfe 3c 000000ef b674d276

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.105371] fdc0: c1504dc8 da97c810 00000028 db565e18 00000000 c1504d c8 db565e18 c1166d08

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.113588] fde0: daddfdfc daddfdf0 c0b85598 c0b84a24 daddfe74 daddfe 00 c095e370 c0b85584

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.121806] fe00: 00000007 daddfe10 c0386824 c095f064 00000000 c1504d c8 db565e20 00000008

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.130023] fe20: dc23fa10 00000008 db565e10 db210000 da97ca5b dc23fa 10 dc23fa10 e06610e2

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.138240] fe40: c092cedc b674d276 daddfe70 dae55300 c1504dc8 db565e 10 dae55000 dc23fa10

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.146457] fe60: 00000000 da97c810 daddfeb4 daddfe78 c095f4e4 c095de fc 00000000 daddff68

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.154675] fe80: da89ba80 b674d276 015507f0 c095f394 dae5a100 000000 40 00000000 daddff68

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.162891] fea0: dae55000 dae5a110 daddfecc daddfeb8 c092b330 c095f3 a0 c092b308 dae5a100

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.171109] fec0: daddfee4 daddfed0 c03854a0 c092b314 00000006 dae5a1 00 daddff1c daddfee8

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.179327] fee0: c0384aec c038545c 00000040 00000000 da803d88 c03849 f4 da89ba80 015507f0

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.187545] ff00: daddff68 00000000 015507f0 00000006 daddff34 daddff 20 c02f90cc c0384a00

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.195761] ff20: 00000006 da89ba80 daddff64 daddff38 c02f92b0 c02f90 b0 00000000 c031ba54

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.203978] ff40: c1504dc8 da89ba80 00000040 00000000 da89ba80 015507 f0 daddffa4 daddff68

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.212195] ff60: c02f9518 c02f9208 00000040 00000000 5b885777 b674d2 76 00000000 00000006

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.220412] ff80: 015507f0 0153bd00 00000004 c01090e4 dadde000 000000 00 00000000 daddffa8

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.228629] ffa0: c0108f00 c02f94c8 00000006 015507f0 00000004 015507 f0 00000006 00000000

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.236846] ffc0: 00000006 015507f0 0153bd00 00000004 015507f0 000000 06 b6a9c728 b6d5bab0

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.245063] ffe0: 00000000 be84c18c b6e1c1bb b6e58af6 00030030 000000 04 00000000 00000000

Message from syslogd@beaglebone at Sep 28 14:25:13 ...
kernel:[ 3615.350576] Code: eaffffd5 e5930018 e3500000 1affff56 (e7f001f2)
Segmentation fault

List the steps to reproduce the problem below (if possible attach code or commands
to run):
import Adafruit_BBIO.ADC as ADC
ADC.setup()

/*******************sudo /opt/scripts/tools/version.sh*****************************/

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[73593ebe3b7d3cc381eeb502d45ccb33a6ec5e78]
eeprom:[A335BNLT00A52414BBBK4307]
model:[SanCloud_BeagleBone_Enhanced]:WiFi AP Broken on Mainline
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
kernel:[4.14.67-ti-r73]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180914.0-0rcnee0~stretch+20180914]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.2-git20180926.0-0rcnee0~stretch+20180927]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.057871] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.059112] gpio-of-helper ocp:cape-universal: ready
END

pdp7, 2018-10-16:
```
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
```

There is an older version of u-boot bootloader on the emmc.  This will prevent device tree overlays from being handled correctly by the bootloader.

Delete the old version of the u-boot bootloader on the eMMC with this command:
`sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k`

The BeagleBone will then boot from the newer version of u-boot bootloader on your SD card.

pdp7, 2018-10-29:
Please refer to #288 with regards to getting your BeagleBone Black up to date.

pdp7, 2018-11-20:
@suraj051 are you still having this issue?

greg112358, 2019-02-11:
@pdp7 
I had this same issue, running:
`sudo dd if=/dev/zero of=/dev/mmcblk1 count=1 seek=1 bs=128k`
fixed it for me.

pdp7, 2019-02-13:
@greg112358 thanks for the feedback

@suraj051 please reopen if still an issue

closed 2019-02-13 by pdp7