adafruit/adafruit-beaglebone-io-python #291: cd: /sys/devices/bone_capemgr.*: No such file or directory

sebastianimsys, 2018-10-26:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

sebastianimsys, 2018-10-26:
Where is the "/slots" or how enable the UART ports in the BBB.
 The output of command  sudo /opt/scripts/tools/version.sh is:

`git:/opt/scripts/:[e9bcff232834702c1c810710706ee815d77b080b]
eeprom:[A335BNLTEIA04816BBBK28A7]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gd5b4c4b656]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.185005] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.197159] gpio-of-helper ocp:cape-universal: ready
END`

sebastianimsys, 2018-10-26:
Hi...Where is the "/slots" or how enable the UART ports in the BBB.
The output of command sudo /opt/scripts/tools/version.sh is:

`root@beaglebone:~# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e9bcff232834702c1c810710706ee815d77b080b]
eeprom:[A335BNLTEIA04816BBBK28A7]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-gd5b4c4b656]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.185005] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.197159] gpio-of-helper ocp:cape-universal: ready
END
`

pdp7, 2018-10-29:
@sebastianimsys 
*Please visit the UART section of the README for this library*:
https://github.com/adafruit/adafruit-beaglebone-io-python/#uart

There is a no longer the cape manager driver in the Linux kernel on current BeagleBoard.org Debian images by @RobertCNelson.  Therefore, the slots file no longer exists in /sys.

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Where_did_the_slots_file_go.3F

> Kernel Overlays are going bye-bye, too many bugs, too many race conditions, no kernel maintainers interested. We've said our farewells and U-Boot Overlays is the way forward: https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays With multiple parties working on the U-Boot infrastructure.

sebastianimsys, 2018-10-29:
Now i have a new problem, i want to use a GPRS module Sara-u201 whit yhis tutorial: https://learn.adafruit.com/fona-tethering-to-raspberry-pi-or-beaglebone-black/usage

When i used this command "cat /var/log/syslog | grep pppd" my output is:
`debian@beaglebone:~$ cat /var/log/syslog | grep pppd
`
`Binary file (standard input) matches`

RobertCNelson, 2018-10-29:
@sebastianimsys if you need a uart at bootup for this module, use one of the uart u-boot overlays that matches the pins you are using..

for example: P9.24 P9.26: https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART1-00A0.dts

But we don't know what pins you want to use..

Regards,

RobertCNelson, 2018-10-29:
@sebastianimsys this is ancient:

> When i used this command "cat /var/log/syslog | grep pppd" my output is:
> `debian@beaglebone:~$ cat /var/log/syslog | grep pppd `
> `Binary file (standard input) matches`

Use:

```bash
journalctl | grep pppd
```

RobertCNelson, 2018-10-30:
@sebastianimsys you need a uart at bootup for this module, use one of the uart u-boot overlays that matches the pins you are using..

for example: P9.24 P9.26: https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART1-00A0.dts

But we don't know what pins you want to use..

Regards,

sebastianimsys, 2018-10-30:
I'm using P9_24 and P9_26, sorry but i don't really understand...This link "https://github.com/beagleboard/bb.org-overlays/blob/master/src/arm/BB-UART1-00A0.dts" shows a scrip, but what i do with that scrip? i must copy it in some folder of the BBB?

RobertCNelson, 2018-10-30:
@sebastianimsys in /boot/uEnv.txt set:

```bash
uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
```
Now P9.24/P9.26 will be an active uart on bootup. Then run your "sudo pon fona" script, making sure it talks to the correct usart...

Regards,

sebastianimsys, 2018-10-30:
I copy  

`uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo`  

on the bottom of uEnv.txt and run "sudo pon fano" and no work,  so i use the commands to configure pin 

`config-pin P9.24 uart  # UART1_TXD`
`config-pin P9.26 uart  # UART1_RXD `

and the answer to those commands are:

`P9_24 pinmux file not found!`
`bash:/sys/devices/platform/ocp/ocp*P9_24_pinmux/state: No such file or directory`

RobertCNelson, 2018-10-30:
@sebastianimsys correct..

```bash
uboot_overlay_addr0=/lib/firmware/BB-UART1-00A0.dtbo
```
Hard "CODES"...
```bash
config-pin P9.24 uart # UART1_TXD
config-pin P9.26 uart # UART1_RXD
```
Something is wrong with your "fano" script, do you have the source?

Regards,

sebastianimsys, 2018-10-30:
Yeah, I'm using the code of this page : "https://learn.adafruit.com/fona-tethering-to-raspberry-pi-or-beaglebone-black/setup"

`connect "/usr/sbin/chat -v -f /etc/chatscripts/gprs -T internet.comcel.com.co"
`
`/dev/ttyO1
`
`115200
`
`noipdefault
`
`usepeerdns
`
`defaultroute
`
`persist
`
`noauth
`
`nocrtscts
`
`local
`

that the code without comments, also try with the UART4 but doesn't work too

pdp7, 2018-11-08:
@sebastianimsys are you still having an issue? thanks

closed 2018-11-20 by pdp7