adafruit/adafruit-beaglebone-io-python #293: Error on installing dafruit_BBIO with pip3

GregoireWulliamoz, 2018-11-02:
- Platform/operating system: 
BeaglBone Black armv7 32/debian 9.5 stretch

- Python version: 
Python 3.5.3 & pip3 9.0.1

-Error message you are receiving, including any Python exception traces:
`Collecting Adafruit_BBIO
  Using cached https://files.pythonhosted.org/packages/3f/a8/f3128e98b25a670956b465595687284c49ea1e4aced2f6b4c8ae4d9a2c14/Adafruit_BBIO-1.1.0.tar.gz
Building wheels for collected packages: Adafruit-BBIO
  Running setup.py bdist_wheel for Adafruit-BBIO ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-yuoc17jh/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpiszvr4zhpip-wheel- --python-tag cp35:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for Adafruit-BBIO
  Running setup.py clean for Adafruit-BBIO
Failed to build Adafruit-BBIO
Installing collected packages: Adafruit-BBIO
  Found existing installation: Adafruit-BBIO 1.1.0
    Uninstalling Adafruit-BBIO-1.1.0:
      Successfully uninstalled Adafruit-BBIO-1.1.0
  Running setup.py install for Adafruit-BBIO ... done
Successfully installed Adafruit-BBIO-1.1.0
`
-List the steps to reproduce the problem below (if possible attach code or commands
to run):

`- Get latest BeaglBone firmware: Debian 9.5 2018-10-07 4GB SD LXQT
- sudo apt-get update
- sudo apt-get install ntpdate 
- sudo ntpdate pool.ntp.org
- sudo apt-get install python3-pip build-essential libssl-dev libffi-dev python-dev
- sudo pip3 install Adafruit_BBIO`

No error on installing with 2.7.13

GregoireWulliamoz, 2018-11-02:
Upgraded to pip 18-1, same issue:
debian@beaglebone:~/environments$ sudo pip3 install --upgrade --force-reinstall Adafruit_BBIO
Collecting Adafruit_BBIO
  Using cached https://files.pythonhosted.org/packages/3f/a8/f3128e98b25a670956b465595687284c49ea1e4aced2f6b4c8ae4d9a2c14/Adafruit_BBIO-1.1.0.tar.gz
Building wheels for collected packages: Adafruit-BBIO
  Running setup.py bdist_wheel for Adafruit-BBIO ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-7_98yld_/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp0oxy7cc4pip-wheel- --python-tag cp35:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for Adafruit-BBIO
  Running setup.py clean for Adafruit-BBIO
Failed to build Adafruit-BBIO
Installing collected packages: Adafruit-BBIO
  Found existing installation: Adafruit-BBIO 1.1.0
    Uninstalling Adafruit-BBIO-1.1.0:
      Successfully uninstalled Adafruit-BBIO-1.1.0
  Running setup.py install for Adafruit-BBIO ... done
Successfully installed Adafruit-BBIO-1.1.0

pdp7, 2018-11-08:
Please run this command and paste the output so I can better understand the configuration of your BeagleBone:
```
sudo /opt/scripts/tools/version.sh 
```

GregoireWulliamoz, 2018-11-09:
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT000C1830BBBG0722]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.119604] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.131460] gpio-of-helper ocp:cape-universal: ready
END

pdp7, 2018-11-20:
I have packaged 1.1.1:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo python3 ./setup.py sdist
running sdist
running egg_info
writing dependency_links to Adafruit_BBIO.egg-info/dependency_links.txt
writing Adafruit_BBIO.egg-info/PKG-INFO
writing top-level names to Adafruit_BBIO.egg-info/top_level.txt
reading manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
warning: sdist: standard file not found: should have one of README, README.rst, README.txt

running check
creating Adafruit_BBIO-1.1.1
creating Adafruit_BBIO-1.1.1/Adafruit_BBIO
creating Adafruit_BBIO-1.1.1/Adafruit_BBIO.egg-info
creating Adafruit_BBIO-1.1.1/overlays
creating Adafruit_BBIO-1.1.1/source
creating Adafruit_BBIO-1.1.1/source/include
creating Adafruit_BBIO-1.1.1/source/include/adafruit
creating Adafruit_BBIO-1.1.1/source/include/adafruit/bbio
creating Adafruit_BBIO-1.1.1/test
copying files to Adafruit_BBIO-1.1.1...
copying Adafruit_I2C.py -> Adafruit_BBIO-1.1.1
copying CHANGELOG.md -> Adafruit_BBIO-1.1.1
copying MANIFEST.in -> Adafruit_BBIO-1.1.1
copying README.md -> Adafruit_BBIO-1.1.1
copying distribute_setup.py -> Adafruit_BBIO-1.1.1
copying setup.py -> Adafruit_BBIO-1.1.1
copying Adafruit_BBIO/Encoder.py -> Adafruit_BBIO-1.1.1/Adafruit_BBIO
copying Adafruit_BBIO/__init__.py -> Adafruit_BBIO-1.1.1/Adafruit_BBIO
copying Adafruit_BBIO/sysfs.py -> Adafruit_BBIO-1.1.1/Adafruit_BBIO
copying Adafruit_BBIO.egg-info/PKG-INFO -> Adafruit_BBIO-1.1.1/Adafruit_BBIO.egg-info
copying Adafruit_BBIO.egg-info/SOURCES.txt -> Adafruit_BBIO-1.1.1/Adafruit_BBIO.egg-info
copying Adafruit_BBIO.egg-info/dependency_links.txt -> Adafruit_BBIO-1.1.1/Adafruit_BBIO.egg-info
copying Adafruit_BBIO.egg-info/top_level.txt -> Adafruit_BBIO-1.1.1/Adafruit_BBIO.egg-info
copying overlays/ADAFRUIT-SPI0-00A0.dts -> Adafruit_BBIO-1.1.1/overlays
copying overlays/ADAFRUIT-SPI1-00A0.dts -> Adafruit_BBIO-1.1.1/overlays
copying overlays/ADAFRUIT-UART1-00A0.dts -> Adafruit_BBIO-1.1.1/overlays
copying overlays/ADAFRUIT-UART2-00A0.dts -> Adafruit_BBIO-1.1.1/overlays
copying overlays/ADAFRUIT-UART4-00A0.dts -> Adafruit_BBIO-1.1.1/overlays
copying overlays/ADAFRUIT-UART5-00A0.dts -> Adafruit_BBIO-1.1.1/overlays
copying overlays/__init__.py -> Adafruit_BBIO-1.1.1/overlays
copying overlays/builder.py -> Adafruit_BBIO-1.1.1/overlays
copying source/c_adc.c -> Adafruit_BBIO-1.1.1/source
copying source/c_adc.h -> Adafruit_BBIO-1.1.1/source
copying source/c_pinmux.c -> Adafruit_BBIO-1.1.1/source
copying source/c_pinmux.h -> Adafruit_BBIO-1.1.1/source
copying source/c_pwm.c -> Adafruit_BBIO-1.1.1/source
copying source/c_pwm.h -> Adafruit_BBIO-1.1.1/source
copying source/c_uart.c -> Adafruit_BBIO-1.1.1/source
copying source/c_uart.h -> Adafruit_BBIO-1.1.1/source
copying source/common.c -> Adafruit_BBIO-1.1.1/source
copying source/common.h -> Adafruit_BBIO-1.1.1/source
copying source/constants.c -> Adafruit_BBIO-1.1.1/source
copying source/constants.h -> Adafruit_BBIO-1.1.1/source
copying source/event_gpio.c -> Adafruit_BBIO-1.1.1/source
copying source/event_gpio.h -> Adafruit_BBIO-1.1.1/source
copying source/py_adc.c -> Adafruit_BBIO-1.1.1/source
copying source/py_gpio.c -> Adafruit_BBIO-1.1.1/source
copying source/py_pwm.c -> Adafruit_BBIO-1.1.1/source
copying source/py_uart.c -> Adafruit_BBIO-1.1.1/source
copying source/spimodule.c -> Adafruit_BBIO-1.1.1/source
copying source/include/adafruit/bbio.h -> Adafruit_BBIO-1.1.1/source/include/adafruit
copying source/include/adafruit/bbio/error.h -> Adafruit_BBIO-1.1.1/source/include/adafruit/bbio
copying source/include/adafruit/bbio/gpio.h -> Adafruit_BBIO-1.1.1/source/include/adafruit/bbio
copying source/include/adafruit/bbio/pwm.h -> Adafruit_BBIO-1.1.1/source/include/adafruit/bbio
copying test/test_adc.py -> Adafruit_BBIO-1.1.1/test
copying test/test_gpio_input.py -> Adafruit_BBIO-1.1.1/test
copying test/test_gpio_output.py -> Adafruit_BBIO-1.1.1/test
copying test/test_gpio_setup.py -> Adafruit_BBIO-1.1.1/test
copying test/test_led.py -> Adafruit_BBIO-1.1.1/test
copying test/test_pwm_setup.py -> Adafruit_BBIO-1.1.1/test
copying test/test_spi.py -> Adafruit_BBIO-1.1.1/test
copying test/test_uart.py -> Adafruit_BBIO-1.1.1/test
Writing Adafruit_BBIO-1.1.1/setup.cfg
creating dist
Creating tar archive
removing 'Adafruit_BBIO-1.1.1' (and everything under it)
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo chown -R debian:debian .
```
and uploaded to PyPI:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ twine upload dist/Adafruit_BBIO-1.1.1.tar.gz 
Uploading distributions to https://upload.pypi.org/legacy/
Enter your username: fustini
Enter your password: 
Uploading Adafruit_BBIO-1.1.1.tar.gz
[================================] 109334/109334 - 00:00:03
```

pdp7, 2018-11-20:
Here is the output I received:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ sudo pip3 install --upgrade --force-reinstall Adafruit_BBIO==1.1.1
Collecting Adafruit_BBIO==1.1.1
  Downloading https://files.pythonhosted.org/packages/53/2b/b0e3dce6113225aae9beb886b2addd4fd5c140ba93c9503d7e4a97021bcc/Adafruit_BBIO-1.1.1.tar.gz (76kB)
    100% |████████████████████████████████| 81kB 793kB/s 
Building wheels for collected packages: Adafruit-BBIO
  Running setup.py bdist_wheel for Adafruit-BBIO ... error
  Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-11oxulmd/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmpmjzp4es1pip-wheel- --python-tag cp35:
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for Adafruit-BBIO
  Running setup.py clean for Adafruit-BBIO
Failed to build Adafruit-BBIO
Installing collected packages: Adafruit-BBIO
  Running setup.py install for Adafruit-BBIO ... done
Successfully installed Adafruit-BBIO-1.1.1
```

@GregoireWulliamoz I am uncertain if this prevents the library from working or not.  Where you unable to use Adafruit_BBIO?