adafruit/adafruit-beaglebone-io-python #294: Beaglebone Blue “ValueError: Set gpio mode failed, missing file or invalid permissions."

deandinglee, 2018-11-15:
Hello，I am running Debian 9.5 2018-10-07 4GB SD IoT on my beaglebone blue,
when i runing following code:

`
import threading
import time
import Adafruit_BBIO.GPIO as GPIO
infPIN = 'GPIO1_25'
count = 0
rpm = 0
GPIO.setup(infPIN, GPIO.IN)

def intr():
    print('test a')
    t = threading.Timer(1, intr)
    t.start()

t = threading.Timer(1,intr)
t.start()
while True:
    inf_state = GPIO.input(infPIN)
    if inf_state != 0 :
        count += 1
        print(count)
    time.sleep(0.1)

`

But I get this error:

`
Traceback (most recent call last):
  File "/var/lib/cloud9/rpm2.py", line 5, in <module>
    GPIO.setup("infPIN", GPIO.IN)
ValueError: Set gpio direction failed, missing file or invalid permissions.

`
I have tried updating kernel and bootloader according to issue #248 and #261, but that didn't work.
And i have tried to use "GPIO1_25"  and "GP0_3", but it still no work 

I will be appreciated if you can help me

pdp7, 2018-11-20:
@deandinglee  Which [physical pin on the BeagleBone Blue](https://github.com/beagleboard/beaglebone-blue/wiki/Pinouts) are you trying to access with GPIO1_25?

cc @jadonk @MarkAYoder

deandinglee, 2018-11-20:
@pdp7  I am trying to access one of GPIO pin (gp03)  on GP0 connector with GPIO_25

pdp7, 2018-11-20:
For reference, from bone.js:
https://github.com/jadonk/bonescript/blob/master/src/bone.js#L1586
```
    {
        "name": "GPIO1_25",
        "gpio": 57,
        "mux": "gpmc_wpn",
        "eeprom": null,
        "key": ["GP0_3", "P2_6"],
        "muxRegOffset": "0x064",
        "options": [
            "gpmc_wpn",
            "gmii2_rxer",
            "gpmc_csn5",
            "rmii2_rxer",
            "mmc2_sdcd",
            "pr1_mii1_txen",
            "uart4_txd",
            "gpio1_25"
        ],
        "ball": {
            "ZCZ": "U16",
            "BSM": "T15"
        }
```
This should be pin 3 on the GP0 connector on the Blue.

I'll have to investigate why this creates an error,

From common.c:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L172
```
  // These are for the Blue
    { "GP0_3", "GP0_3",  57, -1, -1},
```

pdp7, 2018-11-23:
Test program:
```
import Adafruit_BBIO.GPIO as GPIO
infPIN = 'GPIO1_25'
print("infPIN={0}", infPIN)
print("GPIO.setup()")
GPIO.setup(infPIN, GPIO.IN)
print("GPIO.setup() complete")
#inf_state = GPIO.input(infPIN)
```
Running test with strace:
```
debian@beaglebone:~$ strace -o /tmp/x -f python issue294-b.py 
```
Set direction of gpio57 to `in` seems to work OK:
```
1150  access("/sys/class/gpio/gpio57", R_OK|W_OK|X_OK) = -1 ENOENT (No such file or directory)
1150  open("/sys/class/gpio/export", O_WRONLY) = 3
1150  write(3, "57", 2)                 = 2
1150  close(3)                          = 0
1150  open("/sys/class/gpio/gpio57/direction", O_WRONLY) = 3
1150  write(3, "in", 2)                 = 2
1150  close(3)                          = 0
```

However, attempt to set the pinmux in the config-pin (e.g. cape-universal style) fails:
```
1150  open("/sys/devices/platform/ocp/ocp:GPIO1_pinmux/state", O_WRONLY|O_CREAT|O_TRUNC, 0666) = -1 ENOENT (No such file or directory)
```

And then gpio57 is unexported before the program exists:
```
1150  open("/sys/class/gpio/unexport", O_WRONLY) = 3
1150  write(3, "57", 2)                 = 2
1150  close(3)                          = 0
(END)
```

pdp7, 2018-11-23:
```
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ strace -o /tmp/x -f python ~/issue294-b.py 
('infPIN={0}', 'GPIO1_25')
GPIO.setup()
DEBUG: Adafruit_BBIO: set_pin_mode() :: Pinmux file: /sys/devices/platform/ocp/ocp:GPIO1_pinmux/state, mode: gpioDEBUG: Adafruit_BBIO: set_pin_mode(): beaglebone_blue() is TRUE; return BBIO_OK
GPIO.setup() complete
```

```diff
debian@beaglebone:~/ssh/adafruit-beaglebone-io-python$ git diff
diff --git a/source/c_pinmux.c b/source/c_pinmux.c
index 6932ec6..d22272b 100644
--- a/source/c_pinmux.c
+++ b/source/c_pinmux.c
@@ -32,13 +32,23 @@ BBIO_err set_pin_mode(const char *key, const char *mode)
        snprintf(pinmux_dir, sizeof(pinmux_dir), "ocp:%s_pinmux", pin);
        snprintf(path, sizeof(path), "%s/%s/state", ocp_dir, pinmux_dir);
 
+       /* beaglebone blue has complete dtb file and does not need overlays */
+       if(beaglebone_blue()) {
+               fprintf(stderr, "DEBUG: Adafruit_BBIO: set_pin_mode() :: Pinmux file: %s, mode: %s", path, mode); 
+               fprintf(stderr, "DEBUG: Adafruit_BBIO: set_pin_mode(): beaglebone_blue() is TRUE; return BBIO_OK\n");
+               return BBIO_OK;
+       }
+
+
        f = fopen(path, "w");
        if (NULL == f) {
                return BBIO_ACCESS;
        }
        syslog(LOG_DEBUG, "Adafruit_BBIO: set_pin_mode() :: Pinmux file %s access OK", path); 
+       fprintf(stderr, "Adafruit_BBIO: set_pin_mode() :: Pinmux file %s access OK", path); 
        fprintf(f, "%s", mode);
        fclose(f);
        syslog(LOG_DEBUG, "Adafruit_BBIO: set_pin_mode() :: Set pinmux mode to %s for %s", mode, pin);
+       fprintf(stderr, "Adafruit_BBIO: set_pin_mode() :: Set pinmux mode to %s for %s", mode, pin);
        return BBIO_OK;
 }
```

deandinglee, 2018-11-24:
@pdp7 I have tried the current version, There's no error anymore, but it seems like the hardware still not work.  when i run 
**import Adafruit_BBIO.GPIO as GPIO
infPIN = 'GP0_3'
GPIO.setup(infPIN, GPIO.OUT)
GPIO.output(infPIN, GPIO.HIGH)
**
and
**import Adafruit_BBIO.GPIO as GPIO
infPIN = 'GP0_3'
GPIO.setup(infPIN, GPIO.OUT)
GPIO.output(infPIN, GPIO.LOW)
**
the extern LED can not be turn on or turn off

pdp7, 2018-11-24:
@deandinglee thanks for the feedback.  I will investigate further.

pdp7, 2019-12-17:
@deandinglee hi, I'm going through old issues and I wanted to check if this was still an issue. thanks

closed 2020-06-10 by pdp7