adafruit/adafruit-beaglebone-io-python #297: [Request] What is the roadmap like for kernel 4.19.x support?

elfgoh, 2018-12-11:
Given 4.19.x is a long term support kernel, may I know the rough timeline to support the kernel? Also happy to help test if there is a guide on how to do that 🙇

RobertCNelson, 2018-12-11:
@elfgoh currently there are 4 variants of the v4.19.x out for the Beagle's..  It should be pretty seamless from v4.19.x, except for spi names are back to mainline defaults.

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#4.19.x-ti

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Mainline_.284.19.x_lts.29

Regards,

elfgoh, 2018-12-11:
> except for spi names are back to mainline defaults.

Sorry @RobertCNelson , I need a bit more context for this. Havent used the library with SPI and definitely am unfamiliar with the mainline defaults 🙇

RobertCNelson, 2018-12-11:
On mainline, sometime before v4.14.x, the spi port name we re-indexed using the actual node name, thus spi0 in the dts was spi0 in the kernel user-space..

Thus when originally comparing v4.14.x to v4.4.x/v4.9.x: spidev0(v4.14.x) = spidev1(older)...

We reverted that change in v4.14.x, to make it more seamless for v4.4.x -> v4.14.x.

So now with v4.19.x with are back to mainline and will be flushing out all the spidev references once again..  But this time all the spi access software should be fixed... ;) @pdp7 

Regards,

elfgoh, 2018-12-11:
I see. Great! 

Does that mean if I am not using SPI, it *could* probably be safe to use the library with 4.19.x based on the latest stable release?

RobertCNelson, 2018-12-11:
'could' .... i'd say, unless you are using spi, you wouldn't notice you were on v4.19.x vs v4.14.x...

Regards,

elfgoh, 2018-12-11:
Thanks a lot for the clarification @RobertCNelson !

pdp7, 2018-12-11:
@elfgoh thanks for asking about 4.19.x.  the goal of this library is work on the most recent kernel builds by @RobertCNelson.

Please open up new issue(s) if you encounter any functionality not working with 4.19.x kernel builds.

closed 2018-12-11 by pdp7