adafruit/adafruit-beaglebone-io-python #298: Can't install, compile errors

fromeijn, 2018-12-14:
Hey guys,

I'm trying to install this package on a BeagleBone Black with a Kernel 4.14. tried to install it with pip, pip3 and with python(3) setup.py install giving the same results. I have a custom image generated with yocto. Any help would be really appreciated. If more information is needed, please tell me.

```
pip3 install Adafruit_BBIO
Collecting Adafruit_BBIO
  Downloading https://files.pythonhosted.org/packages/53/2b/b0e3dce6113225aae9beb886b2addd4fd5c140ba93c9503d7e4a97021bcc/Adafruit_BBIO-1.1.1.tar.gz (76kB)
    100% |################################| 81kB 620kB/s 
Installing collected packages: Adafruit-BBIO
  Running setup.py install for Adafruit-BBIO ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-b_wdzkc6/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-r0ed8uex/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    copying Adafruit_I2C.py -> build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/overlays
    copying overlays/__init__.py -> build/lib.linux-armv7l-3.5/overlays
    copying overlays/builder.py -> build/lib.linux-armv7l-3.5/overlays
    creating build/lib.linux-armv7l-3.5/Adafruit_BBIO
    copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-3.5/Adafruit_BBIO
    copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-3.5/Adafruit_BBIO
    copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-3.5/Adafruit_BBIO
    running build_ext
    building 'Adafruit_BBIO.GPIO' extension
    creating build/temp.linux-armv7l-3.5
    creating build/temp.linux-armv7l-3.5/source
    arm-poky-linux-gnueabi-gcc -march=armv7-a -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a8 -DNDEBUG -fno-inline -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wno-unused-result -Wsign-compare -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/python3/3.5.6-r1.0=/usr/src/debug/python3/3.5.6-r1.0 -fdebug-prefix-map== -fdebug-prefix-map== -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/python3/3.5.6-r1.0=/usr/src/debug/python3/3.5.6-r1.0 -fdebug-prefix-map== -fdebug-prefix-map== -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.5m -c source/py_gpio.c -o build/temp.linux-armv7l-3.5/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
    source/py_gpio.c:558:14: error: cast between incompatible function types from 'PyObject * (*)(PyObject *, PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *, struct _object *)'} to 'PyObject * (*)(PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *)'} [-Werror=cast-function-type]
        {"setup", (PyCFunction)py_setup_channel, METH_VARARGS | METH_KEYWORDS, "Set up the GPIO channel, direction and (optional) pull/up down control\nchannel        - Either: RPi board pin number (not BCM GPIO 00..nn number).  Pins start from 1\n                 or    : BCM GPIO number\ndirection      - INPUT or OUTPUT\n[pull_up_down] - PUD_OFF (default), PUD_UP or PUD_DOWN\n[initial]      - Initial value for an output channel\n[delay]        - Time in milliseconds to wait after exporting gpio pin"},
                  ^
    source/py_gpio.c:562:25: error: cast between incompatible function types from 'PyObject * (*)(PyObject *, PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *, struct _object *)'} to 'PyObject * (*)(PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *)'} [-Werror=cast-function-type]
        {"add_event_detect", (PyCFunction)py_add_event_detect, METH_VARARGS | METH_KEYWORDS, "Enable edge detection events for a particular GPIO channel.\nchannel      - either board pin number or BCM number depending on which mode is set.\nedge         - RISING, FALLING or BOTH\n[callback]   - A callback function for the event (optional)\n[bouncetime] - Switch bounce timeout in ms for callback"},
                             ^
    source/py_gpio.c:565:27: error: cast between incompatible function types from 'PyObject * (*)(PyObject *, PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *, struct _object *)'} to 'PyObject * (*)(PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *)'} [-Werror=cast-function-type]
        {"add_event_callback", (PyCFunction)py_add_event_callback, METH_VARARGS | METH_KEYWORDS, "Add a callback for an event already defined using add_event_detect()\ngpio         - gpio channel\ncallback     - a callback function\n[bouncetime] - Switch bounce timeout in ms"},
                               ^
    cc1: all warnings being treated as errors
    error: command 'arm-poky-linux-gnueabi-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-b_wdzkc6/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-r0ed8uex/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-b_wdzkc6/Adafruit-BBIO/ 
```

version of gcc:
```
arm-poky-linux-gnueabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-poky-linux-gnueabi-gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/arm-poky-linux-gnueabi/8.2.0/lto-wrapper
Target: arm-poky-linux-gnueabi
Configured with: ../../../../../../work-shared/gcc-8.2.0-r0/gcc-8.2.0/configure --build=x86_64-linux --host=arm-poky-linux-gnueabi --target=arm-poky-linux-gnueabi --prefix=/usr --exec_prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/libexec --datadir=/usr/share --sysconfdir=/etc --sharedstatedir=/com --localstatedir=/var --libdir=/usr/lib --includedir=/usr/include --oldincludedir=/usr/include --infodir=/usr/share/info --mandir=/usr/share/man --disable-silent-rules --disable-dependency-tracking --with-libtool-sysroot=/home/floris/jumpnowtech-thud/poky-thud/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/gcc/8.2.0-r0/recipe-sysroot --with-gnu-ld --enable-shared --enable-languages=c,c++ --enable-threads=posix --enable-multilib --enable-default-pie --enable-c99 --enable-long-long --enable-symvers=gnu --enable-libstdcxx-pch --program-prefix=arm-poky-linux-gnueabi- --without-local-prefix --enable-lto --disable-libssp --enable-libitm --disable-bootstrap --disable-libmudflap --with-system-zlib --with-linker-hash-style=gnu --enable-linker-build-id --with-ppl=no --with-cloog=no --enable-checking=release --enable-cheaders=c_global --without-isl --with-float=hard --with-sysroot=/ --with-build-sysroot=/home/floris/jumpnowtech-thud/poky-thud/build/tmp/work/cortexa8hf-neon-poky-linux-gnueabi/gcc/8.2.0-r0/recipe-sysroot --with-gxx-include-dir=/usr/include/c++/8.2.0 --without-long-double-128 libgcc_cv_powerpc_float128=no --disable-static --enable-nls --enable-initfini-array --with-arch=armv7-a+fp
Thread model: posix
gcc version 8.2.0 (GCC) ```

pdp7, 2018-12-16:
This library only officially supports the BeagleBoard.org Debian images.

On Debian, these packages are needed:
> sudo apt-get install build-essential python-dev python-pip -y

Here is what is in build-essential:
https://packages.debian.org/stretch/build-essential

pdp7, 2018-12-19:
@fromeijn I'd suggest adding Yocto/OpenEmbedded recipes for the software listed in:
https://packages.debian.org/stretch/build-essential
https://packages.debian.org/stretch/python-dev
https://packages.debian.org/stretch/python-pip

I don't have Yocto environment to test myself.

RobertCNelson, 2019-02-05:
@pdp7 grab the buster image, (buster ships 8.2.0 right now)

https://rcn-ee.net/rootfs/bb.org/testing/2019-02-03/buster-iot/

Regards,

TristanDewalle, 2019-02-10:
@fromeijn as quick and (really) dirty way to be able to install (and use) this lib, i removed "-Werror" from CFLAGS list in setup.py (manual installation).

pdp7, 2019-12-17:
I am going to close this in favor of tracking this problem under issue #308 

I believe I have a fix in PR #321

closed 2019-12-17 by pdp7