adafruit/adafruit-beaglebone-io-python #301: Beaglebone Black “ValueError: Set gpio mode failed, missing file or invalid permissions."

jskusk, 2019-01-15:
Hi,
My python2.7 program fails and from strace I can see the permission denied issues.

```
access("/sys/class/gpio/gpio30", R_OK|W_OK|X_OK) = -1 ENOENT (No such file or directory) 
open("/sys/class/gpio/export", O_WRONLY) = 3 
write(3, "30", 2)                       = 2 
close(3)                                = 0 
nanosleep({tv_sec=0, tv_nsec=200000000}, NULL) = 0 
open("/sys/class/gpio/gpio30/direction", O_WRONLY) = 3 
write(3, "out", 3)                      = 3 
close(3)                                = 0 
open("/sys/class/gpio/gpio30/value", O_WRONLY) = 3 
write(3, "0", 1)                        = 1 
close(3)                                = 0 
access("/sys/class/gpio/gpio31", R_OK|W_OK|X_OK) = -1 ENOENT (No such file or directory) 
open("/sys/class/gpio/export", O_WRONLY) = 3 
write(3, "31", 2)                       = 2 
close(3)                                = 0 
nanosleep({tv_sec=0, tv_nsec=200000000}, NULL) = 0 
open("/sys/class/gpio/gpio31/direction", O_WRONLY) = -1 EACCES (Permission denied) 
gettimeofday({tv_sec=1547553455, tv_usec=216996}, NULL) = 0 
getpid()                                = 9969 
send(4, "<139>Jan 15 11:57:35 python[9969"..., 145, MSG_NOSIGNAL) = 145 
write(2, "Traceback (most recent call last"..., 35Traceback (most recent call last): 
) = 35 
write(2, "  File \"test_bbbio.py\", line 17,"..., 45  File "test_bbbio.py", line 17, in <module> 
) = 45 
open("test_bbbio.py", O_RDONLY|O_LARGEFILE) = 3 
fstat64(3, {st_mode=S_IFREG|0644, st_size=727, ...}) = 0 
read(3, "import Adafruit_BBIO.GPIO as GPI"..., 4096) = 727 
write(2, "    ", 4    )                     = 4 
write(2, "GPIO.setup(DO_TTL1, GPIO.OUT) # "..., 62GPIO.setup(DO_TTL1, GPIO.OUT) # red led. (Blink) if in error. 
) = 62 
close(3)                                = 0 
write(2, "ValueError", 10ValueError)              = 10 
write(2, ": ", 2: )                       = 2 
write(2, "Set gpio direction failed, missi"..., 63Set gpio direction failed, missing file or invalid permissions.) = 63 
write(2, "\n", 1 
)                       = 1 
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d2efd1}, {sa_handler=0x5bc19d, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d2efd1}, 8) = 0 
close(-1)                               = -1 EBADF (Bad file descriptor) 
open("/sys/class/gpio/unexport", O_WRONLY) = 3 
write(3, "30", 2)                       = 2 
close(3)                                = 0 
open("/sys/class/gpio/unexport", O_WRONLY) = 3 
write(3, "31", 2)                       = 2 
close(3)                                = 0

```

Python outputs the error for DO_TTL1 and not the previous TTL0.
```
Traceback (most recent call last):
  File "test_bbbio.py", line 17, in <module>
    GPIO.setup(DO_TTL1, GPIO.OUT) # red led. (Blink) if in error.
ValueError: Set gpio direction failed, missing file or invalid permissions.

```
Here is the python source
```
import Adafruit_BBIO.GPIO as GPIO
import time

# TTL
DO_TTL0 = "P9_11"
DO_TTL1 = "P9_13"
DO_RLY0 = "P9_17"
DO_RLY1 = "P9_18"
DO_RLY2 = "P9_21"
DO_RLY3 = "P9_22"
DI0 = "P9_12"
DI1 = "P9_14"
DI2 = "P9_15"
DI3 = "P9_16"

GPIO.setup(DO_TTL0, GPIO.OUT) # green led. (Blink) if await netsync and (solid) if on grid and ready
GPIO.setup(DO_TTL1, GPIO.OUT) # red led. (Blink) if in error.
GPIO.setup(DO_RLY0, GPIO.OUT) # in fault
GPIO.setup(DO_RLY1, GPIO.OUT)
GPIO.setup(DO_RLY2, GPIO.OUT)
GPIO.setup(DO_RLY3, GPIO.OUT)

GPIO.setup(DI0, GPIO.IN)      # stop (Coast and disconnect the grid)
GPIO.setup(DI1, GPIO.IN)      # start/frigivet til start.
GPIO.setup(DI2, GPIO.IN)      # reset all three controllers
GPIO.setup(DI3, GPIO.IN)
```

python -V
`Python 2.7.13`

pip list | grep Ada
`Adafruit-BBIO (1.1.1)`

One would expect permission issues but I can not find any. Here are the /sys/class/gpio folder where both gpio30 gpio31 is missing and are therefore to be exported by Adafruit before usage. From the strace it seems to work fine for gpio30 but not 31. (mind that P9_11 is gpio30 and P9_13 is gpio31)
```
$ ls -lsa | grep -e gpio3[0-2] -e export
0 -rw-rw----  1 root gpio 4096 Jan 15 10:29 export
0 lrwxrwxrwx  1 root gpio    0 Jan 15 09:56 gpio32 -> ../../devices/platform/ocp/4804c000.gpio/gpiochip1/gpio/gpio32
0 -rw-rw----  1 root gpio 4096 Jan 15 10:45 unexport
```

I can solve the issue by exporting manually gpio31 with my debian user (member of gpio)
`echo 31 > export` and therefore I should not have any permission issues.

I use my custom made cape named EPC that make use of the beaglebones I2C pins thus I have disabled I2C and auto cape detection. 
```
root@beaglebone:/opt/scripts/tools# sh version.sh
git:/opt/scripts/:[e9bcff232834702c1c810710706ee815d77b080b]
eeprom:[A335BNLTEIA04816BBBK34D1]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.79-ti-rt-r84]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-CAN0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-CAN1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/EPC-V2-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=0]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2]
pkg:[librobotcontrol]:[1.0.2-git20180829.0-0rcnee0~stretch+20180830]
pkg:[firmware-ti-connectivity]:[20161130-4]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0]
dmesg | grep pinctrl-single
[    1.141118] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.163183] gpio-of-helper ocp:cape-universal: ready
END
```

Above leads me to think that Adafruit could have an bug and I need some help to figure out if so.

Note that I have tried version 1.0.3 and 1.0.10 also.

jskusk, 2019-01-29:
My temporary solution is to use ExecStartPre in systemd. E.g.

```
[Unit]
Description=can_controller
#After=syslog.target network.target
BindsTo=sys-subsystem-net-devices-can0.device
#After=sys-subsystem-net-devices-can0.device
After=canparser.service

[Service]
User=debian
Group=debian
PIDFile=/run/iocont.pid
WorkingDirectory=/home/debian/src/orbiepc/src/epc/iocont
**ExecStartPre=/bin/bash -c "echo 31 > /sys/class/gpio/export"**
ExecStart=/usr/bin/python main.py
ExecReload=/bin/kill -1 $MAINPID
ExecStop=/bin/kill -9 $MAINPID
#KillMode=control-groups
TimeoutSec=10
Restart=on-failure

[Install]
WantedBy=multi-user.target

```

pdp7, 2019-02-05:
@RobertCNelson any insights as to why this configuration needs `ExecStartPre=/bin/bash -c "echo 31 > /sys/class/gpio/export"` in order for P9_13 to be available for GPIO?

RobertCNelson, 2019-02-05:
i don't know... without seeing, EPC-V2-00A0.dtbo

Regards,

pdp7, 2019-02-05:
@jskusk could you please post the device tree source file (.dts) for EPC-V2-00A0.dtbo?

pdp7, 2019-02-13:
@jskusk is this still an issue for you?

pdp7, 2019-02-19:
Please re-open if still an issue

closed 2019-02-19 by pdp7