adafruit/adafruit-beaglebone-io-python #302: Reconfigure LCD data pins in Beaglebone Black

sergpenshin, 2019-01-28:
Hi averyone,
Is it possible to change LCD data pins to another GPIO? The fact is that my cape does not allow the Beaglebone Black to boot normally, because LCD pins  are used while BBB boot. Although if I connect it after boot BBB, then it works well. I tried to reconfigure the pins in the device tree so as not to use the boot pins, but there is no signal on them. Although the cape is attached in the slot.
Here's my device tree source:
https://yadi.sk/d/VjxN_P-Vl4x9Uw

pdp7, 2019-01-28:
Here is a good chart of the pins that can be used for GPIO:
https://elinux.org/Beagleboard:Cape_Expansion_Headers

pdp7, 2019-01-28:
You can disable the HDMI video and audio with change to /boot/uEnv.txt that @RobertCNelson describes on the eLinux wiki:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Disable_on-board_devices

Make sure that you are using the latest Debian image from BeagleBoard.org:
http://beagleboard.org/latest-images

closed 2019-02-13 by pdp7