adafruit/adafruit-beaglebone-io-python #303: Cannot Set P9_25 as GPIO

stumpja, 2019-01-31:
- Platform/operating system:  BeagleBone Black running latest image (Debian 9.5 2018-10-07 4GB SD IoT)

- Python version:  2.7.13

- Error message you are receiving, including any Python exception traces:
```python
debian@beaglebone:/home/Python_Files$ sudo python buttonPress.py
Traceback (most recent call last):
  File "buttonPress.py", line 8, in <module>
    GPIO.setup(DOWN,GPIO.IN,GPIO.PUD_UP)
ValueError: Set gpio mode failed, missing file or invalid permissions.
```

- List the steps to reproduce the problem below:
The issue I'm having has something to do with P9_25. For some reason I cannot set this pin as a GPIO. I have tried using config-pin command line script without success. I have tried the following:

```python
debian@beaglebone:/home/Python_Files$ config-pin p9.25 in
P9_25 pinmux file not found!
bash: /sys/devices/platform/ocp/ocp*P9_25_pinmux/state: No such file or directory
Cannot write pinmux file: /sys/devices/platform/ocp/ocp*P9_25_pinmux/state
debian@beaglebone:/home/Python_Files$

```

Full Code Below:
```python
# 2 Button Input Test Code on BBB
import Adafruit_BBIO.GPIO as GPIO
from time import sleep

UP="P9_42"
DOWN="P9_25"

GPIO.setup(UP,GPIO.IN,GPIO.PUD_UP) # INPUT, PULLUP
GPIO.setup(DOWN,GPIO.IN,GPIO.PUD_UP) #INPUT, PULLUP

GPIO.add_event_detect(UP,GPIO.FALLING)
GPIO.add_event_detect(DOWN,GPIO.FALLING)

BP=0

while(1):
	if GPIO.event_detected(UP):
		BP+=1
	if GPIO.event_detected(DOWN):
		BP-=1
	if BP>10:
		BP=10
	if BP<0:
		BP=0
	print BP
	sleep(.2)
```

My issue looks very similar to what @boineg was experiencing [here](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/224). However, the solution for him was to reflash image. I have already flashed the latest image. Am I doing something wrong.

boineg, 2019-01-31:
Hi, I actually flashed an older image because the latest one (back then) didn't work. For reference, I used bone-debian-9.1-lxqt-armhf-2017-08-31-4gb.img

jancotipsarevic, 2019-01-31:
Hi , I have same issue tryed with Jessie and Stretch (python3), it is my first time using Adafruit_BBIO (I was used to work with PyBBIO). It is strictly necessary to use  cape_universal to work with GPIOs?, If I don´t use cape_universal should I create my own dtbo to deal with GPIOs? .
The problem apears configuring pins as inputs but no problems with outputs (cape_universal dissabled).

 >>> import Adafruit_BBIO.GPIO as GPIO
>>> GPIO.setup("P8_12", GPIO.IN)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Set gpio mode failed, missing file or invalid permissions.

stumpja, 2019-01-31:
Latest issue is when I enable pullups, using a DMM the pin voltage is only ~1.5V. I can only assume both pull-up and pull-down are being enabled simultaneously. I believe this is causing some grief in my program.

Found a similar issue [here](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/5). Looks like issue should be resolved. I'm not sure how it was fixed in this thread, just seems like it should be fixed now. I have updated to the latest Adafruit_BBIO and I am running latest image for BBB.

When using config-pin, results are as follows:

```python
debian@beaglebone:~$ sudo config-pin P9.42 gpio_pu
```
P9_42 to GND = 1.568V
```python
debian@beaglebone:~$ sudo config-pin P9.42 gpio_pd
```
P9_42 to GND = 0.001V
```python
debian@beaglebone:~$ sudo config-pin P9.42 gpio
```
P9_42 to GND = 0.002V
```python
debian@beaglebone:~$ sudo config-pin P9.42 default
```
P9_42 to GND = 0.001V

Suggestions?

pdp7, 2019-02-05:
@RobertCNelson any suggestions for troubleshooting pull-up?

@stumpja reports that `sudo config-pin P9.42 gpio_pu` results in P9_42 to GND being only 1.568V.

RobertCNelson, 2019-02-05:
@stumpja ; P9_42 is a shared, what is P9_92 set to??

stumpja, 2019-02-05:
@RobertCNelson just yesterday I was going through @derekmolloy's book [Exploring BeagleBone](https://www.amazon.com/Exploring-BeagleBone-Techniques-Building-Embedded/dp/1118935128) and saw [here](https://itbrainpower.net/a-gsm/images/BeagleboneBlackP9HeaderTable.pdf) P9.41 and P9.42 both have shared pins. So I thought that the other pin probably had a pull down enabled by default, and that was why I was seeing these issues. I spent a lot of time trying to find if that other pin had a name I could use to set with config-pin without any luck, so thank you!

I am also using P9.41. Looks like it's shared pin name is P9.91 by deduction. After I changed these pins with config-pin to in+, I am now seeing a voltage closer to 3.3V on both. However, where can this information be found?

I wish the 2nd edition of @derekmolloy's book would have been out when I purchased it. I likely will pick it up anyways.

RobertCNelson, 2019-02-05:
@stumpja , so the "P9_91" and "P9_92" pin names where actually come up by the author of "config-pin" for MachineKit, and we just merged that script into the install by default.  So they became those pin names by default, and never officially had any name.. 

Regards,

pdp7, 2019-02-05:
@stumpja is this issue resolved now? thanks

stumpja, 2019-02-05:
@pdp7 this can be closed.

closed 2019-02-05 by stumpja