adafruit/adafruit-beaglebone-io-python #304: BeageBone Black wireless PWM issue

Locho14590, 2019-02-25:
Hi,

I just got the BeagleBone Black Wireless. I am new in Python, Beaglebone. I am running these code:
import Adafrut_BBIO.PWM as PWM
myPWM="P8_13"
PWM.start(myPWM, 0 , 1000)
for i in range (0,5)
       DC = input ("Input Duty Cycle")
       PWM.set_duty_cycle(myPWM, DC)
PWM.stop(myPWM)
PWM.cleanup()

it said that:
DC= input("Input Duty Cycle")
IOError: [Errno 2] No such file or directory: '/sys/devices/platform/bone_capemgr/slots'

Here are the result of some commands that i run on debian@beaglebone:~$ :
~$ cat /etc/dogtag
BeagleBoard.org Debian Image 2018-10-07
~$ cat /etc/debian_version
9.8
~$ cat /proc/cmdline
console=ttyo0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M
~$ cat /sys/devices/platform/bone_capemgr/slots
cat: /sys/devices/platform/bone_capemgr/slots: No such file or directory
~$ cat /boot/uEnv.txt
#this uEnv.txt created by configure_robotics_dt
uname_r=4.14.71-ti-r80
dtb=am335x-boneblack-wireless-roboticscape.dtb
cmdline=coherent_pool=1M

I read some comments, but I do not really sure what should I do to get the signal from the output of PWM pins.
Please help me to figure out.

Thanks

pdp7, 2019-02-25:
Please paste the output of:
/opt/scripts/tools/version.sh

This will provide all the necessary information.

RobertCNelson, 2019-02-25:
@Locho14590 , do you really have a Roboticscape plugged in?

```
#this uEnv.txt created by configure_robotics_dt
dtb=am335x-boneblack-wireless-roboticscape.dtb
```

Regards,

Locho14590, 2019-02-25:
@RobertCNelson ,

Thanks for response. I really appreciated you.
I do not have any Roboticscape on yet. The only thing I need for project is the PWM, and control PWM through the web service. 

debian@beaglebone:~$ cat /opt/scripts/tools/version.sh
#!/bin/sh -e
#

if ! id | grep -q root; then
        echo "must be run as root"
        exit
fi

omap_bootloader () {
        unset test_var
        test_var=$(dd if=${drive} count=6 skip=393248 bs=1 2>/dev/null || true)
        if [ "x${test_var}" = "xU-Boot" ] ; then
                uboot=$(dd if=${drive} count=32 skip=393248 bs=1 2>/dev/null ||                                   true)
                uboot=$(echo ${uboot} | awk '{print $2}')
                echo "bootloader:[${label}]:[${drive}]:[U-Boot ${uboot}]:[locati                                  on: dd MBR]"
        else
                if [ -f /boot/uboot/u-boot.img ] ; then
                        if [ -f /usr/bin/mkimage ] ; then
                                unset uboot
                                uboot=$(/usr/bin/mkimage -l /boot/uboot/u-boot.i                                  mg | grep Description | head -n1 | awk '{print $3}' 2>/dev/null || true)
                                if [ ! "x${uboot}" = "x" ] ; then
                                        echo "bootloader:[${label}]:[${drive}]:[                                  U-Boot ${uboot}]:[location: fatfs /boot/uboot/MLO]"
                                else
                                        unset uboot
                                        uboot=$(/usr/bin/mkimage -l /boot/uboot/                                  u-boot.img | grep Name:| head -n1 | awk '{print $4}' 2>/dev/null || true)
                                        if [ ! "x${uboot}" = "x" ] ; then
                                                echo "bootloader:[${label}]:[${d                                  rive}]:[U-Boot ${uboot}]:[location: fatfs /boot/uboot/MLO]"
                                        fi
                                fi
                        fi
                else
                        if [ -b ${drive}p1 ] ; then
                                #U-Boot SPL 2014.04
                                unset test_var
                                test_var=$(dd if=${drive}p1 count=6 skip=284077                                   bs=1 2>/dev/null || true)
                                if [ "x${test_var}" = "xU-Boot" ] ; then
                                        uboot=$(dd if=${drive}p1 count=34 skip=2                                  84077 bs=1 2>/dev/null || true)
                                        uboot=$(echo ${uboot} | awk '{print $3}'                                  )
                                        echo "bootloader:[${label}]:[${drive}]:[                                  U-Boot ${uboot}]:[location: fatfs /boot/uboot/MLO]"
                                fi
                        fi
                fi
        fi
}

dpkg_check_version () {
        unset pkg_version
        pkg_version=$(dpkg -l | awk '$2=="'$pkg'" { print $3 }' || true)
        if [ ! "x${pkg_version}" = "x" ] ; then
                echo "pkg:[$pkg]:[$pkg_version]"
        else
                echo "WARNING:pkg:[$pkg]:[NOT_INSTALLED]"
        fi
}

dpkg_check_version_replaced () {
        unset pkg_version
        pkg_version=$(dpkg -l | awk '$2=="'$pkg'" { print $3 }' || true)
        if [ ! "x${pkg_version}" = "x" ] ; then
                echo "pkg:[$pkg]:[$pkg_version]:[GOT_REPLACED_BY_NEXT]"
        fi
}

if [ -f /usr/bin/git ] ; then
[... 103 lines cut ...]
        echo "groups:[`groups debian`]"
fi

if [ -d /home/machinekit/ ] ; then
        pkg="firmware-ti-connectivity" ; dpkg_check_version
        echo "groups:[`groups machinekit`]"
fi

if [ -d /home/ubuntu/ ] ; then
        echo "groups:[`groups ubuntu`]"
fi

if [ -d /home/beagle/ ] ; then
        echo "groups:[`groups beagle`]"
fi

echo "cmdline:[`cat /proc/cmdline`]"

echo "dmesg | grep pinctrl-single"
dmesg | grep pinctrl-single
echo "dmesg | grep gpio-of-helper"
dmesg | grep gpio-of-helper
echo "END"

#

RobertCNelson, 2019-02-25:
@Locho14590 , then in /boot/uEnv.txt remove this line:
```
dtb=am335x-boneblack-wireless-roboticscape.dtb
```

Drew ment for you to run the script:

```
sudo /opt/scripts/tools/version.sh
```

Locho14590, 2019-02-25:
@RobertCNelson ,

I am not sure which one that i did correctly.
root@beaglebone:/home/debian# /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBWA51905BBWG0021]
model:[TI_AM335x_BeagleBone_Black_Wireless_RoboticsCape]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.16.0]
device-tree-override:[dtb=am335x-boneblack-wireless-roboticscape.dtb]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190215.0-0rcnee0~stretch+20190215]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M]
dmesg | grep pinctrl-single
[    1.076815] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper

RobertCNelson, 2019-02-25:
@Locho14590 just remove this line from /boot/uEnv.txt and reboot..

```
dtb=am335x-boneblack-wireless-roboticscape.dtb
```
and then the pwm script work fine.

Regards,

Locho14590, 2019-02-25:
@RobertCNelson ,
I am sorry. I am new with Linux. Forgive for me if I ask something stupid. How can I got to /boot/uEnv.txt and modify that line?
root@beaglebone:/home/debian# /boot/uEnv.txt
bash: /boot/uEnv.txt: Permission denied
root@beaglebone:/home/debian# cat /boot/uEnv.txt

# this uEnv.txt created by configure_robotics_dt

uname_r=4.14.71-ti-r80
dtb=am335x-boneblack-wireless-roboticscape.dtb
cmdline=coherent_pool=1M

Thanks,

RobertCNelson, 2019-02-25:
sudo nano /boot/uEnv.txt

(ctrl-x) 
(y) - to save

Regards,

Locho14590, 2019-02-25:
@RobertCNelson ,

I changed it, and run the PWM code

debian@beaglebone:~/my_python$ python digOut.py
what is duty cycle50
Traceback (most recent call last):
  File "digOut.py", line 5, in <module>
    DC = input("what is duty cycle")
IOError: [Errno 2] No such file or directory: '/sys/devices/platform/bone_capemgr/slots'
debian@beaglebone:~/my_python$

RobertCNelson, 2019-02-25:
did you reboot?

Locho14590, 2019-02-25:
@RobertCNelson ,
Sorry, I have not done that.
This is after I reboot:
debian@beaglebone:~/my_python$ python digOut.py
Traceback (most recent call last):
  File "digOut.py", line 3, in <module>
    PWM.start(myPWM, 0, 1000)
RuntimeError: Unknown error

--------------------------------------------------------------
This is my PWM code:
import Adafruit_BBIO.PWM as PWM
myPWM="P8_13"
PWM.start(myPWM, 0, 1000)
for i in range(0,5):
        DC = input("what is duty cycle")
        PWM.set_duty_cycle(myPWM, DC)
PWM.stop(myPWM)
PWM.cleanup()

Locho14590, 2019-02-25:
@RobertCNelson ,

I tried to run the same code in root. It shows this:

root@beaglebone:~/testpwm# python pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 3, in <module>
    PWM.start(myPWM, 0, 1000)
RuntimeError: Problem with the cape manager
root@beaglebone:~/testpwm#

What should I do next to be able to make it work?

Thanks and regard,

RobertCNelson, 2019-02-25:
can you run:
```
sudo /opt/scripts/tools/version.sh
```

Again, as something isn't loading..

Locho14590, 2019-02-25:
@RobertCNelson ,

Thanks for helping me.

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBWA51905BBWG0021]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.16.0]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190215.0-0rcnee0~stretch+20190215]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M]
dmesg | grep pinctrl-single
[    1.046687] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.048258] gpio-of-helper ocp:cape-universal: ready
END
debian@beaglebone:~$

Regard,

RobertCNelson, 2019-02-25:
Yeap, your missing these options in /boot/uEnv.txt, they are the default, but the "this uEnv.txt created by configure_robotics_dt" script ripped them out..

```
enable_uboot_overlays=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
enable_uboot_cape_universal=1
``

Regards,

Locho14590, 2019-02-25:
@RobertCNelson ,
Please  stay with me. There is something still missing.

This is my sudo nano /boot/uEnv.txt

# this uEnv.txt created by configure_robotics_dt

enable_uboot_overlays=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
enalbe_uboot_cape_universal=1
uname_r=4.14.71-ti-r80
cmdline=coherent_pool=1M
-------------------------------------------------------------

I reboot it, then run the pwm
debian@beaglebone:~/my_python$ python digOut.py
Traceback (most recent call last):
  File "digOut.py", line 3, in <module>
    PWM.start(myPWM, 0, 1000)
RuntimeError: Unknown error
debian@beaglebone:~/my_python$

Here is sudo /opt/scripts/tools/version.sh in case you want to see it:

debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTBWA51905BBWG0021]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.16.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enalbe_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190215.0-0rcnee0~stretch+20190215]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M]
dmesg | grep pinctrl-single
[    1.049830] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.051036] gpio-of-helper ocp:cape-universal: ready
END
debian@beaglebone:~$

Regard,

RobertCNelson, 2019-02-25:
enalbe_uboot_cape_universal "enable" is spelled wrong..

Regards,

Locho14590, 2019-02-25:
@RobertCNelson ,

Dear, I found it. I spell wrong. I fixed it. and reboot and it works now.
I am so sorry, but happy in the same time. I appreciated you time and your help.

Thanks and best regard,

pdp7, 2019-03-05:
@Locho14590 is it ok to close this issue now?

pdp7, 2019-03-30:
Please reopen if still an issue

closed 2019-03-30 by pdp7