adafruit/adafruit-beaglebone-io-python #305: No AtributeError

Locho14590, 2019-03-04:
Hi,

I just got the BeagleBoneBlack Wireless. I an new with BBB. I try to run the example code using Web interface to control GPIO. When I run this code as main.py. It said attributeError. Can you give me some help what should I do or what I am missing please?

These are the error and the code that I used.

Thanks and best regard,

Ho

/////////////////////////////////////////////////////////////
debian@beaglebone:~/my_python$ python main.py
Traceback (most recent call last):
  File "main.py", line 57, in <module>
    app = web.application(urls, globals())
AttributeError: 'module' object has no attribute 'application'
debian@beaglebone:~/my_python$

///////////////////////////////////////////////////////////

import web
import Adafruit_BBIO.GPIO as GPIO

class Appliance:

        def _init_(self, name, pin_number):
                self.name = name
                self.pin_number = pin_number
                self.status = 0
                GPIO.setup(self.pin_number, GPIO.OUT)

        def output(self, status):
                if(status == 1):
                        self.status = 1
                        GPIO.output(self.pin_number, GPIO.HIGH)
                elif (status == 0):
                        self.status = 0
                        GPIO.output(self.pin_number, GPIO.LOW)
d = {}
def Access_Create(String, pin_number):
        d[string] = Appliance(string, pin_number)
def Access_Name(string):
        return d[string].name
def Access_Object(string):
        return d[string]
def Access_Delete(String):
        d.pop(string)
urls = (
        '/add', 'add',
        '/sw', 'sw',
        '/delete', 'delete',
        )
class add:
        def GET(self):
                i = web.input(name=None, pin=None)
                Access_Create(i.name, i.pin)
                print "Created an Appliance: " + i.name + "at pin number: " +i.$
class delete:
        def GET(self):
                i = web.input(name=None)
                item = Access_Object(i.name)
                item.output(0)
                Access_Delete(i.name)
                print "Deleted object: " + i.name
class sw:
        def GET(Self):
                i = web.input(name=None)
                item = Access_Object(i.name)
                if(item.status == 0):
                        item.output(1)
                        print i.name + "is turn on"
                elif(item.status == 1):
                        item.output(0)
                        print i.name + "is turn off"

if __name__ == "__main__":
    app = web.application(urls, globals())
    app.run()

Locho14590, 2019-03-04:
Sorry, somehow the typing mess up.

if __name__ == "__main__":
    app = web.application(urls, globals())
    app.run()

Thanks,

pdp7, 2019-03-05:
```
File "main.py", line 57, in 
app = web.application(urls, globals())
AttributeError: 'module' object has no attribute 'application'
```
This does not look like an error with Adafruit_BBIO.

Where does the `web` module come from?

Locho14590, 2019-03-05:
Hi, 

I just got the example from internet. I am trying to control GPIO from web interface.

Thanks

pdp7, 2019-03-06:
@Locho14590 I think the issue is the web interface example that you are using, and not an issue with Adafruit_BBIO.GPIO.

Does the web example run ok without the lines related to GPIO?

Locho14590, 2019-03-06:
Dear,

Yes, I figured out. That is web interface issue.

Thanks

pdp7, 2019-03-30:
@Locho14590 thanks for the update

closed 2019-03-30 by pdp7