adafruit/adafruit-beaglebone-io-python #306: eCAP Support

marcelnauta, 2019-03-07:
Hello,

I am looking to read the duty cycle of a PWM signal and use it alongside the rest of this library's functionalities. As far as I can tell, the best way to do this is to use the enhanced capture module (eCAP), but I do not see any support for that. Am I missing something? The only mention of eCAP I see when I search for it is that it can be used as an additional PWM output.

I am proficient in python, C, C++, have some experience with embedded systems, moderate experience with Linux in general, but next to no experience with Beagle Bone. I have scanned through section 15.3  of the AM335x manual and feel confident I could take this project on myself. Hopefully well enough to give it back to the community. Before I get started, Has anyone done this before? Any recommendations or caveats?

My goal will be to attach a PWM pin to an eCAP pin and be able to occasionally poll the duty cycle accurately, even at high frequencies. I will use a Beagle Bone Green with the default Linux kernel/python version for development.

Thanks,
Marcel

pdp7, 2019-03-14:
@marcelnauta yes, you are correct that there is no support for eCAP input in this library (Adafruit_BBIO).  The eCAP can be used as additional PWM output.

Unfortunately, there is not a supported eCAP Linux kernel as far as I know.  I believe Matt Porter wrote a basic driver several years ago but work never progressed beyond that:
https://lkml.org/lkml/2014/2/3/190

@jadonk @RobertCNelson do you know of any current efforts for eCAP input driver?

RobertCNelson, 2019-03-14:
Has anyone pinged @ohporter ;) Wonder if we could get to a GSOC student to resurrect his old driver..

marcelnauta, 2019-03-14:
Thanks for the update @pdp7. That does make it more involved than I expected, but I am still interested in making an attempt.

pdp7, 2019-12-17:
@marcelnauta did you make any progress on this?

@MarkAYoder / @jadonk - this might be a good task for student to finish

pdp7, 2021-03-31:
FYI - Python integration is still needed but eCAP (pulse input) driver is available as patch posted by Darren Schachter that is known to work:
https://lore.kernel.org/linux-iio/20200818153614.6438-1-dts86@cornell.edu/

OldCrank89, 2023-10-10:
Hi @pdp7  @RobertCNelson  @marcelnauta 

I have reviewed Darren Schachter's eCAP driver code and applied the same changes to my TI AM335x SDK. I have successfully generated the "pulse_tiecap.ko" module. However, when I attempt to test the eCAP module using sysfs (e.g., by navigating to "/sys/bus/iio/devices"), I can see the "trigger0" folder, but I am unable to enable it and view capture register values. Please understand that I am relatively new to Linux. Can anyone provide guidance on how to test Darren Schachter's driver code effectively?

[image]

Thanks in advance.