adafruit/adafruit-beaglebone-io-python #307: Compile error

pdp7, 2019-03-30:
Commit https://github.com/adafruit/adafruit-beaglebone-io-python/commit/0923b85884fd1a021a3ea92b3e30ab5a654e121b ignores new compiler warnings in gcc 8.2.0 but has the regression of causing error on gcc 6.3.0 which is used for the current BeagleBoard.org Debian Stretch images.
```
debian@beaglebone:~/adafruit-beaglebone-io-python$ sudo make install3
python3 setup.py build # --force
running build
running build_py
creating build
creating build/lib.linux-armv7l-3.5
copying Adafruit_I2C.py -> build/lib.linux-armv7l-3.5
creating build/lib.linux-armv7l-3.5/overlays
copying overlays/builder.py -> build/lib.linux-armv7l-3.5/overlays
copying overlays/__init__.py -> build/lib.linux-armv7l-3.5/overlays
creating build/lib.linux-armv7l-3.5/Adafruit_BBIO
copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-3.5/Adafruit_BBIO
copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-3.5/Adafruit_BBIO
copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-3.5/Adafruit_BBIO
running build_ext
building 'Adafruit_BBIO.GPIO' extension
creating build/temp.linux-armv7l-3.5
creating build/temp.linux-armv7l-3.5/source
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-QMv3zq/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.5m -c source/py_gpio.c -o build/temp.linux-armv7l-3.5/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-cast-function-type -Wno-error=format-truncation= -Wno-error=sizeof-pointer-memaccess
cc1: error: -Werror=format-truncation=: no option -Wformat-truncation=
cc1: error: unrecognized command line option ‘-Wno-cast-function-type’ [-Werror]
cc1: all warnings being treated as errors
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
Makefile:32: recipe for target 'build3' failed
make: *** [build3] Error 1
```

pdp7, 2019-03-30:
Changes in commit https://github.com/adafruit/adafruit-beaglebone-io-python/commit/0923b85884fd1a021a3ea92b3e30ab5a654e121b to ignore new compiler warnings in gcc 8.2.0 are reverted in commit https://github.com/adafruit/adafruit-beaglebone-io-python/commit/e06bec0d244a48265a3855e643167f8d1faae04a

TODO: determine how to avoid compile failure in gcc 8.2.0 (which is in the Debian 10.x "Buster" test image from @RobertCNelson)  without causing a regression with the current Debian 9.x "Stretch" images

pdp7, 2019-03-30:
Resolved in https://github.com/adafruit/adafruit-beaglebone-io-python/commit/e06bec0d244a48265a3855e643167f8d1faae04a

For gcc 8.2.0 in buster, see #308

closed 2019-03-30 by pdp7

webrouse, 2019-12-05:
@pdp7 

Hi, I have same problem here:
https://github.com/aircrack-ng/rtl8812au/blob/1af7edca86138533d04d7b7eb08a883d9ea4872a/os_dep/linux/os_intfs.c#L2804-L2810

```
/home/user/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-sunxi_cortexa53/rtl8812au-ct-custom-1af7edca/os_dep/linux/os_intfs.c:2805:2: error: 'wdev' undeclared (first use in this function); did you mean 'cdev'?
  wdev->iftype = NL80211_IFTYPE_MONITOR;
  ^~~~
  cdev
```