adafruit/adafruit-beaglebone-io-python #308: Compiler error in gcc

pdp7, 2019-03-30:
I was testing the new Debian "buster" image by @RobertCNelson and we noticed that the Adafruit_BBIO build was failing on compiler warnings.  This seems to be due to the introduction of new compiler warnings in the more recent version of gcc that is on the buster image: https://rcn-ee.net/rootfs/bb.org/testing/2019-03-03/buster-iot/bone-debian-buster-iot-armhf-2019-03-03-4gb.img.xz a shot

NOTE: commit https://github.com/adafruit/adafruit-beaglebone-io-python/commit/0923b85884fd1a021a3ea92b3e30ab5a654e121b caused a regression in older versions of gcc.  See issue #307 

TODO: determine how to avoid compile failure in gcc 8.2.0 without causing a regression for gcc 6.3.0 in the current Debian 9.x "Stretch" images

eliasbakken, 2019-07-12:
Here is a workaround to get it to compile on Buster. Please note it has not been tested, just that the compilation passes. 

CFLAGS="-Wno-cast-function-type -Wno-format-truncation -Wno-sizeof-pointer-memaccess -Wno-stringop-overflow" python setup.py install

pdp7, 2019-07-12:
@eliasbakken thanks, do you know if this would cause a regression in earlier versions of gcc where those options didn't exist yet?

eliasbakken, 2019-07-12:
Sorry, I'm not familiar with the compiler options on older GCCs...

pdp7, 2019-12-17:
@jlinford I have merged #321. Please check if this resolves the build errors that you encountered.  Thank you.

jlinford, 2020-01-05:
It's fixed!  Thanks!

closed 2020-01-05 by pdp7