adafruit/adafruit-beaglebone-io-python #309: Problems with UART on BBGW

hulkco, 2019-03-30:
Isue reference [#186](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/186)

hello everyone, I read  a lot notes on the Internet about how to enable UART ports in the Beaglebone, without success so far, I would appreciate any help

- Platform/operating system: Beaglebone Green wireless with Debian 9.5 2018-10-07 4GB SD LXQT
- Python 3.5.3

**Output of: sudo /opt/scripts/tools/version.sh**
```txt
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLTGW1ABBGW16056659]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
kernel:[4.14.71-ti-r80]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190320.1-0rcnee0~stretch+20190320]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.127925] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.139296] gpio-of-helper ocp:cape-universal: ready
END
```
**Contents of: /boot/uEnv.txt**
```txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0
uname_r=4.14.71-ti-r80
#uuid=
#dtb=
###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###
cmdline=coherent_pool=1M net.ifnames=0 quiet
#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e
#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs
##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
optargs=capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4
```
**Commands entered to config pin as UART.**
```txt
debian@beaglebone:/var/lib/cloud9$ config-pin P9_13 uart                                                              
debian@beaglebone:/var/lib/cloud9$ cat /sys/devices/platform/ocp/ocp:P9_13_pinmux/state
uart
```

**Source code program:**
```python
import Adafruit_BBIO.UART as UART
import serial

UART.setup("UART1")

ser = serial.Serial(port = "/dev/ttyS4", baudrate=9600)
ser.close()
ser.open()
if ser.isOpen():
    print ("Serial is open!")
ser.write("Hello UART!!!")

ser.close()
```

pdp7, 2019-03-30:
@hulkco thanks for providing all this information.  I believe the bootloader `U-Boot 2018.09-00002-g0b54a51eee` on your eMMC is recent enough, so the root cause of the issue you are experiencing is likely something else.

I will try to reproduce the issue.

__Have you only wired up P9.13 for UART4_TXD?__

There is also P9.11 for UART4_RXD:
http://beagleboard.org/static/images/cape-headers-serial.png

pdp7, 2019-04-03:
@hulkco apologies for the delay.  I've been focusing on the recent activity in Adafruit_Blinka around BeagleBone Black and PocketBeagle.  Is it ok with you for me to finish up that work before I try to reproduce and troubleshoot this issue?  thanks!

hulkco, 2019-04-03:
Hello @pdp7, of course I agree, I will be attentive to the changes and improvements you make to solve this, Regards

hulkco, 2019-04-04:
Hello, I already solved the issue, it was not software problem, it was one of the cables I was using, it was broken, I changed it and everything worked, Thanks for your help, I close the issue

closed 2019-04-04 by hulkco