adafruit/adafruit-beaglebone-io-python #311: PocketBeagle - naming/functions of P1_3

s-light, 2019-04-01:
i think there is a conflicting naming of the Pin P1_3 in [common.c#L192](https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/common.c#L192)
```
{ "USB1_VBUS_OUT", "P1_3", 0, -1, -1},
```
the comment above the table points to the source at bone.js - [there we can find #L2070](https://github.com/jadonk/bonescript/blob/master/src/bone.js#L2070):
```
{
        "name": "USB1_DRVVBUS",
        "gpio": 109,
        "key": "P1_3",
        "muxRegOffset": "0x134",
        "mux": "USB1_DRVVBUS",
        "eeprom": null,
        "options": [
            "USB1_DRVVBUS",
            "NA",
            "NA",
            "NA",
            "NA",
            "NA",
            "NA",
            "gpio3_13"
        ],
        "ball": {
            "ZCZ": "F15",
            "BSM": "M14"
        }
},
```

at [System Manual - Figure 42. Expansion Header Popular Functions - Color Coded](https://github.com/beagleboard/pocketbeagle/wiki/System-Reference-Manual#figure-42-expansion-header-popular-functions---color-coded)
we find P1_3 named as `USB1 V_EN` and also as `GPIO 109`
in [this google docs table](https://docs.google.com/spreadsheets/d/1FRGvYOyW1RiNSEVprvstfJAVeapnASgDXHtxeDOjgqw/edit#gid=0) (i don't know anymore where if found the link to this..) you find the Silkscreen as `USB1_EN` and the Mode0 as `USB1_DRVVBUS`.
silkscreen printing can be verified at [Figure 48. PocketBeagle Back BW](https://github.com/beagleboard/pocketbeagle/wiki/System-Reference-Manual#figure-48-pocketbeagle-back-bw)

So VBUS_OUT seems a little bit off to me...

pdp7, 2019-04-02:
@s-light thanks for raising this issue.  I will investigate further and ask @jadonk about that pin

pdp7, 2019-04-02:
note: I believe the gpio3_13 relates to GPIO 109 because gpiochip 3 is (3 * 32) base plus 13 which is 109

pdp7, 2019-04-02:
@RobertCNelson do you recall why P1_3 is "USB1_VBUS_OUT"?
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/5df30a8b735470e9b47f4f416a955043d85576a9/source/common.c#L192

@jadonk do you think USB1_DRVVBUS is the correct name?

pdp7, 2019-04-02:
I got clarification from @jadonk that he put EN on USB1_DRVVBUS.  USB1_DRVVBUS is the name from the datasheet.

I think it makes sense to change Adafruit_BBIO to mat

pdp7, 2019-04-02:
I got clarification from @jadonk that he put EN on USB1_DRVVBUS.  USB1_DRVVBUS is the name from the datasheet.

I think it makes sense to change Adafruit_BBIO to match bone.js and use USB1_DRVVBUS as the name

pdp7, 2019-04-02:
oops, hit close on accident

s-light, 2019-04-03:
thanks for your take on this! 
yeah it was mainly confusing me where that name came from....

so with b867bbf you have fixed this - so we can close this issue now?!

closed 2019-04-03 by pdp7