adafruit/adafruit-beaglebone-io-python #312: Adafruit_BBIO.GPIO not working on Graphical Desktop Version of OS

1stMonkeyInSpace, 2019-04-24:
I have looked through all of the similar issues that I can find, but I don't see anything that solves my problem.

I am using a Beaglebone Black with the latest Stretch OS images from the BB website. OS has been updated, kernel version 5.0.5, and I have run through the Adafruit_BBIO installation again. I am typing these commands directly into the terminal. 

In python 2.7.13 and python 3.5.3:
When I try to set the pinmode to in with "GPIO.setup("P8_10", GPIO.IN), I get "ValueError: Set gpio mode failed, missing file or invalid permissions."

From the terminal using config-pin:
$config-pin P8_10 gpio
ERROR: open() for /sys/devices/platform/ocp/ocp:P8_10_pinmux/state failed, No such file or directory
$config-pin -l P8_10
ERROR: open() for /sys/firmware/devicetree/base/ocp/P8_10_pinmux/pinctrl-named failed, No such file or directory.

When I check these directories, these files are not there in the Stretch LXQT  version of the OS, but they are there in the IoT version of the OS. What do I need to do to use the GPIO pins with the Graphical desktop OS?

Thanks in advance.

pdp7, 2019-04-25:
@1stMonkeyInSpace 
I think this is probably related to either the uboot bootloader version and/or the device tree overlays the are being loaded.

Please paste the output of:
```
sudo /opt/scripts/tools/version.sh 
```
and:
```
/boot/uEnv.txt
```

1stMonkeyInSpace, 2019-04-25:
Thanks pdp7.

Output of version.sh:

git:/opt/scripts/:[24f2e3495113a63c2dcc2c0bdc0d7660b54e4e8f]
eeprom:[A335BNLT000C1828BBBG2286]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g9d95a561c9]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
kernel:[5.0.5-bone7]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190227.0-0rcnee0~stretch+20190227]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190107.0-0rcnee0~stretch+20190108]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep remote
[    1.825231] remoteproc remoteproc0: wkup_m3 is available
[    2.347540] remoteproc remoteproc0: powering up wkup_m3
[    2.347559] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.351276] remoteproc remoteproc0: remote processor wkup_m3 is now up
[    7.084780] remoteproc remoteproc1: 4a334000.pru is available
[    7.100046] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[    6.968233] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[    7.084780] remoteproc remoteproc1: 4a334000.pru is available
[    7.084867] pru-rproc 4a334000.pru: PRU rproc node pru@34000 probed successfully
[    7.100046] remoteproc remoteproc2: 4a338000.pru is available
[    7.100114] pru-rproc 4a338000.pru: PRU rproc node pru@38000 probed successfully
dmesg | grep pinctrl-single
[    1.540099] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper

Contents of uEnv.txt:

#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=5.0.5-bone7
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

1stMonkeyInSpace, 2019-04-30:
@pdp7 

I didn't tag you in the reply, so I guess you missed it. Any ideas?

Thanks.

RobertCNelson, 2019-04-30:
```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2018.09-00002-g9d95a561c9]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.03-00001-gd12d09f]:[location: dd MBR]
```

u-boot on the eMMC is too old, 

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```
and reboot...

Regards,

1stMonkeyInSpace, 2019-05-06:
@RobertCNelson @pdp7 

Well, that command corrupted the Beaglebone and the SD card. 

I flashed the SD card with the latest image, then used that as a flasher image for the eMMC. Everything appears to be working now. Thanks for your help, and I'll mark this issue closed

closed 2019-05-06 by 1stMonkeyInSpace