adafruit/adafruit-beaglebone-io-python #314: Beaglebone Black PWM and input, both does not work: Tried everything available on the forums.

Akshaybj0221, 2019-09-10:
Hi guys,
@pdp7 @RobertCNelson 

I am really new to using BBB. Currently I am using the Adafruit BBIO library for making use of the PWM as well as Input functionality of it. In my case, neither the PWM works nor Input functionality. I have tried mostly all that I could find on the forums that seemed relevant, but nothing works.

**Scripts:**
PWM Test Script:
```
import Adafruit_BBIO.PWM as PWM
LED = "P9_14"
PWM.start(LED, 0, 1000)
PWM.stop(LED)
PWM.cleanup()
```

INPUT test script:
```
import Adafruit_BBIO.GPIO as GPIO
from time import sleep

topButton = "P9_11"
bottomButton = "P9_13"

GPIO.setup(topButton, GPIO.IN)
GPIO.setup(bottomButton, GPIO.IN)

while(1):
    if GPIO.input(topButton):
        print "Top button is ON"
    else:
        print "Top button OFF"

    if GPIO.input(bottomButton):
        print "Bottom button in ON"
    else:
        print "Bottom button in OFF"

    if GPIO.input(topButton) and GPIO.input(bottomButton):
        break

    sleep(0.2)

print "Good bye"
GPIO.cleanup()
```

**Errors:**

For Input test script, I get the following error:
```
Traceback (most recent call last):
  File "7_inputButtonPush.py", line 7, in <module>
    GPIO.setup(topButton, GPIO.IN)
ValueError: Set gpio direction failed, missing file or invalid permissions.

```

And for PWM test script, I get the following error:
```
debian@beaglebone:~/python_scripts$ python 5_testPWM.py 
Traceback (most recent call last):
  File "5_testPWM.py", line 7, in <module>
    PWM.start(greenLED, 0, 1000)
RuntimeError: Unknown error
```

**Additional Data:**
**1.**
```
sudo /opt/scripts/tools/version.sh
[sudo] password for debian: 
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[A335BNLT00C02016BBBK2BE6]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
kernel:[5.2.13-bone9]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.409666] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.419656] gpio-of-helper ocp:cape-universal: ready
END
```

**2.**
This gives me same error:
```
$ dmesg | grep bone
-bash: $: command not found
```

**3.**
So I used `cat /etc/dogtag` as suggested by @MajdBy. I get the output:
`BeagleBoard.org Debian Image 2018-10-07
`
Seems like I have the latest BBB version and still I am seeing this problem.

**4.** 
I have also tried (Did not work)
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=100
sudo reboot
```

**5.**
Also tried updating bootloader (Did update successfully)
```
sudo /opt/scripts/tools/update_kernel.sh --bone-channel --testing
sudo /opt/scripts/tools/developers/update_bootloader.sh 
sudo reboot
```

I have already taken the backup to all the data and would really appreciate it if anyone could help me.

Also I am sorry if this is the wrong thread. Please point me if I should carry on the conversation somewhere else.

Thanks!

RobertCNelson, 2019-09-10:
@Akshaybj0221 , i'd really upgrade this package:

```
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
```

Do:

```
sudo apt update ; sudo apt install --only-upgrade bb-cape-overlays
```

```
ValueError: Set gpio direction failed, missing file or invalid permissions.
```
This might fix that persmission issue:

```
sudo apt install bb-customizations
```

ps, that image is soo old, you should look at the latest monthy snapshot:

https://rcn-ee.net/rootfs/bb.org/testing/2019-09-08/stretch-iot/bone-debian-9.10-iot-armhf-2019-09-08-4gb.img.xz

or:

https://rcn-ee.net/rootfs/bb.org/testing/2019-09-08/stretch-lxqt/bone-debian-9.10-lxqt-armhf-2019-09-08-4gb.img.xz

as it should fix a lot of the current issues..

Regards,

Akshaybj0221, 2019-09-10:
1. @RobertCNelson Thanks a lot for the quick response. But unfortunately none of them worked.
I ran  `sudo apt update ; sudo apt install --only-upgrade bb-cape-overlays`
I got this error:

Output:
```
debian@beaglebone:~$ sudo apt update ; sudo apt install --only-upgrade bb-cape-overlays
[sudo] password for debian: 
Ign:1 http://deb.debian.org/debian stretch InRelease
Get:2 http://deb.debian.org/debian stretch-updates InRelease [91.0 kB]
Hit:3 http://repos.rcn-ee.com/debian stretch InRelease          
Hit:4 http://deb.debian.org/debian-security stretch/updates InRelease
Hit:5 http://deb.debian.org/debian stretch Release
Fetched 91.0 kB in 42s (2,138 B/s)                                                                                                                                                       
Reading package lists... Done
Building dependency tree       
Reading state information... Done
192 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be upgraded:
  bb-cape-overlays
1 upgraded, 0 newly installed, 0 to remove and 191 not upgraded.
Need to get 98.4 kB of archives.
After this operation, 116 kB of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf bb-cape-overlays armhf 4.4.20190812.0-0rcnee0~stretch+20190812 [98.4 kB]
Fetched 98.4 kB in 0s (246 kB/s)      
(Reading database ... 76554 files and directories currently installed.)
Preparing to unpack .../bb-cape-overlays_4.4.20190812.0-0rcnee0~stretch+20190812_armhf.deb ...
Unpacking bb-cape-overlays (4.4.20190812.0-0rcnee0~stretch+20190812) over (4.4.20180928.0-0rcnee0~stretch+20180928) ...
Setting up bb-cape-overlays (4.4.20190812.0-0rcnee0~stretch+20190812) ...
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130) ...
update-initramfs: Generating /boot/initrd.img-5.2.13-bone9
WARNING: Setting CRYPTSETUP in /etc/initramfs-tools/initramfs.conf is deprecated and will stop working in the future. Use /etc/cryptsetup-initramfs/conf-hook instead.
cryptsetup: WARNING: failed to detect canonical device of /dev/mmcblk0p1
cryptsetup: WARNING: could not determine root device from /etc/fstab
```

2.
On running `sudo apt install bb-customizations`
I got the same error:
```
debian@beaglebone:~$ sudo apt install bb-customizations
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be upgraded:
  bb-customizations
1 upgraded, 0 newly installed, 0 to remove and 190 not upgraded.
Need to get 5,694 B of archives.
After this operation, 6,144 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian stretch/main armhf bb-customizations all 1.20190730.0-0rcnee0~stretch+20190730 [5,694 B]
Fetched 5,694 B in 0s (24.2 kB/s)      
(Reading database ... 76584 files and directories currently installed.)
Preparing to unpack .../bb-customizations_1.20190730.0-0rcnee0~stretch+20190730_all.deb ...
Unpacking bb-customizations (1.20190730.0-0rcnee0~stretch+20190730) over (1.20180815-0rcnee0~stretch+20180815) ...
Setting up bb-customizations (1.20190730.0-0rcnee0~stretch+20190730) ...
Installing new version of config file /etc/udev/rules.d/80-gpio-noroot.rules ...
Installing new version of config file /etc/udev/rules.d/81-pwm-noroot.rules ...
Installing new version of config file /etc/udev/rules.d/82-gpio-config-pin.rules ...
Installing new version of config file /etc/udev/rules.d/83-eqep-noroot.rules ...
groupadd: group 'gpio' already exists
groupadd: group 'pwm' already exists
groupadd: group 'eqep' already exists
update-initramfs: deferring update (trigger activated)
Processing triggers for initramfs-tools (0.130) ...
update-initramfs: Generating /boot/initrd.img-5.2.13-bone9
WARNING: Setting CRYPTSETUP in /etc/initramfs-tools/initramfs.conf is deprecated and will stop working in the future. Use /etc/cryptsetup-initramfs/conf-hook instead.
cryptsetup: WARNING: failed to detect canonical device of /dev/mmcblk0p1
cryptsetup: WARNING: could not determine root device from /etc/fstab
```

Would you want me to try something else, or would you just suggest me to download the latest images that you sent me?

Akshaybj0221, 2019-09-11:
@RobertCNelson One new update. I rebooted the BBB after running the above 2 commands, where I got the output as mentioned above. And it seems like now the 'Input' script works!

Now only one thing left to fix, the PWM issue. Any suggestions on that?

Also I would really appreciate it if you could give some insight or an explanation as to what was actually going on and how it got fixed? I reckon that you deduced that the 'bb-cape-overlays' package was old and we just updated it. Can I know in the future, how can I know if a package is latest or not?

Also as you suggested to use the monthly snapshots of the image, is there a way I can just update my OS without having to burn an SD card again. Basically like we update the OS of a computer, where all the data is intact, and just the packages gets updated?

Thanks!

AkshayMatternet, 2019-09-13:
Want to update this issue. Used the latest OS image (IOT), the one suggested by @RobertCNelson . Everything works fine on this one. The link to the image is here:
https://rcn-ee.net/rootfs/bb.org/testing/2019-09-08/stretch-iot/bone-debian-9.10-iot-armhf-2019-09-08-4gb.img.xz

hstarmans, 2019-10-02:
I can confirm this, also make sure you try this with the universal cape activated in /boot/uEnv.txt.
I had problems with my custom cape, which worked with everything except the python module.

pdp7, 2019-10-02:
@Akshaybj0221 ok to close this issue now or is there still a problem?  thanks

AkshayMatternet, 2019-10-02:
Yes. Ok to close this issue. Thanks! :)

closed 2019-10-03 by pdp7