adafruit/adafruit-beaglebone-io-python #315: running config-pin 9.91 out before flipping pin 9.41 breaks library

hstarmans, 2019-10-08:
The code works but the behavior is strange.
If just flip the pin 9.41 with the python library ; it works.
If I run the following, before flipping pin 9.41 with the python library.
```
config-pin 9.91 out
````
This will break the python library. It is unable to flip it back to in and the pin voltage is still changed but now it varies between 0 and 1.4V. To fix it I apply ```config-pin 9.91 in``` before running the code.
I think the library itself should fix this somehow.

pdp7, 2019-10-09:
thanks for reporting this issue.  P9.91 and P9.41 are both the same physical pins so I think that might be why this issue is occurring.  I will try to reproduce this on a BeagleBone and investigate further.

hstarmans, 2019-10-13:
pretty sure it also happens to other pins also had it with pin 9.12 so don't think it is related to P9.91 or P9.41. 
What also might be nice is the possibility to write other configurations, e.g. pruout.
```python
def sysfs_write( path, data ):
    if isinstance( data, str ):
        data = data.encode()
    with open( path, 'wb', buffering=0 ) as f:
        f.write( data )

def config_pin( pin, mode ):
    sysfs_write( '/sys/bus/platform/devices/ocp:'+pin+'_pinmux/state', mode )

config_pin( 'P8_11', 'pruout' )
```
Got this code from [mvduijn](https://github.com/mvduin/py-uio/issues/4#issuecomment-541319930)

pdp7, 2019-12-02:
@hstarmans Adafruit_BBIO does set the pinmux state in a similar way as the example you provided:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pinmux.c#L53

I noticed in your original post:
```
config-pin 9.91 out
```
but `out` is not a valid pinmux state.  `config-pin -l p9.91` will show you the valid states.

pdp7, 2019-12-10:
@hstarmans is this still an issue?

hstarmans, 2019-12-11:
no this is no longer an issue..

closed 2019-12-11 by hstarmans