adafruit/adafruit-beaglebone-io-python #316: config-pin overlay BB-UART1 throws Segmentation fault Error

meberli, 2019-10-17:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

RobertCNelson, 2019-10-17:
@meberli loading overlays from linux is not support..

```
root@beaglebone:~# config-pin overlay BB-UART1
Loading BB-UART1 overlay
```
all overlays need to be added thru u-boot:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays

Regards,

meberli, 2019-10-18:
@RobertCNelson thanks. 
I now added BB-UART1 in uEnv.txt like this:
`cape_enable=bone_capemgr.enable_partno=BB-UART1`
on startup everything seems to be ok:
```
Oct 18 10:03:44 beaglebone kernel: [    1.500855] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,00C0,2017BBBK0C20'
Oct 18 10:03:44 beaglebone kernel: [    1.500885] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
Oct 18 10:03:44 beaglebone kernel: [    1.528519] bone_capemgr bone_capemgr: slot #0: No cape found
Oct 18 10:03:44 beaglebone kernel: [    1.557068] bone_capemgr bone_capemgr: slot #1: No cape found
Oct 18 10:03:44 beaglebone kernel: [    1.583967] bone_capemgr bone_capemgr: slot #2: No cape found
Oct 18 10:03:44 beaglebone kernel: [    1.612514] bone_capemgr bone_capemgr: slot #3: No cape found
Oct 18 10:03:44 beaglebone kernel: [    1.612534] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-UART1' VER 'N/A' PR '0'
Oct 18 10:03:44 beaglebone kernel: [    1.612571] bone_capemgr bone_capemgr: slot #4: override
Oct 18 10:03:44 beaglebone kernel: [    1.612585] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
Oct 18 10:03:44 beaglebone kernel: [    1.612598] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-UART1'
Oct 18 10:03:44 beaglebone kernel: [    1.612681] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-UART2' VER 'N/A' PR '0'
Oct 18 10:03:44 beaglebone kernel: [    1.612691] bone_capemgr bone_capemgr: slot #5: override
Oct 18 10:03:44 beaglebone kernel: [    1.612701] bone_capemgr bone_capemgr: Using override eeprom data at slot 5
Oct 18 10:03:44 beaglebone kernel: [    1.612712] bone_capemgr bone_capemgr: slot #5: 'Override Board Name,00A0,Override Manuf,BB-UART2'
Oct 18 10:03:44 beaglebone kernel: [    1.613178] bone_capemgr bone_capemgr: initialized OK.
```
but I still do not have the devices present:
```
crw--w---- 1 root tty     243,  0 Oct 18 10:22 /dev/ttyGS0
lrwxrwxrwx 1 root root          5 Oct 18 10:03 /dev/ttyO0 -> ttyS0
crw--w---- 1 root tty       4, 64 Oct 18 10:22 /dev/ttyS0
crw-rw---- 1 root dialout   4, 65 Oct 18 10:03 /dev/ttyS1
crw-rw---- 1 root dialout   4, 66 Oct 18 10:03 /dev/ttyS2
crw-rw---- 1 root dialout   4, 67 Oct 18 10:03 /dev/ttyS3
crw-rw---- 1 root dialout   4, 68 Oct 18 10:03 /dev/ttyS4
crw-rw---- 1 root dialout   4, 69 Oct 18 10:03 /dev/ttyS5
```
Am I missing something?

RobertCNelson, 2019-10-18:
@meberli , sorry that's still kernel overlays, let's start from scratch to see where you are and what we need to upgrade to get things working..

Please run: (and share the output of)

```
sudo /opt/scripts/tools/version.sh
```

Regards,

pdp7, 2019-10-25:
@meberli are you still having this issue?

meberli, 2019-10-25:
sorry I was away this week ..
@pdp7  yes still the same issue.
@RobertCNelson output is:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[b4e40ed94bc1e15bb291d10af48e0e9893864197]
eeprom:[A335BNLT00C02017BBBK0C20]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gf15b99f                                                                             0b6]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]:[loca                                                                             tion: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-                                                                             00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev us                                                                             ers systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisd                                                                             k weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/m                                                                             mcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.def                                                                             ault_quality=100 quiet]
dmesg | grep remote
[    1.260787] remoteproc remoteproc0: wkup_m3 is available
[    1.481038] remoteproc remoteproc0: powering up wkup_m3
[    1.481165] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf,                                                                              size 217168
[    1.485430] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   10.043417] remoteproc remoteproc1: 4a334000.pru is available
[   10.045340] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   10.014364] pruss 4a300000.pruss: creating PRU cores and other child platform                                                                              devices
[   10.043417] remoteproc remoteproc1: 4a334000.pru is available
[   10.043544] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a32600                                                                             4/pruss@0/pru@34000 probed successfully
[   10.045340] remoteproc remoteproc2: 4a338000.pru is available
[   10.045445] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a32600                                                                             4/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.944703] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.956996] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

pdp7, 2019-10-25:
I think the issue is that there is an old u-boot on the eMMC;
> bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2016.01-00001-g4eb802e]:[loca                                                                             tion: dd MBR]
@RobertCNelson what is the command to clear the old u-boot from the emmc?

silver2row, 2019-12-01:
Hello,

Here: 

    sudo blkdiscard /dev/mmcblk1

This will erase your eMMC. Please back-up all info/data you want to keep. This will erase the entire contents of your eMMC.

Seth

pdp7, 2019-12-02:
@meberli please try this to delete the u-boot bootloader on the eMMC (you willl need to be booting from SD card):
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

RobertCNelson, 2019-12-03:
@meberli "cape_enable=bone_capemgr.enable_partno=BB-UART1" is not supported:

You must use the uboot_overlay_addrX= variables in /boot/uEnv.txt
```
uboot_overlay_addr0=/lib/firmware/<file0>.dtbo
```
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Override_external_capes

Regards,

pdp7, 2019-12-03:
@meberli upon reviewing with @RobertCNelson today, we believe the issue is that the overlay is specificied incorrectly in uEnv.txt

Please look at:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Override_external_capes

For example:
```
uboot_overlay_addr0=/lib/firmware/BB-UART1.dtbo
```

meberli, 2019-12-04:
After restarting with a new board and new installation, the issue was gone...

@RobertCNelson Thanks for the hint,  I will test this with my old setup.

pdp7, 2019-12-04:
@meberli good to hear!  and thanks to @RobertCNelson for always being a fountain of knowledge :)

pdp7, 2019-12-17:
@meberli please re-open if still an issue

closed 2019-12-17 by pdp7