adafruit/adafruit-beaglebone-io-python #317: Add support for BeagleBone AI with Adafruit BBIO.GPIO

jomoengineer, 2019-10-25:
The new BeagleBone AI has a AM57x processor and a different gpio mapping than the previous BeagleBone boards, thus the BBIO GPIO Python libs do not work properly on the BB AI.

Ex:
On the BB AI P9_15 is mapped to gpio76
where
On the BB Black, P8_39 is mapped to gpio76

Also, although not really a Adafruit issue, the config-pin tool does not work with the BB AI:
<pre><code>
$ config-pin -q P9_15
P9_15 pinmux file not found!
Pin has no cape: P9_15
</pre></code>

pdp7, 2019-12-17:
@MarkAYoder @jadonk Adafruit_BBIO still needs support for BeagleBone AI.  This might be a good task for a student (if you know of anyone, Mark, or once GSoC starts).

MalavPatel3501, 2020-06-03:
Here i create my custom .dts file and test all interfaces .if you facing issue regarding any interfaces then please follow below link :

https://github.com/MalavPatel3501/BeagleBone-AI--mpBBAI-IO-Python-Library

In this project you got .dts file and all pin muxing steps. Moreover i puts some example code in this project.

simongtan, 2020-09-27:
BBAI compatibility would be really useful - thanks guys!

pnallin, 2021-07-01:
@pdp7 any "release date" for BB-AI? Or, is there anyone working on it?

pdp7, 2021-07-02:
@pnallin unfortunately there is not active work on it.  However, I think Adafruit Blinka is a good alternative to this library (Adafruit_BBIO): https://github.com/adafruit/Adafruit_Blinka

There is support for the AI board by @makermelissa 
https://github.com/adafruit/Adafruit_Blinka/commit/8273fc647a35d3e8f6d11e75f6d9ea4223445ac6