adafruit/adafruit-beaglebone-io-python #318: ValueError: Set gpio mode failed, missing file or invalid permissions.

leo19980219, 2019-11-12:
Hi,
I'm a beginner of Beaglebone. I try to set GPIO as input. But always get an error.
**ValueError: Set gpio mode failed, missing file or invalid permissions.**
How can I solved the problem?
Here is my python code.
```
import Adafruit_BBIO.GPIO as GPIO
import time
 
GPIO.setup("P9_12", GPIO.IN)
 
while True:
    status = GPIO.input("P9_12")
    print(status)
    time.sleep(1)
```
I think the following message is needed for debugging.
```
$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[c4140f1eca1c862e0652338a391b3e610a82ed48]
eeprom:[A335BNLT00C02917BBBK3C44]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-01-28]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
UBOOT: Loaded Overlay:[BB-UART1-00A0]
UBOOT: Loaded Overlay:[BB-UART2-00A0]
UBOOT: Loaded Overlay:[BB-UART4-00A0]
UBOOT: Loaded Overlay:[BB-UART5-00A0]
kernel:[4.9.78-ti-r94]
nodejs:[v6.12.3]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20191110.0-0rcnee0~stretch+20191110]
pkg:[bb-wl18xx-firmware]:[1.20170829-0rcnee2~stretch+20180104]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[roboticscape]:[0.3.4-git20170602-0rcnee4~stretch+20171108]:[GOT_REPLACED_BY_NEXT]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee0~stretch+20170830]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4,BB-UART5 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep remote
[    1.648925] remoteproc remoteproc0: wkup_m3 is available
[    1.925763] remoteproc remoteproc0: powering up wkup_m3
[    1.925787] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[    1.926042] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   34.889107] remoteproc remoteproc1: 4a334000.pru0 is available
[   34.966313] remoteproc remoteproc2: 4a338000.pru1 is available
dmesg | grep pru
[   34.740106] ti-pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   34.889107] remoteproc remoteproc1: 4a334000.pru0 is available
[   34.889231] pru-rproc 4a334000.pru0: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a334000 probed successfully
[   34.966313] remoteproc remoteproc2: 4a338000.pru1 is available
[   34.966431] pru-rproc 4a338000.pru1: PRU rproc node /ocp/pruss_soc_bus@4a326000/pruss@4a300000/pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.416313] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    2.145445] pinctrl-single 44e10800.pinmux: pin PIN48 already requested by 0-0070; cannot claim for 481aa000.serial
[    2.167972] pinctrl-single 44e10800.pinmux: pin-48 (481aa000.serial) status -22
[    2.181573] pinctrl-single 44e10800.pinmux: could not request pin 48 (PIN48) from group pinmux_bb_uart5_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    1.417337] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```
```
$ dmesg | grep bone
[    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4,BB-UART5 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet
[    1.931769] bone_capemgr bone_capemgr: Baseboard: 'A335BNLT,00C0,2917BBBK3C44'
[    1.931798] bone_capemgr bone_capemgr: compatible-baseboard=ti,beaglebone-black - #slots=4
[    1.960202] bone_capemgr bone_capemgr: slot #0: No cape found
[    1.991252] bone_capemgr bone_capemgr: slot #1: No cape found
[    2.023957] bone_capemgr bone_capemgr: slot #2: No cape found
[    2.056270] bone_capemgr bone_capemgr: slot #3: No cape found
[    2.062090] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-UART1' VER 'N/A' PR '0'
[    2.062100] bone_capemgr bone_capemgr: slot #4: override
[    2.067466] bone_capemgr bone_capemgr: Using override eeprom data at slot 4
[    2.067480] bone_capemgr bone_capemgr: slot #4: 'Override Board Name,00A0,Override Manuf,BB-UART1'
[    2.076615] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-UART2' VER 'N/A' PR '0'
[    2.076625] bone_capemgr bone_capemgr: slot #5: override
[    2.081991] bone_capemgr bone_capemgr: Using override eeprom data at slot 5
[    2.082005] bone_capemgr bone_capemgr: slot #5: 'Override Board Name,00A0,Override Manuf,BB-UART2'
[    2.091152] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-UART4' VER 'N/A' PR '0'
[    2.091167] bone_capemgr bone_capemgr: slot #6: override
[    2.096532] bone_capemgr bone_capemgr: Using override eeprom data at slot 6
[    2.096544] bone_capemgr bone_capemgr: slot #6: 'Override Board Name,00A0,Override Manuf,BB-UART4'
[    2.105658] bone_capemgr bone_capemgr: enabled_partno PARTNO 'BB-UART5' VER 'N/A' PR '0'
[    2.105668] bone_capemgr bone_capemgr: slot #7: override
[    2.111031] bone_capemgr bone_capemgr: Using override eeprom data at slot 7
[    2.111043] bone_capemgr bone_capemgr: slot #7: 'Override Board Name,00A0,Override Manuf,BB-UART5'
[    2.120776] bone_capemgr bone_capemgr: initialized OK.
[    2.132987] bone_capemgr bone_capemgr: slot #5: dtbo 'BB-UART2-00A0.dtbo' loaded; overlay id #1
[    2.138549] bone_capemgr bone_capemgr: slot #4: dtbo 'BB-UART1-00A0.dtbo' loaded; overlay id #0
[    2.138795] bone_capemgr bone_capemgr: slot #6: dtbo 'BB-UART4-00A0.dtbo' loaded; overlay id #2
[    2.220589] bone_capemgr bone_capemgr: slot #7: dtbo 'BB-UART5-00A0.dtbo' loaded; overlay id #3
[    6.314465] systemd[1]: Set hostname to <beaglebone>.

```

pdp7, 2019-11-12:
@RobertCNelson is this too old version of u-boot on eMMC?
> bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]

pdp7, 2019-11-12:
@leo19980219 what happens when you run the Python script as root (using sudo)?

leo19980219, 2019-11-12:
@pdp7 Running script in root mode also show the same error message.

leo19980219, 2019-11-16:
@pdp7 When I run
```
config-pin P8.03 gpio
```
It shows 
P8_01 pinmux file not found!
Pin has no cape: P8_03

Is that related to this issue? It means I should update my bootloader?

pdp7, 2019-12-02:
@leo19980219 to be safe, let's delete the u-boot bootloader on the eMMC since it may be too old (2018) and is causing an issue.

@RobertCNelson do you have that command to zero out the bootloader on eMMC?

pdp7, 2019-12-02:
@leo19980219 to be safe, let's delete the u-boot bootloader on the eMMC since it may be too old (2018) and is causing an issue.  You willl need to be booting from SD card:
```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```
reboot and then run version.sh again to see the version of u-boot that is listed.

pdp7, 2019-12-03:
@leo19980219 could you please post the contents of uEnv.txt?

I believe that you need to add:
```
enable_uboot_overlays=1
enable_uboot_cape_universal=1
disable_uboot_overlay_video=1
```
and remove:
```
bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.enable_partno=BB-UART1,BB-UART2,BB-UART4,BB-UART5
```

pdp7, 2019-12-03:
@leo19980219 I talked to @RobertCNelson and the u-boot bootloader on the eMMC since is too old (2018) and is causing an issue.

You should boot from eMMC and upgrade the eMMC bootloader with:
```
sudo /opt/scripts/tools/developers/update_bootloader.sh
```
reboot and then run version.sh again to see the version of u-boot that is listed.  It should be version `2019.04`.

pdp7, 2019-12-17:
@leo19980219 I am going to close but please re-open if still issue

leo19980219, 2019-12-23:
Sorry for replying late.
I update the kernel and boot with SD card.
But still got the same issue.
[image]

Here is the information
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLT00C02917BBBK3C44]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 quiet capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.110131] remoteproc remoteproc0: wkup_m3 is available
[    1.188386] remoteproc remoteproc0: powering up wkup_m3
[    1.188501] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.192445] remoteproc remoteproc0: remote processor wkup_m3 is now up
[    6.262932] remoteproc remoteproc1: 4a334000.pru is available
[    6.269730] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[    6.214125] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[    6.262932] remoteproc remoteproc1: 4a334000.pru is available
[    6.263983] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[    6.269730] remoteproc remoteproc2: 4a338000.pru is available
[    6.269852] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.865722] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.866938] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```
The content of /boot/uEnv.txt
```
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.108-ti-r113
#uuid=
#dtb=
optargs=quiet capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN
###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo
uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```

leo19980219, 2019-12-23:
I accidentally fixed  this by booting with pressing button "S2".
Maybe it related to the bootloader in eMMC ?

RobertCNelson, 2019-12-23:
@leo19980219 correct the version of u-boot is way too old on the eMMC, which breaks everything..

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```

leo19980219, 2019-12-24:
Thanks for replying.
After update u-boot it works fine.

pdp7, 2019-12-24:
@leo19980219 did clearing the old u-boot on eMMC resolve the issue?

leo19980219, 2019-12-25:
> @leo19980219 did clearing the old u-boot on eMMC resolve the issue?

@pdp7 Yes!

closed 2019-12-25 by pdp7