adafruit/adafruit-beaglebone-io-python #319: Warnings makes the project crash when installing on Debian Buster on BBB

JeffLabonte, 2019-11-14:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **debian buster**

- Python version (run `python -version` or `python3 -version`):  **python 3.7.5**

- Error message you are receiving, including any Python exception traces:
The error is the same for a ARM device, in my case a BBB.
  *** 
    x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c source/py_gpio.c -o build/temp.linux-x86_64-3.7/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
    source/py_gpio.c:558:14: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
      558 |    {"setup", (PyCFunction)py_setup_channel, METH_VARARGS | METH_KEYWORDS, "Set up the GPIO channel, direction and (optional) pull/up down control\nchannel        - Either: RPi board pin number (not BCM GPIO 00..nn number).  Pins start from 1\n                 or    : BCM GPIO number\ndirection      - INPUT or OUTPUT\n[pull_up_down] - PUD_OFF (default), PUD_UP or PUD_DOWN\n[initial]      - Initial value for an output channel\n[delay]        - Time in milliseconds to wait after exporting gpio pin"},
          |              ^
    source/py_gpio.c:562:25: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
      562 |    {"add_event_detect", (PyCFunction)py_add_event_detect, METH_VARARGS | METH_KEYWORDS, "Enable edge detection events for a particular GPIO channel.\nchannel      - either board pin number or BCM number depending on which mode is set.\nedge         - RISING, FALLING or BOTH\n[callback]   - A callback function for the event (optional)\n[bouncetime] - Switch bounce timeout in ms for callback"},
          |                         ^
    source/py_gpio.c:565:27: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
      565 |    {"add_event_callback", (PyCFunction)py_add_event_callback, METH_VARARGS | METH_KEYWORDS, "Add a callback for an event already defined using add_event_detect()\ngpio         - gpio channel\ncallback     - a callback function\n[bouncetime] - Switch bounce timeout in ms"},
          |                           ^
    cc1: all warnings being treated as errors

***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **apt install build-essential python3-dev python3-smbus python3-pip && pip3 install Adafruit_BBIO**

silver2row, 2019-11-24:
Hello,

I am using Buster too. I have not been able to install the Adafruit-BBIO library just yet. I will return service when I figure out how.

Seth

silver2row, 2019-11-25:
Hello Again,

See. This is my issue. It was on the machine, my BBBW, and then it just went away "magically." I am using Buster 4.19.x. I am backing up my files and directories now. I am starting w/ a fresh image.

...

It may be easier to restart than to fix a broken install but I just wanted to relay this info. in case you were thinking of still fixing the issue.

Seth

P.S. I have to bow out for now. I am sorry.

pdp7, 2019-12-02:
I believe it is related to GCC flags:
Compiler error in gcc 8.2.0 on Debian 10.x (buster)
https://github.com/adafruit/adafruit-beaglebone-io-python/issues/308

JeffLabonte, 2019-12-05:
I was able to compile the project when I removed the `-Werror`. Obviously, it isn't the way that I should operate and I am not sure if the library is going to work as expected ( It should, those are only warnings ).

pdp7, 2019-12-11:
Refer to PR https://github.com/adafruit/adafruit-beaglebone-io-python/pull/321

pdp7, 2019-12-17:
I am going to close this in favor of tracking this problem under issue #308 

I believe I have a fix in PR #321

closed 2019-12-17 by pdp7