adafruit/adafruit-beaglebone-io-python #320: runtimeerror: problem with a sysfs file

RadhaKrishnanme, 2019-11-15:
Hi,
I encounter a problem resulting in "runtimeerror: problem with a sysfs file" when I try to set P9_16 PWM frequency other than 2000Hz. It works fine when the frequency is 2000. 
I read various github pages and implemented it. None of it sorted this issue.
The uEnv.text output is pasted below:
`
debian@beaglebone:~$ cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.19.50-bone35
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=/lib/firmware/BB-BONE-4D4C-01-00A1.dtbo
#uboot_overlay_addr1=/lib/firmware/<file1>.dtbo
#uboot_overlay_addr2=/lib/firmware/<file2>.dtbo
#uboot_overlay_addr3=/lib/firmware/<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/<file4>.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###Override capes with eeprom
#uboot_overlay_addr4=/lib/firmware/BB-SPIDEV1-00A0.dtbo
uboot_overlay_addr4=/lib/firmware/PSS_PRU.dtbo
#uboot_overlay_addr5=/lib/firmware/BB-BONE-4D4C-01-00A1.dtbo
uboot_overlay_addr5=/lib/firmware/PSS_LCD.dtbo
uboot_overlay_addr6=/lib/firmware/BB-UART4-00A0.dtbo
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

#these lines are to enable SPI1 and disable SPI0 and HDMI
#cape_disable=bone_capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN
cape_disable=bone_capemgr.disable_partno=BB-HDMI-TDA998x
cape_disable=bone_capemgr.disable_partno=BB-SPIDEV0
#cape_enable=bone_capemgr.enable_partno=BB-SPIDEV1
``
The verison is:
`
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
eeprom:[▒▒▒▒*]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2018-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-dirty]:[location: dd MBR]
kernel:[4.19.50-bone35]
nodejs:[v6.14.4]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/PSS_PRU.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/PSS_LCD.dtbo]
uboot_overlay_options:[uboot_overlay_addr6=/lib/firmware/BB-UART4-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20180928.0-0rcnee0~stretch+20180928]
pkg:[bb-wl18xx-firmware]:[1.20180517-0rcnee0~stretch+20180517]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.3-git20181005.0-0rcnee0~stretch+20181005]
pkg:[firmware-ti-connectivity]:[20170823-1rcnee1~stretch+20180328]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.disable_partno=BB-SPIDEV0 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    0.524128] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    0.838419] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by ocp:helper; cannot claim for ocp:P9_30_pinmux
[    0.849506] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    0.857042] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_default_pin  on device pinctrl-single
[    0.880108] pinctrl-single 44e10800.pinmux: pin PIN100 already requested by ocp:helper; cannot claim for ocp:P9_31_pinmux
[    0.891150] pinctrl-single 44e10800.pinmux: pin-100 (ocp:P9_31_pinmux) status -22
[    0.898689] pinctrl-single 44e10800.pinmux: could not request pin 100 (PIN100) from group pinmux_P9_31_default_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.533833] gpio-of-helper ocp:cape-universal: ready
END
`
Error in dmesg:
`
[ 8841.070859] ehrpwm 48302200.pwm: period value conflicts with channel 0
`

the state of the P9_16 pin is found to be PWM but I cannot change the frequency other than 2000 Hz. 
`/sys/devices/platform/ocp/ocp:P9_16_pinmux/state: pwm`

Kindly help me in this regard.

pdp7, 2019-12-01:
I see that the image you have is from October 2018.

Please try the latest Debian image on BeagleBoard.org (http://beagleboard.org/latest-images):
Debian 9.9 2019-08-03 4GB SD IoT

RobertCNelson, 2019-12-01:
@RadhaKrishnanme , i see a small problem.. Do you seem to have issues booting some of our images?  Your eeprom from the script looks corrupt..  To be safe, if you take a picture of the board and add that to this issue, i can walk you thru the correct eeprom rewrite..

```
git:/opt/scripts/:[1aa73453b2c980b75e31e83dab7dd8b6696f10c7]
**eeprom:[▒▒▒▒*]**
model:[TI_AM335x_BeagleBone_Black]
```
Regards,

pdp7, 2019-12-03:
@RadhaKrishnanme another option would be if you have a serial cable, to monitor the output of u-boot at boot and take a screenshot.

pdp7, 2019-12-10:
@RadhaKrishnanme are you still having this issue?

RadhaKrishnanme, 2019-12-11:
@pdp7 Sorry for the delayed reply. I will take a screenshot of the output of u-boot and will paste it. Thanks.

RadhaKrishnanme, 2019-12-11:
@pdp7 Attached are the screenshots for your reference. Kindly let me know the procedure to overcome theissue.
[image]
[image]

pdp7, 2019-12-11:
@RadhaKrishnanme thanks, could you also take a screenshot of the output of:
`udo /opt/scripts/tools/version.sh`

@RobertCNelson is there any other information that would be useful to troubleshoot this issue?  thanks

RadhaKrishnanme, 2019-12-12:
@pdp7 Attached is the screenshot as instructed.
[image]

RobertCNelson, 2019-12-12:
Just re-flash with a new image, you have a lot of random changes thru out the boot process from /boot/uEnv.txt custom overlays (PSS_PRU.dtbo/PSS_LCD.dtbo files..).. Pins P9_30 adn P9_31 utilized in multiple overlays, so those are fighting..  PRU-RPROC for 4.14.x-ti loading on a 4.19.x bone kernel.. Just start over, with a fresh image. Too much random to debug..

Regards,

RadhaKrishnanme, 2019-12-12:
@RobertCNelson Thanks for letting me know.

RadhaKrishnanme, 2019-12-12:
I will re-flash the new image.

pdp7, 2019-12-17:
@RadhaKrishnanme did the new image solve the problem?

RadhaKrishnanme, 2019-12-19:
@pdp7 After consulting with my supervisors, we came up with different solution to overcome this issue rather than flashing with new image. Kindly mark this issue as closed. Thanks for your help.

closed 2019-12-23 by pdp7