adafruit/adafruit-beaglebone-io-python #322: SPI No such file or directory

leo19980219, 2019-12-22:
Hi,
When I try to create SPI object , I get an error "No such file or directory".
```
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from Adafruit_BBIO.SPI import SPI
>>> spi = SPI(0,0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
FileNotFoundError: [Errno 2] No such file or directory
```
There is also a problem with ```spi = SPI(0,1)```
However what's wierd is it's fine with ```spi = SPI(1,0)``` and ```spi = SPI(1,1)```
How can I fix the problem?

Additional info:
[image]

pdp7, 2019-12-23:
Please run:
```
sudo /opt/scripts/tools/version.sh
```
and paste the output here so I can see more about how your BeagleBone is configured.

leo19980219, 2019-12-23:
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLT00C02917BBBK3C44]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.01-00002-g9aa111a004]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 quiet capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.118094] remoteproc remoteproc0: wkup_m3 is available
[    1.366373] remoteproc remoteproc0: powering up wkup_m3
[    1.366492] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.370432] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   10.261409] remoteproc remoteproc1: 4a334000.pru is available
[   10.294295] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   10.223308] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   10.261409] remoteproc remoteproc1: 4a334000.pru is available
[   10.261537] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   10.294295] remoteproc remoteproc2: 4a338000.pru is available
[   10.294431] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.873903] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.124672] pinctrl-single 44e10800.pinmux: pin PIN103 already requested by 481a0000.spi; cannot claim for 48038000.mcasp
[    1.135880] pinctrl-single 44e10800.pinmux: pin-103 (48038000.mcasp) status -22
[    1.143434] pinctrl-single 44e10800.pinmux: could not request pin 103 (PIN103) from group mcasp0_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.875136] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

leo19980219, 2019-12-23:
Also, I try ```config-pin p9.17 spi_cs```
It got an error.
[image]

RobertCNelson, 2019-12-23:
@leo19980219 feel free to follow: https://github.com/adafruit/adafruit-beaglebone-io-python/issues/318#issuecomment-568570810

leo19980219, 2019-12-24:
I follow #318 but SPI still not work.
I'm wondering if there is something I do wrong?
Here is the information after updating bootloader.
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLT00C02917BBBK3C44]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
kernel:[4.14.108-ti-r113]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190801.0-0rcnee0~stretch+20190801]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 quiet capemgr.disable_partno=BB-BONELT-HDMI,BB-BONELT-HDMIN bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.289473] remoteproc remoteproc0: wkup_m3 is available
[    1.383026] remoteproc remoteproc0: powering up wkup_m3
[    1.383052] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.383297] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   13.503637] remoteproc remoteproc1: 4a334000.pru is available
[   13.511632] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   13.464722] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   13.503637] remoteproc remoteproc1: 4a334000.pru is available
[   13.503755] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   13.511632] remoteproc remoteproc2: 4a338000.pru is available
[   13.511752] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.946843] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.958912] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

RobertCNelson, 2019-12-24:
@leo19980219 

> I follow #318 but SPI still not work.

Please actually state "exactly" what you tried to do.  Then let's see the output of:

ls -lha /dev/spi*

Regards,

leo19980219, 2019-12-24:
What I'm tyring to do is creating SPI(0,0) object.
What's wierd is SPI(0,0) SPI(0,1) went wrong, but SPI(1,0) SPI(1,1) was fine.
[image]
Here is the output of ls -lha /dev/spi*
[image]

pdp7, 2019-12-24:
@leo19980219 I see in https://github.com/adafruit/adafruit-beaglebone-io-python/issues/318 that you cleared out the old u-boot on eMMC per @RobertCNelson.  does this issue still occur?

pdp7, 2019-12-24:
@leo19980219 I think it would be useful to see what file paths Adafruit_BBIO is trying to open.

Please put those Python lines into a script file and run:
```
sudo strace -f /tmp/strace,log python3 ./name-of-script.py
```
and paste the output in reply or create a GitHub Gist and link to it (if too big for a reply)

if strace does not exist, install it with:
```
sudo apt-get install strace
```

RobertCNelson, 2019-12-24:
> What I'm tyring to do is creating SPI(0,0) object.
> What's wierd is SPI(0,0) SPI(0,1) went wrong, but SPI(1,0) SPI(1,1) was fine.
> [image]
> Here is the output of ls -lha /dev/spi*
> [image]

On v4.19.x the node names for SPI are now using the mainline "defaults", prior they started with 0...  Thus the old SPI(0,0) is now SPI(1,0) and the old SPI SPI(1,0) is SPI(2,0)...

Regards,

leo19980219, 2019-12-24:
> > What I'm tyring to do is creating SPI(0,0) object.
> > What's wierd is SPI(0,0) SPI(0,1) went wrong, but SPI(1,0) SPI(1,1) was fine.
> > [image]
> > Here is the output of ls -lha /dev/spi*
> > [image]
> 
> On v4.19.x the node names for SPI are now using the mainline "defaults", prior they started with 0... Thus the old SPI(0,0) is now SPI(1,0) and the old SPI SPI(1,0) is SPI(2,0)...
> 
> PS, at the bottom of the file you can see the symlinks: /dev/spi/0.0 -> /dev/spidev1.0 that's to help @pdp7 make it compatible ;)
> 
> Regards,

This is exactly my problem !
SPI(1,0) SPI(1,1) SPI(2,0) SPI(2,1) all work fine.

Thanks for helping.
I'm OK if you close the issue.

closed 2019-12-24 by pdp7