adafruit/adafruit-beaglebone-io-python #323: GPIO input error

R-e-a-p-e-R, 2020-02-02:
I am trying to get a GPIO pin in beaglebone as an input pin. 
I am using the beaglebone black with Ubuntu 18.04 OS on it.
I tried using the adafruit beaglebone python library and it works perfectly for GPIO output pins and also the I2C.
But when I try the following:

import Adafruit_BBIO.GPIO as GPIO
GPIO.setup("P8_14", GPIO.IN)

I get the following result:
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
ValueError: Set gpio mode failed, missing file or invalid permissions.

I searched for this issue and itseems a lot of poeple are facing the same issue.
I tried different approaches and nothing worked.
I also followed #247 but did not get any positive results.
In step 6 of the solution which was posted in #247 I did not get the result which was given.
specifically 

dmesg | grep gpio-of-helper

the above command does not give any result

[image]

Please not the solution mentioned in #247 was for debian but I have UBUNTU 18.04 running on my beaglebone.

Any help regarding this would be appreciated.

pdp7, 2020-02-02:
@R-e-a-p-e-R I think u-boot 2018.03 might be too old.

@RobertCNelson how would the u-boot get updated on the ubuntu image?

RobertCNelson, 2020-02-02:
@R-e-a-p-e-R you need to provide some debugging info, not a screenshot from https://github.com/adafruit/adafruit-beaglebone-io-python/issues/247

Please run and provide the ouput of "./version.sh"..
```
debian@beaglebone:~$ cd /opt/scripts/tools/
debian@beaglebone:/opt/scripts/tools$ git pull
debian@beaglebone:/opt/scripts/tools$ sudo ./version.sh
```

@pdp7 github supports an issue template:

https://help.github.com/en/github/building-a-strong-community/configuring-issue-templates-for-your-repository

This repo should request the version.sh information. ;)

Regards,

RobertCNelson, 2020-02-02:
@pdp7 found the issue template:

https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/.github/ISSUE_TEMPLATE.md

can we just rip it all out and ask users to run:

sudo /opt/scripts/tools/version.sh

From any image downloaded off https://beagleboard.org/ or https://rcn-ee.com/

Regards,

pdp7, 2020-02-03:
Thanks @RobertCNelson.  very good point!

R-e-a-p-e-R, 2020-02-04:
@RobertCNelson Sorry for the late reply. I tried doing what you said. One problem: i am not able to pull from your repo.
when i run `git pull`:
I get the following: `fatal: unable to access 'https://github.com/RobertCNelson/boot-scripts/': Failed to connect to github.com port 443: No route to host`

also I ran the next command: `/opt/scripts/tools$ sudo ./version.sh`

The following is the output:
`ubuntu@arm:/opt/scripts/tools$ sudo ./version.sh
git:/opt/scripts/:[e457c010efc76c4e6fecccb7a5f7ff9b0597c4c0]
eeprom:[A335BNLT00C04414BBBK4234]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2019-04-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2015.10-rc3-00001-gfb375b]:[location: dd MBR]
kernel:[4.14.108-ti-r104]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~bionic+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~bionic+20190227]
pkg:[kmod]:[24-1ubuntu3.2rcnee0~bionic+20190208]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait fixrtc coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep remote
[    1.260331] remoteproc remoteproc0: wkup_m3 is available
[    1.466895] remoteproc remoteproc0: powering up wkup_m3
[    1.466919] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.467192] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.025920] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.027106] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END`

RobertCNelson, 2020-02-04:
@R-e-a-p-e-R , oh you have an easy fix.  The eMMC bootloader is too old blocking the one isntalled on teh microSD.. Just nuke the u-boot on the eMMC:

```
sudo dd if=/dev/zero of=/dev/mmcblk1 bs=1M count=10
```
Regards,

R-e-a-p-e-R, 2020-02-04:
@RobertCNelson , I did what you suggested and this is the output:

>  10+0 records in
> 10+0 records out
> 10485760 bytes (10 MB, 10 MiB) copied, 1.08146 s, 9.7 MB/s

After that I tried to make a GPIO as an input and got the same following error:

> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> ValueError: Set gpio mode failed, missing file or invalid permissions.

pdp7, 2020-02-11:
Could you run `sudo /opt/scripts/tools/version.sh` again so we can be sure that it old u-boot version is gone?

Also, could you try another pin to see if the results are different?

P9_12 should work for GPIO.

pdp7, 2020-02-21:
@R-e-a-p-e-R please reopen if still an issue

closed 2020-02-21 by pdp7