adafruit/adafruit-beaglebone-io-python #324: GPIO.setup reset pin status

vitotafuni, 2020-02-02:
GPIO.setup when called reset the status of the pin
this behaviour is not correct nor is the same on the RPI GPIO lib 

code to check the problem
```
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from Adafruit_BBIO import GPIO
>>> GPIO.setup("P8_7",GPIO.OUT)
>>> GPIO.output("P8_7", GPIO.HIGH)
>>> GPIO.setup("P8_7",GPIO.OUT)
```
Attended behaviour is that P8_7 pin state is HIGH but it goes LOW just after calling GPIO.setup

pdp7, 2020-02-03:
Thanks for reporting.

In order to understand your system configuration better, please run:
```
sudo /opt/scripts/tools/version.sh
```

and paste the output in a reply.

pdp7, 2020-02-11:
@vitotafuni please reopen and run `sudo /opt/scripts/tools/version.sh` if this is still an issue

closed 2020-02-11 by pdp7

bgigous, 2024-11-18:
I came across this issue after trying to read the value (0 or 1) of a GPIO pin using this library. As @vitotafuni mentions, ideally the `GPIO.setup("P8_7",GPIO.OUT)` call only "sets up" the pin for use and makes no change to the pin state.

However, under the hood, it seems like that line actually translates to `GPIO.setup("P8_7",GPIO.OUT, initial=GPIO.LOW)`. So, if P8_07 is set HIGH before the `setup()` call, it is set LOW regardless of it was HIGH or LOW.

Since calling `setup()` is required and cannot be avoided before interacting with a pin through the library, it's currently impossible to get the desired behavior of "read GPIO pin value without changing its state," and this is regardless of which direction (in or out) is passed to `setup()`.

From my cursory look at the source code, it looks like this block will need to change:

```c
// py_gpio.c
static PyObject *py_setup_channel(__attribute__ ((unused)) PyObject *self, PyObject *args, PyObject *kwargs)
{
   // ...
   int initial = 0;
   // ...
   if (!PyArg_ParseTupleAndKeywords(args, kwargs, "si|iii", kwlist, &channel, &direction, &pud, &initial, &delay))
      return NULL;
   // ...
   if (direction == OUTPUT) {

      // Set the pin value and bail if we get an error.
      res = gpio_set_value(gpio, initial);
      // ...
   }
```

If the `initial` kwarg is not passed to `setup()`, the pin value is always changed to 0 for outputs. **So, it'd be great if users are able to see the value of a GPIO output pin using the library without affecting the value of the pin.**

Output of `sudo beagle-version` on a BeagleBone Green with Debian 12:
```
eeprom:[A335BNLTBBG1BBG12325003D]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Bookworm IoT Image 2023-10-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2022.04-00038-gbaca7b469d (Nov 06 2024 - 17:14:19 -0700)]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0.kernel]
UBOOT: Loaded Overlay:[BB-I2C2-RTC-DS3231]
UBOOT: Loaded Overlay:[BB-SPIDEV0-00A0.kernel]
UBOOT: Loaded Overlay:[BB-UART1-00A0.kernel]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0.kernel]
kernel:[5.10.168-ti-r77]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr4=BB-I2C2-RTC-DS3231.dtbo]
uboot_overlay_options:[uboot_overlay_addr5=BB-UART1-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr6=BB-SPIDEV0-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
WARNING:pkg:[bb-cape-overlays]:[NOT_INSTALLED]
pkg:[bb-customizations]:[1.20240119.0-0~bookworm+20240119]
pkg:[bb-usb-gadgets]:[1.20231003.1-0~bookworm+20231003]
pkg:[bb-wl18xx-firmware]:[1.20230414.0-0~bookworm+20230414]
pkg:[kmod]:[30+20221128-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20230210-5]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input render netdev i2c bluetooth gpio admin tisdk weston-launch cloud9ide spi]
cmdline:[console=ttyS0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   13.481926] remoteproc remoteproc0: wkup_m3 is available
[   26.020669] systemd[1]: Reached target remote-fs.target - Remote File Systems.
[   56.324136] remoteproc remoteproc0: powering up wkup_m3
[   56.324208] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[   56.324514] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   83.880732] remoteproc remoteproc1: 4a334000.pru is available
[   83.922347] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[... 59 lines cut ...]
[   13.749117] gpio-of-helper ocp:cape-universal: Allocated GPIO id=52 name='P9_19'
[   13.749313] gpio-of-helper ocp:cape-universal: Allocated GPIO id=53 name='P9_20'
[   13.749503] gpio-of-helper ocp:cape-universal: Allocated GPIO id=54 name='P9_21'
[   13.749710] gpio-of-helper ocp:cape-universal: Allocated GPIO id=55 name='P9_22'
[   13.749929] gpio-of-helper ocp:cape-universal: Allocated GPIO id=56 name='P9_23'
[   13.750147] gpio-of-helper ocp:cape-universal: Allocated GPIO id=57 name='P9_24'
[   13.750675] gpio-of-helper ocp:cape-universal: Allocated GPIO id=58 name='P9_25'
[   13.750916] gpio-of-helper ocp:cape-universal: Allocated GPIO id=59 name='P9_26'
[   13.751124] gpio-of-helper ocp:cape-universal: Allocated GPIO id=60 name='P9_27'
[   13.751327] gpio-of-helper ocp:cape-universal: Allocated GPIO id=61 name='P9_28'
[   13.751624] gpio-of-helper ocp:cape-universal: Allocated GPIO id=62 name='P9_29'
[   13.751845] gpio-of-helper ocp:cape-universal: Allocated GPIO id=63 name='P9_30'
[   13.752040] gpio-of-helper ocp:cape-universal: Allocated GPIO id=64 name='P9_31'
[   13.752269] gpio-of-helper ocp:cape-universal: Allocated GPIO id=65 name='P9_41'
[   13.752503] gpio-of-helper ocp:cape-universal: Allocated GPIO id=66 name='P9_91'
[   13.752707] gpio-of-helper ocp:cape-universal: Allocated GPIO id=67 name='P9_42'
[   13.752921] gpio-of-helper ocp:cape-universal: Allocated GPIO id=68 name='P9_92'
[   13.752933] gpio-of-helper ocp:cape-universal: ready
dmesg | grep wlcore
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

Thank you for your work on this library, @pdp7 !