adafruit/adafruit-beaglebone-io-python #326: GPIO input error on Ubuntu 18.04

JoaoPaulo-Ferreira, 2020-03-07:
Thank you for opening an issue on the Adafruit BeagleBone Python library repository.

In order to understand your system configuration better, please run:
```
sudo /opt/scripts/tools/version.sh
```

and paste the output in a reply.

This script should be present for any image downloaded from:
https://beagleboard.org/ or https://rcn-ee.com/

RobertCNelson, 2020-03-07:
@JoaoPaulo-Ferreira , can you also include the adafruit python error? ;)

I'm guessing it was gpio, so if it was, let's double check something..

```
ls -lha /sys/class/gpio/
```

PS, yeah i really need to someday update the ubuntu images. ;)

JoaoPaulo-Ferreira, 2020-03-08:
@RobertCNelson  you're right. When I try to set a gpio IN it shows this message:

$ ValueError: Set gpio mode failed, missing file or invalid permissions.

The output of $ ls -lha /sys/class/gpio/

drwxrwxr-x  2 root gpio    0 Mar  8 18:17 .
drwxr-xr-x 58 root root    0 Jan  1  2000 ..
--w--w----  1 root gpio 4.0K Mar  8 18:17 export
lrwxrwxrwx  1 root gpio    0 Mar  8 18:17 gpiochip0 -> ../../devices/platform/ocp/44c00000.interconnect/44c00000.interconnect:segment@200000/0
lrwxrwxrwx  1 root gpio    0 Mar  8 18:17 gpiochip32 -> ../../devices/platform/ocp/48000000.interconnect/48000000.interconnect:segment@0/48042
lrwxrwxrwx  1 root gpio    0 Mar  8 18:17 gpiochip64 -> ../../devices/platform/ocp/48000000.interconnect/48000000.interconnect:segment@1000004
lrwxrwxrwx  1 root gpio    0 Mar  8 18:17 gpiochip96 -> ../../devices/platform/ocp/48000000.interconnect/48000000.interconnect:segment@1000006
--w--w----  1 root gpio 4.0K Mar  8 18:17 unexport

RobertCNelson, 2020-03-12:
@JoaoPaulo-Ferreira , so @pdp7 is in charge of this project, i just help out..

the gpio permissions are fine..

can you attach a simple gpio example (with the pins you are actually using?)

I mostly just verify u-boot/kernel are setup properly, @pdp7 knows what the library actually does..

Regards,

pdp7, 2020-03-12:
@RobertCNelson thanks for the assistance

@JoaoPaulo-Ferreira I see the gpio files are root:gpio 
What is the user you logged in as on the Beagle?
The "id" command should show this

RobertCNelson, 2020-03-13:
As long @JoaoPaulo-Ferreira is logged in as ubuntu, he should have gpio enabled..

```
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep admin ]
```

JoaoPaulo-Ferreira, 2020-03-13:
Hi @pdp7 , 
I'm logged as Ubuntu.
Here is the output of $id:

uid=1000(ubuntu) gid=1000(ubuntu) groups=1000(ubuntu),4(adm),15(kmem),20(dialout),24(cdrom),25(floppy),29(audio),30(dip),44(video),46(plugdev)

RobertCNelson, 2020-03-13:
@JoaoPaulo-Ferreira perfect!  So what pins are you trying to use in your library?

JoaoPaulo-Ferreira, 2020-03-13:
@RobertCNelson, I've tried 2 examples. It crashes when I call GPIO.setup("P9_23", GPIO.IN) in first example. The second one I try to use PWM to control a servo and it doesn't work too.

RobertCNelson, 2020-03-13:
Notes...
```
debian@bbb-pwr01-ser09:/sys/devices/platform/ocp/ocp:P9_23_pinmux$ uname -r ; ls -lha
5.5.1-bone6
total 0
drwxrwxr-x  3 root gpio    0 Mar 13 01:24 .
drwxr-xr-x 55 root root    0 Mar 13 01:24 ..
lrwxrwxrwx  1 root gpio    0 Mar 13 01:25 driver -> ../../../../bus/platform/drivers/bone-pinmux-helper
-rw-rw-r--  1 root gpio 4.0K Mar 13 01:24 driver_override
-r--r--r--  1 root gpio 4.0K Mar 13 01:24 modalias
lrwxrwxrwx  1 root gpio    0 Mar 13 01:25 of_node -> ../../../../firmware/devicetree/base/ocp/P9_23_pinmux
drwxrwxr-x  2 root gpio    0 Mar 13 01:24 power
-rw-rw-r--  1 root gpio 4.0K Mar 13 01:24 state
lrwxrwxrwx  1 root gpio    0 Mar 13 01:25 subsystem -> ../../../../bus/platform
-rw-rw-r--  1 root gpio 4.0K Mar 13 01:24 uevent
```

JoaoPaulo-Ferreira, 2020-03-13:
@RobertCNelson should I run something?

JoaoPaulo-Ferreira, 2020-03-13:
Output of uname -r ; ls -lha:

4.14.108-ti-r107
total 0
drwxr-xr-x 41 root root    0 Mar 12 21:01 .
drwxr-xr-x 21 root root    0 Mar 12 21:01 ..
drwxr-xr-x  3 root root    0 Mar 12 21:01 40300000.ocmcram
drwxr-xr-x  5 root root    0 Mar 12 21:01 44e07000.gpio
drwxr-xr-x  4 root root    0 Mar 12 21:01 44e09000.serial
drwxr-xr-x  4 root root    0 Mar 12 21:01 44e0b000.i2c
drwxr-xr-x  4 root root    0 Mar 12 21:01 44e0d000.tscadc
drwxr-xr-x  5 root root    0 Mar 12 21:01 44e35000.wdt
drwxr-xr-x  4 root root    0 Mar 12 21:01 44e3e000.rtc
drwxr-xr-x  9 root root    0 Mar 12 21:01 47400000.usb
drwxr-xr-x  3 root root    0 Mar 12 21:01 48038000.mcasp
drwxr-xr-x  3 root root    0 Mar 12 21:01 48042000.timer
drwxr-xr-x  3 root root    0 Mar 12 21:01 48044000.timer
drwxr-xr-x  3 root root    0 Mar 12 21:01 48046000.timer
drwxr-xr-x  3 root root    0 Mar 12 21:01 48048000.timer
drwxr-xr-x  3 root root    0 Mar 12 21:01 4804a000.timer
drwxr-xr-x  5 root root    0 Mar 12 21:01 4804c000.gpio
drwxr-xr-x  4 root root    0 Mar 12 21:01 48060000.mmc
drwxr-xr-x  4 root root    0 Mar 12 21:01 480c8000.mailbox
drwxr-xr-x  3 root root    0 Mar 12 21:01 480ca000.spinlock
drwxr-xr-x  4 root root    0 Mar 12 21:01 4819c000.i2c
drwxr-xr-x  5 root root    0 Mar 12 21:01 481ac000.gpio
drwxr-xr-x  5 root root    0 Mar 12 21:01 481ae000.gpio
drwxr-xr-x  4 root root    0 Mar 12 21:01 481d8000.mmc
drwxr-xr-x  4 root root    0 Mar 12 21:01 4830e000.lcdc
drwxr-xr-x  3 root root    0 Mar 12 21:01 48310000.rng
drwxr-xr-x  4 root root    0 Mar 12 21:01 49000000.edma
drwxr-xr-x  3 root root    0 Mar 12 21:01 49800000.tptc
drwxr-xr-x  3 root root    0 Mar 12 21:01 49900000.tptc
drwxr-xr-x  3 root root    0 Mar 12 21:01 49a00000.tptc
drwxr-xr-x  7 root root    0 Mar 12 21:01 4a100000.ethernet
drwxr-xr-x  4 root root    0 Mar 12 21:01 4a300000.pruss
drwxr-xr-x  3 root root    0 Mar 12 21:01 4c000000.emif
drwxr-xr-x  3 root root    0 Mar 12 21:01 53100000.sham
drwxr-xr-x  3 root root    0 Mar 12 21:01 53500000.aes
drwxr-xr-x  3 root root    0 Mar 12 21:01 56000000.sgx
-rw-r--r--  1 root root 4.0K Mar 12 21:01 driver_override
-r--r--r--  1 root root 4.0K Mar 12 21:01 modalias
drwxr-xr-x  3 root root    0 Mar 12 21:01 ocp:cape-universal
drwxr-xr-x  6 root root    0 Mar 12 21:01 ocp:l4_wkup@44c00000
drwxrwxr-x  3 root gpio    0 Jan  1  2000 ocp:P9_19_pinmux
drwxrwxr-x  3 root gpio    0 Jan  1  2000 ocp:P9_20_pinmux
lrwxrwxrwx  1 root root    0 Mar 12 21:01 of_node -> ../../../firmware/devicetree/base/ocp
drwxr-xr-x  2 root root    0 Mar 12 21:01 power
lrwxrwxrwx  1 root root    0 Mar 12 21:01 subsystem -> ../../../bus/platform
-rw-r--r--  1 root root 4.0K Mar 12 21:01 uevent

JoaoPaulo-Ferreira, 2020-03-13:
@RobertCNelson There isn't a directory ocp:P9_23_pinmux inside ocp..

RobertCNelson, 2020-03-13:
@JoaoPaulo-Ferreira we have a problem...  "4.14.108-ti-r107" does not equal "kernel:[5.5.1-bone6]" as stated in your very first post..

SO PLEASE RUN:
```sudo /opt/scripts/tools/version.sh```

Before you change anything else...

Regards,

JoaoPaulo-Ferreira, 2020-03-13:
OK.
Thanks @RobertCNelson.

pdp7, 2020-03-23:
@JoaoPaulo-Ferreira were you able to run `sudo /opt/scripts/tools/version.sh` again?

JoaoPaulo-Ferreira, 2020-03-25:
@pdp7 Sure.

git:/opt/scripts/:[43e2e0e554b10d779fdbc730bb0d5d197e467d02]
eeprom:[A335BNLT00C01319BBBK0428]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[rcn-ee.net console Ubuntu Image 2019-06-15]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
kernel:[4.14.108-ti-r107]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190610.0-0rcnee0~bionic+20190610]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~bionic+20190227]
pkg:[kmod]:[24-1ubuntu3.2rcnee0~bionic+20190208]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
groups:[ubuntu : ubuntu adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep admin ]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 q]
dmesg | grep remote
[    1.264065] remoteproc remoteproc0: wkup_m3 is available
[    1.474337] remoteproc remoteproc0: powering up wkup_m3
[    1.474361] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.474611] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.023908] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    1.025180] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

pdp7, 2020-03-25:
@JoaoPaulo-Ferreira are you still having the GPIO permission error?

JoaoPaulo-Ferreira, 2020-03-25:
@pdp7 Yes, I do.

RobertCNelson, 2020-03-25:
@JoaoPaulo-Ferreira you can try, enabling this in /boot/uEnv.txt

```
enable_uboot_cape_universal=1
```

BUT i THINK, your rootfs's kernel/overlays are too old for that..

PS, after you last few messages, i got off my ass and finally updated:

https://elinux.org/BeagleBoardUbuntu#Demo_Image

the new 2020-03-12 will probally just work out of the box..

Regards,

JoaoPaulo-Ferreira, 2020-03-31:
@RobertCNelson It's working now.
I just enabling uboot universal. 

During this time I saw the 2020-03 ubuntu image and I tried with the library and didn't work.
Maybe was the same uboot universal problem...

Anyway, it's working now. 
Thank you @RobertCNelson and @pdp7.

pdp7, 2020-04-01:
@JoaoPaulo-Ferreira good to hear!

closed 2020-04-01 by pdp7