adafruit/adafruit-beaglebone-io-python #328: ValueError: Set gpio mode failed, missing file or invalid permissions.

Shawn2448, 2020-04-14:
I am writing python code on my beaglebone black and tried to set inputs just as described in the README. I get this error:

ValueError: Set gpio mode failed, missing file or invalid permissions.

Does anyone know why?

pdp7, 2020-04-21:
Please paste the output of this command:
```
sudo /opt/scripts/tools/version.sh
```

pdp7, 2020-06-09:
@Shawn2448 please re-open if still a problem and post the output of:
```
sudo /opt/scripts/tools/version.sh
```

closed 2020-06-09 by pdp7

johanhenselmans, 2023-04-02:
Im a having the same problems on a beaglebone that worked fine until I updated it via apt update && apt upgrade -y

here is the output of /opt/scripts/tools/version.sh

git:/opt/scripts/:[1b1122751f7051bd8996f353756ba6ff30e71820]
eeprom:[A335BNLT0A5B2313BBBK0121]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster Console Image 2021-10-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2019.04-g923f8b8 (Jan 02 2022 - 19:05:15 +0000)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-g923f8b8]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0.kernel]
UBOOT: Loaded Overlay:[BB-BONE-LCD4-01-00A1.kernel]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0.kernel]
kernel:[4.19.94-ti-r73]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~buster+20210821]
pkg:[bb-customizations]:[1.20221108.0-0~buster+20221108]
pkg:[bb-usb-gadgets]:[1.20220816.0-0~buster+20220816]
pkg:[bb-wl18xx-firmware]:[1.20221201.0-0~buster+20221201]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio admin tisdk weston-launch cloud9ide]
cmdline:[console=ttyS0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BB-BONE-LCD4-01, coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   44.289283] remoteproc remoteproc0: wkup_m3 is available
[   44.302855] remoteproc remoteproc0: powering up wkup_m3
[   44.302888] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[   44.303304] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.020068] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.033513] gpio-of-helper ocp:cape-universal: ready
END

Trying to set the pins required is done via:
leftButton=digitalio.DigitalInOut(board.P9_15)
leftButton.direction = digitalio.Direction.INPUT
rightButton=digitalio.DigitalInOut(board.P9_23)
rightButton.direction = digitalio.Direction.INPUT
upButton=digitalio.DigitalInOut(board.P9_16)
upButton.direction = digitalio.Direction.INPUT
downButton=digitalio.DigitalInOut(board.P9_30)
downButton.direction = digitalio.Direction.INPUT
enterButton=digitalio.DigitalInOut(board.P9_24)
enterButton.direction = digitalio.Direction.INPUT

the beaglebone is connected to a 4D LCD cape, see https://4dsystems.com.au/4dcape-43 (that has buttons hooked up to SYS_RESET, GPIO1_16, GPIO1_17, GPIO1_19 GPIO3_16 GPIO0_15 and PWR_BUT)

When I try to set the pin manually via config-pin I get this reponse:

echo 48 > /sys/class/gpio/export 

root@woodytestbox:~# config-pin p9.15 gpio
ERROR: open() for /sys/devices/platform/ocp/ocp:P9_15_pinmux/state failed, No such file or directory

show-pins from the buttons shows this:
P9.15                             16 fast rx      7 gpio 1.16 << hi  sysfs (pinmux_bb_lcd_keymap_pins)
P9.23                             17 fast rx      7 gpio 1.17 << hi  sysfs (pinmux_bb_lcd_keymap_pins)
P9.16                             19 fast rx      7 gpio 1.19 << hi  P9_16 (pinmux_bb_lcd_keymap_pins)
P9.30                            102 fast rx      7 gpio 3.16 << hi  P9_30 (pinmux_bb_lcd_keymap_pins)
P9.24                             97 fast rx      7 gpio 0.15 << hi  P9_24 (pinmux_bb_lcd_keymap_pins)