adafruit/adafruit-beaglebone-io-python #329: Unable to sample Analog input BB enhanced

BlueskyEnergy, 2020-04-20:
**Hi**,

this is the situation:

root@beaglebone:/opt# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.71-bone-rt-r17
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom

# universal for gpio's
uboot_overlay_addr0=/lib/firmware/cape-universalh-00A0.dtbo

# overlay for SPI bus on p9.28 - p9.31
#uboot_overlay_addr1=/lib/firmware/BB-SPIDEV1-00A0.dtbo

# overlay for 1-wire temperature on p8.12
uboot_overlay_addr2=/lib/firmware/BB-W1-P8.12-00A0.dtbo

# overlay for can bus on p9.24 and p9.26
#uboot_overlay_addr3=/lib/firmware/BB-CAN1-00A0.dtbo

# overlay for i2c on p9.19 and p9.20
#uboot_overlay_addr4=/lib/firmware/BB-I2C2-00A0.dtbo

###
###Additional custom capes
#uboot_overlay_addr4=/lib/firmware/cape-universal-00A0.dtbo
#uboot_overlay_addr5=/lib/firmware/<file5>.dtbo
#uboot_overlay_addr6=/lib/firmware/<file6>.dtbo
#uboot_overlay_addr7=/lib/firmware/<file7>.dtbo
###
###Custom Cape
#dtb_overlay=/lib/firmware/<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.4.x-ti kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-4-TI-00A0.dtbo
###pru_rproc (4.14.x-ti kernel)
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_uio (4.4.x-ti, 4.14.x-ti & mainline/bone kernel)
#uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
#enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 quiet video=HDMI-A-1:1024x768@60e

#Use an overlayfs on top of a read-only root filesystem:
#cmdline=coherent_pool=1M net.ifnames=0 quiet overlayroot=tmpfs

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
##################cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

#################cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
################cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
##############cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
#############cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
############cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
###########cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
##########cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
#########cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
########cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
#######cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
######cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
#####cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
####cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
###cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
##cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

root@beaglebone:/opt# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[e9bcff232834702c1c810710706ee815d77b080b]
eeprom:[A335BNLTSE0A2919BBE3E7CD]
model:[SanCloud_BeagleBone_Enhanced]:WiFi AP Broken on Mainline
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.71-bone-rt-r17]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_addr0=/lib/firmware/cape-universalh-00A0.dtbo]
uboot_overlay_options:[uboot_overlay_addr2=/lib/firmware/BB-W1-P8.12-00A0.dtbo]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_adc=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20191017.0-0rcnee0~stretch+20191017]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet]
dmesg | grep pinctrl-single
[    1.041985] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.377741] pinctrl-single 44e10800.pinmux: pin PIN107 already requested by ocp:P9_25_pinmux; cannot claim for 48038000.mcasp
[    1.389352] pinctrl-single 44e10800.pinmux: pin-107 (48038000.mcasp) status -22
[    1.396711] pinctrl-single 44e10800.pinmux: could not request pin 107 (PIN107) from group mcasp0_pins  on device pinctrl-single
[    1.480636] pinctrl-single 44e10800.pinmux: pin PIN40 already requested by ocp:P8_45_pinmux; cannot claim for 0-0070
[    1.491240] pinctrl-single 44e10800.pinmux: pin-40 (0-0070) status -22
[    1.497808] pinctrl-single 44e10800.pinmux: could not request pin 40 (PIN40) from group nxp_hdmi_bonelt_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    1.052356] gpio-of-helper ocp:cape-universal: ready
END

Thank you for your help.

BlueskyEnergy, 2020-04-20:
when  I try to run:
import Adafruit_BBIO.ADC as ADC
ADC.setup()

#read returns values 0-1.0
value = ADC.read("P9_40")

#read_raw returns non-normalized value
value = ADC.read_raw("P9_40")

#read returns values 0-1.0
value = ADC.read("P9_39")

#read_raw returns non-normalized value
value = ADC.read_raw("P9_39")

It returns:
#python test_adc.py
Traceback (most recent call last):
  File "test_adc.py", line 2, in <module>
    ADC.setup()
RuntimeError: Unable to setup ADC system. Possible causes are:
  - A cape with a conflicting pin mapping is loaded
  - A device tree object is loaded that uses the same name for a fragment: helper

BlueskyEnergy, 2020-04-20:
root@beaglebone:/opt/adc# cat strace.log
2523  execve("/usr/bin/python", ["python", "test_adc.py"], [/* 19 vars */]) = 0
2523  brk(NULL)                         = 0x1c5c000
2523  uname({sysname="Linux", nodename="beaglebone", ...}) = 0
2523  access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
2523  mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f14000
2523  access("/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
2523  open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
2523  fstat64(3, {st_mode=S_IFREG|0644, st_size=31160, ...}) = 0
2523  mmap2(NULL, 31160, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f0c000
2523  close(3)                          = 0
2523  access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
2523  open("/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
2523  read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\341F\0\0004\0\0\0"..., 512) = 512
2523  lseek(3, 105412, SEEK_SET)        = 105412
2523  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1560) = 1560
2523  lseek(3, 70056, SEEK_SET)         = 70056
2523  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2523  fstat64(3, {st_mode=S_IFREG|0755, st_size=106972, ...}) = 0
2523  mmap2(NULL, 143952, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ecb000
2523  mprotect(0xb6edc000, 61440, PROT_NONE) = 0
2523  mmap2(0xb6eeb000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0xb6eeb000
2523  mmap2(0xb6eed000, 4688, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6eed000
2523  close(3)                          = 0
2523  access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
2523  open("/lib/arm-linux-gnueabihf/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
2523  read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0`\t\0\0004\0\0\0"..., 512) = 512
2523  lseek(3, 8684, SEEK_SET)          = 8684
2523  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
2523  lseek(3, 8328, SEEK_SET)          = 8328
2523  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2523  fstat64(3, {st_mode=S_IFREG|0644, st_size=9804, ...}) = 0
2523  mmap2(NULL, 73912, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6eb8000
2523  mprotect(0xb6eba000, 61440, PROT_NONE) = 0
2523  mmap2(0xb6ec9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6ec9000
2523  close(3)                          = 0
2523  access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
2523  open("/lib/arm-linux-gnueabihf/libutil.so.1", O_RDONLY|O_CLOEXEC) = 3
2523  read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\24\n\0\0004\0\0\0"..., 512) = 512
2523  lseek(3, 8720, SEEK_SET)          = 8720
2523  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
2523  lseek(3, 8364, SEEK_SET)          = 8364
2523  read(3, "A2\0\0\0aeabi\0\1(\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 51) = 51
2523  fstat64(3, {st_mode=S_IFREG|0644, st_size=9840, ...}) = 0
2523  mmap2(NULL, 73904, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6ea5000
2523  mprotect(0xb6ea7000, 61440, PROT_NONE) = 0
2523  mmap2(0xb6eb6000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6eb6000
2523  close(3)                          = 0
2523  access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
2523  open("/lib/arm-linux-gnueabihf/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
2523  read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\220\30\0\0004\0\0\0"..., 512) = 512
2523  lseek(3, 70288, SEEK_SET)         = 70288
2523  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
2523  lseek(3, 69924, SEEK_SET)         = 69924
2523  read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
2523  fstat64(3, {st_mode=S_IFREG|0644, st_size=71408, ...}) = 0
2523  mmap2(NULL, 135464, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e83000
2523  mprotect(0xb6e94000, 61440, PROT_NONE) = 0
2523  mmap2(0xb6ea3000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x10000) = 0xb6ea3000
2523  close(3)                          = 0
2523  access("/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
2523  open("/lib/arm-linux-gnueabihf/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
2523  read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0h=\0\0004\0\0\0"..., 512) = 512
2523  lseek(3, 422364, SEEK_SET)        = 422364
2523  read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
2523  lseek(3, 422004, SEEK_SET)        = 422004
2523  read(3, "A4\0\0\0aeabi\0\1*\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 53) = 53
2523  fstat64(3, {st_mode=S_IFREG|0644, st_size=423484, ...}) = 0
2523  mmap2(NULL, 487548, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e0b000
2523  mprotect(0xb6e72000, 61440, PROT_NONE) = 0
[... 990 lines cut ...]
2523  fstat64(3, {st_mode=S_IFREG|0644, st_size=2237, ...}) = 0
2523  read(3, "TZif2\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\7\0\0\0\7\0\0\0\0"..., 4096) = 2237
2523  _llseek(3, -28, [2209], SEEK_CUR) = 0
2523  read(3, "\nCET-1CEST,M3.5.0,M10.5.0/3\n", 4096) = 28
2523  close(3)                          = 0
[... cut ...]
2523  close(4)                          = 0
2523  write(2, "RuntimeError", 12)      = 12
2523  write(2, ": ", 2)                 = 2
2523  write(2, "Unable to setup ADC system. Poss"..., 184) = 184
2523  write(2, "\n", 1)                 = 1
2523  rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d41fd1}, {sa_handler=0x59319d, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6d41fd1}, 8) = 0
2523  exit_group(1)                     = ?
2523  +++ exited with 1 +++

BlueskyEnergy, 2020-04-20:
python -c "import Adafruit_BBIO; print Adafruit_BBIO"
<module 'Adafruit_BBIO' from '/usr/local/lib/python2.7/dist-packages/Adafruit_BBIO-1.0.10-py2.7-linux-armv7l.egg/Adafruit_BBIO/__init__.pyc'>

BlueskyEnergy, 2020-04-20:
this directory /sys/bus/iio/devices is empty

RobertCNelson, 2020-04-20:
@BlueskyEnergy i "really" don't want to touch this image:

```
dogtag:[BeagleBoard.org Debian Image 2018-08-30]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
kernel:[4.14.71-bone-rt-r17]
```

If you heart and sweat are in this eMMC image, i don't want to damage it..

Latest Images release is 2020-04-06

https://beagleboard.org/latest-images

and a list of all varients of 2020-04-06..

https://elinux.org/Beagleboard:Latest-images-testing

Regards,

RobertCNelson, 2020-04-20:
If we just look at the ADC problem you have this line enabled in /boot/uEnv.txt

disable_uboot_overlay_adc=1

So i'd expect the ADC to not work.. ;)

Regards,

BlueskyEnergy, 2020-04-20:
@RobertCNelson I commented the line you mentioned from:
disable_uboot_overlay_adc=1
to:
#disable_uboot_overlay_adc=1
and the BB doesn´t reboot anymore.

RobertCNelson, 2020-04-20:
@BlueskyEnergy Ah CRAP!! and totally sorry...  See when i said i didn't want to touch it this error, i shouldn't have touched it..

Grab the console (microSD) image:

https://elinux.org/Beagleboard:Latest-images-testing#Debian_10_.28Buster.29_Console

Login:

```
sudo mkdir disk
sudo mount /dev/mmcblk1p1 ./disk/
sudo nano ./disk/boot/uEnv.txt
sudo umount ./disk/
sudo reboot
```

REgards,

pdp7, 2020-06-09:
@BlueskyEnergy are you still having this issue?

BlueskyEnergy, 2020-06-09:
> @BlueskyEnergy are you still having this issue?

I solved updating the kernel, as mentioned from Robert, it was to hold.
Thanks again for the support.

closed 2020-06-09 by BlueskyEnergy