adafruit/adafruit-beaglebone-io-python #331: unable to get Correct output  , SPI

aljandali, 2020-04-29:
Unable to configure pins BBB

I am beginner with Beaglebone and would like to use SPI to transfer data to a Microcontroller , but i have a problem
When I try to run Adafruit-SPI-test :

> from Adafruit_BBIO.SPI import SPI
> 
> print("SPI(1,0)")
> spi = SPI(1,0)
> print(spi.xfer2([32, 11, 110, 22, 220]))
> spi.close()

 I get the following output:

> SPI(1,0)
> [0, 0, 0, 0, 0]
if i connect it with logic analyser i get a strange data 

the output of sudo /opt/scripts/tools/version.sh :

> git:/opt/scripts/:[b61125c1485bee929340cacc06c85c6fcfd678bc]
> eeprom:[A335BNLT000C1818BBBG1658]
> model:[TI_AM335x_BeagleBone_Black]
> dogtag:[BeagleBoard.org Debian Stretch imgtec Image 2020-04-06]
> bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
> UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
> UBOOT: Loaded Overlay:[BB-ADC-00A0]
> UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
> UBOOT: Loaded Overlay:[BB-NHDMI-TDA998x-00A0]
> UBOOT: Loaded Overlay:[BB-SPIDEV0-00A0]
> UBOOT: Loaded Overlay:[BB-SPIDEV1-00A0]
> kernel:[4.4.155-ti-r155]
> nodejs:[v6.17.0]
> /boot/uEnv.txt Settings:
> uboot_overlay_options:[disable_uboot_overlay_audio=1]
> uboot_overlay_options:[enable_uboot_overlays=1]
> uboot_overlay_options:[uboot_overlay_addr4=/lib/firmware/BB-SPIDEV0-00A0.dtbo]
> uboot_overlay_options:[uboot_overlay_addr5=/lib/firmware/BB-SPIDEV1-00A0.dtbo]
> uboot_overlay_options:[disable_uboot_overlay_audio=1]
> uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
> uboot_overlay_options:[enable_uboot_cape_universal=1]
> pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
> pkg:[bb-cape-overlays]:[4.14.20200403.2-0~stretch+20200424]
> pkg:[bb-wl18xx-firmware]:[1.20200420.0-0rcnee0~stretch+20200420]
> pkg:[kmod]:[23-2rcnee1~stretch+20171005]
> WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
> pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
> groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
> cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
> dmesg | grep remote
> [    1.258623]  remoteproc0: wkup_m3 is available
> [    1.258639]  remoteproc0: Note: remoteproc is still under development and considered experimental.
> [    1.258648]  remoteproc0: THE BINARY FORMAT IS NOT YET FINALIZED, and backward compatibility isn't yet guaranteed.
> [    1.360645]  remoteproc0: powering up wkup_m3
> [    1.361022]  remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
> [    1.368029]  remoteproc0: remote processor wkup_m3 is now up
> dmesg | grep pru
> dmesg | grep pinctrl-single
> [    1.054467] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
> dmesg | grep gpio-of-helper
> [    1.055249] gpio-of-helper ocp:cape-universal: Allocated GPIO id=0
> [    1.055397] gpio-of-helper ocp:cape-universal: Allocated GPIO id=1
> [    1.055408] gpio-of-helper ocp:cape-universal: ready
> lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

END

What am I doing wrong?

Best regards,

jan

pdp7, 2020-05-14:
@aljandali sorry, I somehow missed this issue.  Is this still a problem?

pdp7, 2020-06-09:
sudo /opt/scripts/tools/version.sh

closed 2020-06-09 by pdp7