adafruit/adafruit-beaglebone-io-python #332: Error when installing Adafruit_BBIO Library with pip

frank2597, 2020-05-09:
Hi, i am currently having some issues installing the Adafruit_BBIO library on a Beaglebone black(Debian 10 buster).
I am receiving the error message "Failed building wheel for Adafruit-BBIO", the information is below:

In order to understand your system configuration better, please run:

sudo /opt/scripts/tools/version.sh
```
root@beaglebone:~# sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[46d33928c4a396ef472829bcf22761546e550a6f]
eeprom:[A335BNLT00C01619BBBK10B5]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-07-07]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gb96b20de74]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
kernel:[4.19.50-ti-r20]
nodejs:[v10.15.2]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.4.20190610.0-0rcnee0~buster+20190610]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~buster+20190227]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190114-1]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c cloud9ide gpio pwm eqep admin spi tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 cape_universal=enable quiet]
dmesg | grep remote
[    1.820831] remoteproc remoteproc0: wkup_m3 is available
[    1.822345] remoteproc remoteproc1: 4a334000.pru is available
[    1.822854] remoteproc remoteproc2: 4a338000.pru is available
[    2.126063] remoteproc remoteproc0: powering up wkup_m3
[    2.126180] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.129720] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    1.822345] remoteproc remoteproc1: 4a334000.pru is available
[    1.822542] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    1.822854] remoteproc remoteproc2: 4a338000.pru is available
[    1.822937] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.401369] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.414427] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

When i try to run the code 
`sudo pip install Adafruit_BBIO
`
i get the following output which contains the error message

```
root@beaglebone:~# sudo pip install Adafruit_BBIO
Collecting Adafruit_BBIO
  Using cached https://files.pythonhosted.org/packages/53/2b/b0e3dce6113225aae9beb886b2addd4fd5c140ba93c9503d7e4a97021bcc/Adafruit_BBIO-1.1.1.tar.gz
Building wheels for collected packages: Adafruit-BBIO
  Running setup.py bdist_wheel for Adafruit-BBIO ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-n7NVjA/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-qbYwCh --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  copying Adafruit_I2C.py -> build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/Adafruit_BBIO
  copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
  copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
  copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
[... 38 lines cut ...]
    copying overlays/__init__.py -> build/lib.linux-armv7l-2.7/overlays
    copying overlays/builder.py -> build/lib.linux-armv7l-2.7/overlays
    running build_ext
    building 'Adafruit_BBIO.GPIO' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/source
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-C6mgOs/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
    source/py_gpio.c:558:14: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
        {"setup", (PyCFunction)py_setup_channel, METH_VARARGS | METH_KEYWORDS, "Set up the GPIO channel, direction and (optional) pull/up down control\nchannel        - Either: RPi board pin number (not BCM GPIO 00..nn number).  Pins start from 1\n                 or    : BCM GPIO number\ndirection      - INPUT or OUTPUT\n[pull_up_down] - PUD_OFF (default), PUD_UP or PUD_DOWN\n[initial]      - Initial value for an output channel\n[delay]        - Time in milliseconds to wait after exporting gpio pin"},
                  ^
    source/py_gpio.c:562:25: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
        {"add_event_detect", (PyCFunction)py_add_event_detect, METH_VARARGS | METH_KEYWORDS, "Enable edge detection events for a particular GPIO channel.\nchannel      - either board pin number or BCM number depending on which mode is set.\nedge         - RISING, FALLING or BOTH\n[callback]   - A callback function for the event (optional)\n[bouncetime] - Switch bounce timeout in ms for callback"},
                             ^
    source/py_gpio.c:565:27: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
        {"add_event_callback", (PyCFunction)py_add_event_callback, METH_VARARGS | METH_KEYWORDS, "Add a callback for an event already defined using add_event_detect()\ngpio         - gpio channel\ncallback     - a callback function\n[bouncetime] - Switch bounce timeout in ms"},
                               ^
    cc1: all warnings being treated as errors
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-n7NVjA/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-wOX_dm/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-n7NVjA/Adafruit-BBIO/

```

I will be very grateful if someone could help me resolve this issue, thanks.

pdp7, 2020-05-09:
@frank2597 if it is possible for you, please try this newer image:
[AM3358 Debian 10.3 2020-04-06 4GB SD IoT](http://beagleboard.org/latest-images)

If you aren't able to try that, then I will download the 2019-07-07 image and try to reproduce.

frank2597, 2020-05-09:
Hello pdp7, thanks for the reply. I have tried the newer image however i received the same error message.

frank2597, 2020-05-10:
Hello @pdp7 i recently discovered an alternative to install the library by running the following commands:

```
sudo apt-get update
sudo apt-get install build-essential python-dev python-pip python-smbus -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python setup.py install
cd ..
sudo rm -rf adafruit-beaglebone-io-python
```
So i am considering closing this issue. @pdp7 I was hoping that if you have the time, if you could please check out my new issue that i encountered at the following link:

https://github.com/adafruit/adafruit-beaglebone-io-python/issues/333 

Thank you very much!

closed 2020-05-11 by frank2597