adafruit/adafruit-beaglebone-io-python #333: BeagleBone Black Error when using I2C to communicate with sensor

frank2597, 2020-05-10:
Hello, i am trying to use the Adafruit_bno55 Library with a beaglebone black to attain data across the i2c bus from the BNO055 sensor. However when i try to run the test code that they offer here:

(https://learn.adafruit.com/bno055-absolute-orientation-sensor-with-raspberry-pi-and-beaglebone-black/software)

 I receive an error message which will be displayed further below. 

**In order for you to understand my system configuration better, i ran the code:**
```
sudo /opt/scripts/tools/version.sh
```
and pasted the output in below:
```
debian@beaglebone:/var/lib/cloud9$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT00C01619BBBK10B5]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   69.143961] remoteproc remoteproc0: wkup_m3 is available
[   69.312300] remoteproc remoteproc0: powering up wkup_m3
[   69.312332] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   69.312611] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   72.265343] remoteproc remoteproc1: 4a334000.pru is available
[   72.284564] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   72.265343] remoteproc remoteproc1: 4a334000.pru is available
[   72.265538] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   72.284564] remoteproc remoteproc2: 4a338000.pru is available
[   72.284721] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.202674] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.216283] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

When i type the following code that's supposed to read data from the BNO055 sensor on the i2c bus i get the following error:

```
root@beaglebone:~/Adafruit_Python_BNO055/examples# sudo python simpletest.py
Traceback (most recent call last):
  File "simpletest.py", line 45, in <module>
    if not bno.begin():
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 382, in begin
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 358, in _config_mode
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 416, in set_mode
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 302, in _write_byte
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/I2C.py", line 116, in write8
  File "build/bdist.linux-armv7l/egg/Adafruit_PureIO/smbus.py", line 322, in write_byte_data
IOError: [Errno 16] Device or resource busy
```

I suspect that there might be some issue with the device tree overlays but i am unsure. Also when i scan the i2c bus i notice that the BNO055 sensor is not showing:

```
root@beaglebone:~/Adafruit_Python_BNO055/examples# i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- 34 -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: UU -- -- -- -- -- -- --
root@beaglebone:~/Adafruit_Python_BNO055/examples#
```
I can assure you that the senor is not broken and works fine since i tried it on other microprocessors such as the raspberry pi and i was able to detect it there as well. However i can not detect it on the beaglebone black.  The sensor default address is 0x28 and it should show but does not. 

I will appreciate any help in resolving this issue. Thank you.

pdp7, 2020-05-11:
@frank2597 it might be an issue of the library looking at the wrong i2c controller.

Please run:
```
sudo strace -Ff -e open i2cdetect -y -r 2
```
If you don't have 'strace', this should install it:
```
sudo apt-get install strace
```
Here are my results:
```
debian@beaglebone:~$ sudo strace -Ff -e open i2cdetect -y -r 2
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
open("/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
open("/dev/i2c/2", O_RDWR)              = -1 ENOENT (No such file or directory)
open("/dev/i2c-2", O_RDWR)              = 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```
(with the techlab cape which has a couple i2c devices).

Then run Python with:
```
sudo strace -o /tmp/strace.log -fF python simpletest.py
```
and then grep for /dev/i2c:
```
sudo grep /dev/i2c /tmp/strace.log
```
This should help determine if both `i2cdetect` and python are looking at the same /dev/i2cX entry

frank2597, 2020-05-11:
Hello @pdp7 ,
 
After running 
`sudo strace -Ff -e open i2cdetect -y -r 2`
I got the output:
```
root@beaglebone:~# sudo strace -Ff -e open i2cdetect -y -r 2
strace: deprecated option -F ignored
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- 28 -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
+++ exited with 0 +++
```
I than ran python with: 
`sudo strace -o /tmp/strace.log -fF python simpletest.py`
and got this putput:
```
root@beaglebone:~/Adafruit_Python_BNO055/examples# sudo strace -o /tmp/strace.log -fF python simpletest.py
strace: deprecated option -F ignored
Traceback (most recent call last):
  File "simpletest.py", line 45, in <module>
    if not bno.begin():
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 382, in begin
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 358, in _config_mode
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 416, in set_mode
  File "build/bdist.linux-armv7l/egg/Adafruit_BNO055/BNO055.py", line 302, in _write_byte
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/I2C.py", line 116, in write8
  File "build/bdist.linux-armv7l/egg/Adafruit_PureIO/smbus.py", line 322, in write_byte_data
IOError: [Errno 16] Device or resource busy
```
Not sure why but my outputs with trace are a bit different than yours.

frank2597, 2020-05-12:
Hey i think you were right. Looking at line 242 at the following link:

`import Adafruit_GPIO.I2C as I2C`

[(https://github.com/adafruit/Adafruit_Python_BNO055/blob/master/Adafruit_BNO055/BNO055.py)](url)

And the imported file,i2c.py, at this link:
[https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/I2C.py](url)

Which contains the following code:

```
def get_default_bus():
    """Return the default bus number based on the device platform.  For a
    Raspberry Pi either bus 0 or 1 (based on the Pi revision) will be returned.
    For a Beaglebone Black the first user accessible bus, 1, will be returned.
    """
    plat = Platform.platform_detect()
    if plat == Platform.RASPBERRY_PI:
        if Platform.pi_revision() == 1:
            # Revision 1 Pi uses I2C bus 0.
            return 0
        else:
            # Revision 2 Pi uses I2C bus 1.
            return 1
    elif plat == Platform.BEAGLEBONE_BLACK:
        # Beaglebone Black has multiple I2C buses, default to 1 (P9_19 and P9_20).
        return 1
    else:
        raise RuntimeError('Could not determine default I2C bus for platform.')
```
The beaglebone black's i2c bus is being set to 1 when it should be 2. 
The current issue that i am now having is trying to locate this file in the cmd terminal so i can edit it with nano, but i cant seem to locate it.

pdp7, 2020-05-12:
@frank2597 I believe the package installed through pip would get installed to something like:
/usr/local/lib/python3.5/dist-packages/Adafruit_BBIO-1.1.1-py3.5-linux-armv7l.egg/Adafruit_I2C.py

However, the best way to make changes would probably be to clone, build and install:
https://github.com/adafruit/Adafruit_Python_GPIO/

frank2597, 2020-05-13:
Hey @pdp7,

I am still new to the workings of git so i was hoping that you could please clear somethings up for me.

You said that the package installed throught pip would like like: /usr/local/lib/python3.5/dist-packages/Adafruit_BBIO-1.1.1-py3.5-linux-armv7l.egg/Adafruit_I2C.py
but mine will look different since i installed it by cloning the link://github.com/adafruit/adafruit-beaglebone-io-python.git

Now correct me if I'm wrong but i believe that whether  you install it via pip or by cloning it to the beagleone, the contents that are being installed are from
the same git repository https://github.com/adafruit/adafruit-beaglebone-io-python ? I ask this since i cant seem to find the i2c.py file in the repository, yet when you install it with pip the i2c.py file will be located where you said it would be. 

My second question is how this git repository; https://github.com/adafruit/Adafruit_Python_BNO055 

with file located in:

https://github.com/adafruit/Adafruit_Python_BNO055/blob/master/Adafruit_BNO055/BNO055.py

can import Adafruit_GPIO.I2C? I ask this because i have not installed Adafruit_GPIO, yet i receiveD no errors when running simpletest.py. I understand that simpletest.py can import BNO055 from Adafruit_BNO055 since i installed Adafruit_Python_BNO055 repository

`from Adafruit_BNO055 import BNO055`

but i do not understand how Adafruit_BNO055/BNO055.py can import Adafruit_GPIO.I2C cause i did not manually install/clone it
`import Adafruit_GPIO.I2C as I2C`

Unless Adafruit_GPIO was installed somehow when i installed Adafruit_Python_BNO055 repository?

frank2597, 2020-05-14:
Hey @pdp7,
I forked the repositories Adafruit_Python_GPIO and  
Adafruit_Python_BNO055 which are now located in my github:
https://github.com/frank2597/Adafruit_Python_GPIO
https://github.com/frank2597/Adafruit_Python_BNO055

in Adafruit_Python_GPIO/Adafruit_GPIO/I2C.py  i changed the I2C bus to 2 on line 55: 
`return 2`

in Adafruit_Python_GPIO/setup.py  i changed line 32 to: 
`url  = 'https://github.com/frank2597/Adafruit_Python_GPIO/',`

in Adafruit_Python_BNO055/setup.py i changed lines 27 and 28 to:
` url  = 'https://github.com/frank2597/Adafruit_Python_BNO055/',`

`dependency_links =['https://github.com/frank2597/Adafruit_Python_GPIO/tarball/master#egg=Adafruit-GPIO-0.9.3'],`

then i cloned Adafruit_Python_BNO055 and installed the dependency Adafruit_Python_GPIO with setup.py

However unzipping the  Adafruit_GPIO-1.0.4-py2.7.egg file and looking into the I2C.py file i noticed that my changes have not been reflected and the I2C bus is still set to 1. It seems its still cloning the original repository and not my edited forked version. Do you have any idea what I'm may be doing wrong?

frank2597, 2020-05-14:
Hello @pdp7 
I hit a few roadblocks but i was finally able to edit the I2C.py file. I believe the i2c comunication is working now since im getting some output for the selft test on the BNO055.
Unfortunately after changing the i2c bus to 2 i am now getting the following error:

```
 root@beaglebone:~/Adafruit_Python_BNO055/examples# python simpletest.py
System status: 5
Self test result (0x0F is normal): 0x0F
Software version:   785
Bootloader version: 21
Accelerometer ID:   0xFB
Magnetometer ID:    0x32
Gyroscope ID:       0x0F

Reading BNO055 data, press Ctrl-C to quit...
Traceback (most recent call last):
  File "simpletest.py", line 68, in <module>
    heading, roll, pitch = bno.read_euler()
  File "/root/Adafruit_Python_BNO055/Adafruit_BNO055/BNO055.py", line 645, in read_euler
    heading, roll, pitch = self._read_vector(BNO055_EULER_H_LSB_ADDR)
  File "/root/Adafruit_Python_BNO055/Adafruit_BNO055/BNO055.py", line 633, in _read_vector
    data = self._read_bytes(address, count*2)
  File "/root/Adafruit_Python_BNO055/Adafruit_BNO055/BNO055.py", line 320, in _read_bytes
    return bytearray(self._i2c_device.readList(address, length))
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/I2C.py", line 136, in readList
  File "build/bdist.linux-armv7l/egg/Adafruit_PureIO/smbus.py", line 255, in read_i2c_block_data
TypeError: one character string expected
```
could you please help me resolve this? Thanks

pdp7, 2020-05-14:
@frank2597 It appears you updated the comment rather than adding a new comment.
> Not sure why but my outputs with strace are a bit different than yours.
The output of strace should be in: /tmp/strace.log 
This should reveal what system call is resulting in "IOError: [Errno 16] Device or resource busy"

frank2597, 2020-05-14:
@pdp7 
yep i hit a few roadblocks regarding editing the i2c.py file but i figured it out so i deleted a few comments so you wouldn't deal with them

pdp7, 2020-05-14:
can you please post the results of `grep i2c  /tmp/strace.log`?

frank2597, 2020-05-14:
`sudo strace -Ff -e open i2cdetect -y -r 2`

It seems to be infinitely long. Does this portion seem correct?  
```
1581  openat(AT_FDCWD, "/usr/lib/python2.7/plat-arm-linux-gnueabihf/apport_pyth$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/plat-arm-linux-gnueabihf/apport_pyth$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/plat-arm-linux-gnueabihf/apport_pyth$
1581  stat64("/usr/lib/python2.7/lib-tk", {st_mode=S_IFDIR|0755, st_size=4096, $
1581  stat64("/usr/lib/python2.7/lib-tk", {st_mode=S_IFDIR|0755, st_size=4096, $
1581  stat64("/usr/lib/python2.7/lib-tk/apport_python_hook", 0xbeb34b80) = -1 E$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-tk/apport_python_hook.arm-linux-$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-tk/apport_python_hook.so", O_RDO$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-tk/apport_python_hookmodule.so",$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-tk/apport_python_hook.py", O_RDO$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-tk/apport_python_hook.pyc", O_RD$
1581  stat64("/usr/lib/python2.7/lib-old", 0xbeb33a60) = -1 ENOENT (No such fil$
1581  stat64("/usr/lib/python2.7", {st_mode=S_IFDIR|0755, st_size=20480, ...}) $
1581  stat64("/usr/lib/python2.7/lib-old", 0xbeb34a50) = -1 ENOENT (No such fil$
1581  stat64("/usr/lib/python2.7/lib-dynload", {st_mode=S_IFDIR|0755, st_size=4$
1581  stat64("/usr/lib/python2.7/lib-dynload", {st_mode=S_IFDIR|0755, st_size=4$
1581  stat64("/usr/lib/python2.7/lib-dynload/apport_python_hook", 0xbeb34b80) =$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-dynload/apport_python_hook.arm-l$
1581  openat(AT_FDCWD, "/usr/lib/python2.7/lib-dynload/apport_python_hook.so", $
```

frank2597, 2020-05-14:
Here is the output from: `sudo strace -o /tmp/strace.log -fF python simpletest.py`
```
root@beaglebone:~# grep i2c /tmp/strace.log
1718  openat(AT_FDCWD, "/dev/i2c-2", O_RDWR|O_LARGEFILE) = 4
1718  read(5, "         self._i2c_device.write8"..., 4096) = 4096
1718  read(5, "         self._i2c_device.write8"..., 4096) = 4096
1718  read(5, "         self._i2c_device.write8"..., 4096) = 4096
1718  write(2, "return bytearray(self._i2c_devic"..., 61) = 61
```

pdp7, 2020-05-15:
> Here is the output
> 
> ```
> root@beaglebone:~# grep i2c /tmp/strace.log
> 1718  openat(AT_FDCWD, "/dev/i2c-2", O_RDWR|O_LARGEFILE) = 4
> 1718  read(5, "         self._i2c_device.write8"..., 4096) = 4096
> 1718  read(5, "         self._i2c_device.write8"..., 4096) = 4096
> 1718  read(5, "         self._i2c_device.write8"..., 4096) = 4096
> 1718  write(2, "return bytearray(self._i2c_devic"..., 61) = 61
> ```

Could you post the full trace as a GitHub Gist?

> 1718  openat(AT_FDCWD, "/dev/i2c-2", O_RDWR|O_LARGEFILE) = 4

This means the i2c device is on the file descriptor 4.  There should be more lines after it with fd 4 to show what happens.

Also could you run strace with i2cdetect for the adapter where it sees the device?  And please paste the log file in a gist.

You can change the log filename in "strace -o <FILE>" to be whatever you would like.

Oh and yes, "-F" is unnecessary.  It is a habit of my mine from older systems.  "strace" just needs "-f" to make sure it traces the child processes as well.

frank2597, 2020-05-20:
@pdp7 Hey, ill be using the Adafruit_CircuitPython_BNO055 library so ill be closing this issue. Thanks!

closed 2020-05-20 by frank2597