adafruit/adafruit-beaglebone-io-python #334: BBGW/balenaOS Unable to Setup GPIO

crbn60, 2020-05-12:
This appears similar to a variety of issues I see where the library is unable to locate the subsystem, at least that's my interpretation. Running as root using the library installed from master:

```shell
# python -c "import Adafruit_BBIO.ADC as ADC; ADC.setup();"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
RuntimeError: Unable to setup ADC system. Possible causes are: 
  - A cape with a conflicting pin mapping is loaded 
  - A device tree object is loaded that uses the same name for a fragment: helper
```

I'm running a Beaglebone Green Wireless with balenaOS on top of it. In lieu of the version script:

```shell
# python -VVV
Python 3.8.2 (default, Apr 27 2020, 19:06:36) 
[GCC 6.3.0 20170516]
# uname -a
Linux 1c72c55 4.14.53+ #1 SMP PREEMPT Mon Mar 9 08:28:48 UTC 2020 armv7l GNU/Linux
```

Some excerpts from `dimes`:

```
[    0.193535] OMAP GPIO hardware version 0.1
[    0.244209] omap_i2c 44e0b000.i2c: could not find pctldev for node /ocp/l4_wkup@44c00000/scm@210000/pinmux@800/pinmux_i2c0_pins, deferring probe
[    0.245647] omap_i2c 4819c000.i2c: bus 2 rev0.11 at 100 kHz
[    0.528596] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    0.529724] gpio-of-helper ocp:cape-universal: ready
[    0.532244] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
[    0.533842] omap_uart 44e09000.serial: no wakeirq for uart0
[    0.534080] 44e09000.serial: ttyO0 at MMIO 0x44e09000 (irq = 30, base_baud = 3000000) is a OMAP UART0
[    1.312619] random: fast init done
[    1.331649] console [ttyO0] enabled
[    1.336218] omap_uart 481a6000.serial: no wakeirq for uart3
[    1.342296] 481a6000.serial: ttyO3 at MMIO 0x481a6000 (irq = 31, base_baud = 3000000) is a OMAP UART3
[    1.353912] omap_rng 48310000.rng: Random Number Generator ver. 20
[    1.422841] CAN device driver interface
[    1.547984] omap_rtc 44e3e000.rtc: registered as rtc0
[    1.554437] i2c /dev entries driver
[    1.560946] omap_wdt: OMAP Watchdog Timer Rev 0x01: initial timeout 60 sec
[    1.568670] softdog: initialized. soft_noboot=0 soft_margin=60 sec soft_panic=0 (nowayout=0)
[    1.578726] cpuidle: enable-method property 'ti,am3352' found operations
```

What else can I  determine from the system?

Thank you,

A.

RobertCNelson, 2020-05-12:
Awesome! balenaOS are still using kernel overlays! We gave up on that for beagleboard.org, it's too un-reliable, supporting kernel overlays just ends up being a waste of developer resources...

After the overlay loads, do you get the iio adc driver on that system?

```
debian@bbb-pwr01-ser09:~$ ls -lha /sys/bus/iio/devices/
total 0
drwxr-xr-x 2 root root 0 May 12 03:01 .
drwxr-xr-x 4 root root 0 May 11 17:23 ..
lrwxrwxrwx 1 root root 0 May 11 17:23 iio:device0 -> ../../../devices/platform/ocp/44e0d000.tscadc/44e0d000.tscadc:adc.0.auto/iio:device0
```

Regards,

crbn60, 2020-05-12:
```
# ls -lha /sys/bus/iio/devices/
total 0
drwxr-xr-x 2 root root 0 May 12 02:48 .
drwxr-xr-x 4 root root 0 May 12 02:48 ..
```

Doesn't look promising.

crbn60, 2020-05-12:
Some more digging makes me thing that my issue is most likely system level and not with the Python library.

What can I do to verify the system setup, before further troubleshooting at a higher level?

This page indicates that further setup is required:

https://www.balena.io/docs/reference/OS/advanced/#setting-device-tree-overlays-dtoverlay-and-parameters-dtparam

crbn60, 2020-05-12:
I added the following directly to the `/mnt/boot/uEnv.txt` file on my development device:

```
enable_uboot_overlays=1
disable_uboot_overlay_video=1
```

And `ADC.setup()` now succeeds. On to the next challenge.

Thanks for your help.

closed 2020-05-12 by crbn60