adafruit/adafruit-beaglebone-io-python #335: fails to compile using gcc 8.3 (default compiler on latest bbb images) due to -Werror

mvduin, 2020-05-28:
On the latest beaglebone images, Adafruit_BBIO fails to compile because it uses -Werror and gcc has added a bunch of new warnings.

The simple fix is removing -Werror.  It's a bad idea to use -Werror by default in open source projects exactly because a newer compiler version may add new warnings and as a result break compilation.

Obviously it would also be a good idea to examine the warnings and either fix the problem (if there is one) or suppress the warning (if there isn't).

pdp7, 2020-05-28:
@mvduin thank you, I will take a look at this

pdp7, 2020-07-28:
Note: this is how @RobertCNelson is handling it on the current Buster images:
https://github.com/RobertCNelson/omap-image-builder/blob/master/target/chroot/beagleboard.org-buster.sh#L175

mvduin, 2020-07-28:
I'm deeply puzzled why this issue has remained unfixed for long enough to require workaround by rcn, especially since fixing it would take about 20 seconds.  Just remove -Werror.  I strongly disagree with rcn's workaround: you should non't suppress warnings using -Wno-something unless you know for certain that all warnings in that category are harmless and safe to ignore.  (This is definitely _not_ the case for all of them, also not for -Wno-strict-aliasing as I mentioned in #336)

mvduin, 2020-07-28:
(Fixing the issues for which you're getting warnings (or suppressing them when appropriate) would of course also be nice, but obviously has significantly lower priority than fixing the fact that right now it simply **does not compile**.)

mvduin, 2020-07-28:
Ah it seems you fixed the issue but didn't close it?

pdp7, 2020-07-29:
@mvduin Thanks for opening this issue.  You are quite right that I should have setup.py months ago.

pdp7, 2020-07-29:
This should be resolved with https://github.com/adafruit/adafruit-beaglebone-io-python/commit/24a617aecf7cee89d0b62452c86b899e2c02e51b
```
debian@beaglebone:~/adafruit-beaglebone-io-python$ cat /etc/dogtag 
BeagleBoard.org Debian Buster IoT Image 2020-04-06
debian@beaglebone:~/adafruit-beaglebone-io-python$ cat /etc/debian_version 
10.3
debian@beaglebone:~/adafruit-beaglebone-io-python$ sudo python3 ./setup.py install
running install
running bdist_egg
running egg_info
writing Adafruit_BBIO.egg-info/PKG-INFO
writing dependency_links to Adafruit_BBIO.egg-info/dependency_links.txt
writing top-level names to Adafruit_BBIO.egg-info/top_level.txt
reading manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/overlays
copying build/lib.linux-armv7l-3.7/overlays/__init__.py -> build/bdist.linux-armv7l/egg/overlays
copying build/lib.linux-armv7l-3.7/overlays/builder.py -> build/bdist.linux-armv7l/egg/overlays
creating build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/SPI.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/GPIO.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/__init__.py -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/PWM.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/ADC.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/UART.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/Encoder.py -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/sysfs.py -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_I2C.py -> build/bdist.linux-armv7l/egg
byte-compiling build/bdist.linux-armv7l/egg/overlays/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/overlays/builder.py to builder.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/Encoder.py to Encoder.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/sysfs.py to sysfs.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_I2C.py to Adafruit_I2C.cpython-37.pyc
creating stub loader for Adafruit_BBIO/GPIO.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/PWM.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/ADC.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/SPI.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/UART.cpython-37m-arm-linux-gnueabihf.so
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/GPIO.py to GPIO.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/PWM.py to PWM.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/ADC.py to ADC.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/SPI.py to SPI.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/UART.py to UART.cpython-37.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
writing build/bdist.linux-armv7l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
Adafruit_BBIO.__pycache__.ADC.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.GPIO.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.PWM.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.SPI.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.UART.cpython-37: module references __file__
creating 'dist/Adafruit_BBIO-1.1.1-py3.7-linux-armv7l.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing Adafruit_BBIO-1.1.1-py3.7-linux-armv7l.egg
removing '/usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.1.1-py3.7-linux-armv7l.egg' (and everything under it)
creating /usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.1.1-py3.7-linux-armv7l.egg
Extracting Adafruit_BBIO-1.1.1-py3.7-linux-armv7l.egg to /usr/local/lib/python3.7/dist-packages
Adafruit-BBIO 1.1.1 is already the active version in easy-install.pth

Installed /usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.1.1-py3.7-linux-armv7l.egg
Processing dependencies for Adafruit-BBIO==1.1.1
Finished processing dependencies for Adafruit-BBIO==1.1.1
debian@beaglebone:~/adafruit-beaglebone-io-python$ gcc --version
gcc (Debian 8.3.0-6) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

closed 2020-07-29 by pdp7