adafruit/adafruit-beaglebone-io-python #336: Using -Wno-strict-aliasing is a very bad idea.

mvduin, 2020-05-28:
`-Wno-strict-aliasing` suppresses warnings that indicate undefined behaviour (due to violating strict aliasing rules), which may cause miscompilation and crashes.

It's fine if you don't want to deal with the strict aliasing rules (I don't either), but then you should use `-fno-strict-aliasing` (which allows code to simply ignore strict aliasing rules without causing undefined behaviour) instead of `-Wno-strict-aliasing`.

pdp7, 2020-05-28:
Thanks for letting me know about this. I will try out the ` -fno-strict-aliasing` option

pdp7, 2020-07-29:
Now that -Werror is gone, I decided to just have -Wall in CFLAGs.  Thanks for poking me @mvduin.  I should have made the changes long ago.

closed 2020-07-29 by pdp7