adafruit/adafruit-beaglebone-io-python #338: Push new release to PyPI

sambristow-syrp, 2020-07-15:
The most recent version on PyPI is from Nov 20, 2018. 

Some of the fixes that have been merged in the last 18 months are quite important - particularly related to #308 which will prevent users from installing from PIP on more recent systems.

**System Details:**
- Buildroot 2020.05
- Linux 5.6.16
- Python 3.8.2
- GCC 9.3.0 (arm-buildroot-linux-gnueabihf-gcc)

Trying to install from pypi fails with the following output:
```
...
<redacted>/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Os -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Os -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -fPIC -DBBBVERSION41 -Isource/include/ -I<redacted>/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/usr/include/python3.8 -c source/py_gpio.c -o build/temp.linux-x86_64-3.8/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
source/py_gpio.c:558:14: error: cast between incompatible function types from 'PyObject * (*)(PyObject *, PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *, struct _object *)'} to 'PyObject * (*)(PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *)'} [-Werror=cast-function-type]
  558 |    {"setup", (PyCFunction)py_setup_channel, METH_VARARGS | METH_KEYWORDS, "Set up the GPIO channel, direction and (optional) pull/up down control\nchannel        - Either: RPi board pin number (not BCM GPIO 00..nn number).  Pins start from 1\n                 or    : BCM GPIO number\ndirection      - INPUT or OUTPUT\n[pull_up_down] - PUD_OFF (default), PUD_UP or PUD_DOWN\n[initial]      - Initial value for an output channel\n[delay]        - Time in milliseconds to wait after exporting gpio pin"},
      |              ^
source/py_gpio.c:562:25: error: cast between incompatible function types from 'PyObject * (*)(PyObject *, PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *, struct _object *)'} to 'PyObject * (*)(PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *)'} [-Werror=cast-function-type]
  562 |    {"add_event_detect", (PyCFunction)py_add_event_detect, METH_VARARGS | METH_KEYWORDS, "Enable edge detection events for a particular GPIO channel.\nchannel      - either board pin number or BCM number depending on which mode is set.\nedge         - RISING, FALLING or BOTH\n[callback]   - A callback function for the event (optional)\n[bouncetime] - Switch bounce timeout in ms for callback"},
      |                         ^
source/py_gpio.c:565:27: error: cast between incompatible function types from 'PyObject * (*)(PyObject *, PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *, struct _object *)'} to 'PyObject * (*)(PyObject *, PyObject *)' {aka 'struct _object * (*)(struct _object *, struct _object *)'} [-Werror=cast-function-type]
  565 |    {"add_event_callback", (PyCFunction)py_add_event_callback, METH_VARARGS | METH_KEYWORDS, "Add a callback for an event already defined using add_event_detect()\ngpio         - gpio channel\ncallback     - a callback function\n[bouncetime] - Switch bounce timeout in ms"},
      |                           ^
cc1: all warnings being treated as errors
error: command '<redacted>/buildroot/output/host/bin/arm-buildroot-linux-gnueabihf-gcc' failed with exit status 1
...
```

pdp7, 2020-07-17:
Thanks, that is a good point.  I'll get a new release together and push it out.

pdp7, 2020-07-31:
I have published the new v1.2.0 release to PyPI:
https://pypi.org/project/Adafruit-BBIO/

closed 2020-07-31 by pdp7