adafruit/adafruit-beaglebone-io-python #339: Is support for more recent version of Debian (10.3) planned any time soon?

hurak, 2020-07-18:
I am aware that the currently recommended image is Debian 9.9 "Stretch" IoT (2019-08-03) and that Adafruit_BBIO supports Linux kernels 3.8 through 4.14, but I still wanted to ask: is support for more recent version of Debian (10.3) planned any time soon?

With the [currently latest recommended (for BB) Debian 10.3 (kernel 4.19.94-ti-r42)](https://beagleboard.org/latest-images), the Adafruit_BBIO package fails to install.
```
debian@beaglebone:~$ sudo pip install Adafruit_BBIO   
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting Adafruit_BBIO
  Using cached https://files.pythonhosted.org/packages/53/2b/b0e3dce6113225aae9beb886b2addd4fd5c140ba93c9503d7e4a97021bcc/Adafruit_BBIO-1.1.1.tar.gz
Building wheels for collected packages: Adafruit-BBIO
  Running setup.py bdist_wheel for Adafruit-BBIO ... error
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-BYXchU/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-t4YOhS --python-tag cp27:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-2.7
  copying Adafruit_I2C.py -> build/lib.linux-armv7l-2.7
  creating build/lib.linux-armv7l-2.7/Adafruit_BBIO
  copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
  copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
  copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
  creating build/lib.linux-armv7l-2.7/overlays
  copying overlays/builder.py -> build/lib.linux-armv7l-2.7/overlays
  copying overlays/__init__.py -> build/lib.linux-armv7l-2.7/overlays
  running build_ext
  building 'Adafruit_BBIO.GPIO' extension
  creating build/temp.linux-armv7l-2.7
  creating build/temp.linux-armv7l-2.7/source
  arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-RT6aMn/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
  source/py_gpio.c:558:14: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
      {"setup", (PyCFunction)py_setup_channel, METH_VARARGS | METH_KEYWORDS, "Set up the GPIO channel, direction and (optional) pull/up down control\nchannel        - Either: RPi board pin number (not BCM GPIO 00..nn number).  Pins start from 1\n                 or    : BCM GPIO number\ndirection      - INPUT or OUTPUT\n[pull_up_down] - PUD_OFF (default), PUD_UP or PUD_DOWN\n[initial]      - Initial value for an output channel\n[delay]        - Time in milliseconds to wait after exporting gpio pin"},
                ^
  source/py_gpio.c:562:25: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
      {"add_event_detect", (PyCFunction)py_add_event_detect, METH_VARARGS | METH_KEYWORDS, "Enable edge detection events for a particular GPIO channel.\nchannel      - either board pin number or BCM number depending on which mode is set.\nedge         - RISING, FALLING or BOTH\n[callback]   - A callback function for the event (optional)\n[bouncetime] - Switch bounce timeout in ms for callback"},
                           ^
  source/py_gpio.c:565:27: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
      {"add_event_callback", (PyCFunction)py_add_event_callback, METH_VARARGS | METH_KEYWORDS, "Add a callback for an event already defined using add_event_detect()\ngpio         - gpio channel\ncallback     - a callback function\n[bouncetime] - Switch bounce timeout in ms"},
                             ^
  cc1: all warnings being treated as errors
  error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for Adafruit-BBIO
  Running setup.py clean for Adafruit-BBIO
Failed to build Adafruit-BBIO
Installing collected packages: Adafruit-BBIO
  Running setup.py install for Adafruit-BBIO ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-BYXchU/Adafruit-BBIO/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-BMDqSz/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    copying Adafruit_I2C.py -> build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/Adafruit_BBIO
    copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
    copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
    copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-2.7/Adafruit_BBIO
    creating build/lib.linux-armv7l-2.7/overlays
    copying overlays/builder.py -> build/lib.linux-armv7l-2.7/overlays
    copying overlays/__init__.py -> build/lib.linux-armv7l-2.7/overlays
    running build_ext
    building 'Adafruit_BBIO.GPIO' extension
    creating build/temp.linux-armv7l-2.7
    creating build/temp.linux-armv7l-2.7/source
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-RT6aMn/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv7l-2.7/source/py_gpio.o -Wall -Werror -Wextra -Wno-missing-field-initializers -Wno-strict-aliasing
    source/py_gpio.c:558:14: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
        {"setup", (PyCFunction)py_setup_channel, METH_VARARGS | METH_KEYWORDS, "Set up the GPIO channel, direction and (optional) pull/up down control\nchannel        - Either: RPi board pin number (not BCM GPIO 00..nn number).  Pins start from 1\n                 or    : BCM GPIO number\ndirection      - INPUT or OUTPUT\n[pull_up_down] - PUD_OFF (default), PUD_UP or PUD_DOWN\n[initial]      - Initial value for an output channel\n[delay]        - Time in milliseconds to wait after exporting gpio pin"},
                  ^
    source/py_gpio.c:562:25: error: cast between incompatible function types from ‘PyObject * (*)(PyObject *, PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *, struct _object *)’} to ‘PyObject * (*)(PyObject *, PyObject *)’ {aka ‘struct _object * (*)(struct _object *, struct _object *)’} [-Werror=cast-function-type]
[... 35 lines cut ...]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
[... cut ...]
[    0.909339] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.114193] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.125596] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.133182] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

pdp7, 2020-07-19:
@hurak It is my intention for it to work on the current Debian 10 image.  I should update the README.  I will take a look at the install error.

But, also, Adafruit_BBIO should already installed by default on the BeagleBoard.org Debian images that @RobertCNelson creates.  Can you try using the library?

askpatrickw, 2020-07-24:
I've also encountered this same issue installing Adafruit_BBIO in a virtualenv on BBBW.
[error.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/4973916/error.txt)

Is there a known good set of steps installing Adafruit_BBIO on the BBBW?
Thanks!

mvduin, 2020-07-28:
If Adafruit_BBIO is already installed on the system (like it is by default on the official images) then you can just create your venv with something like `python3 -m venv --system-site-packages path/to/your/venv` to give your venv access to system-wide installed packages.

Alternatively, you can install the git version, where the issue has been fixed, e.g.:
```bash
python3 -m venv my-venv
cd my-venv
source bin/activate
pip3 install wheel
git clone https://github.com/adafruit/adafruit-beaglebone-io-python
pip3 install ./adafruit-beaglebone-io-python/
```

pdp7, 2020-07-29:
@askpatrickw are you still having an issue installing?

I tried to reproduce with "BeagleBoard.org Debian Buster IoT Image 2020-04-06" which is Buster 10.3, but I did not run into problems.

I noticed in the original issue posting at the top you were not using python3.  Please make sure you are using pip3 and python3.

askpatrickw, 2020-07-29:
I switched to my Raspberry pi...

pdp7, 2020-07-29:
@hurak I believe the problem was:
`sudo pip install Adafruit_BBIO`
should instead be:
`sudo pip3 install Adafruit_BBIO`

I've update the readme:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/README.md

pdp7, 2020-07-29:
@askpatrickw Sorry for the issues you encountered.  It should be working fine on the current April 2020 image with Buster 10.3.  If you do try again, please let me know if you run into any problems.

I've update readme:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/README.md

mucolon, 2020-07-29:
@hurak @askpatrickw I was getting the same error too. Unfortunately, I forgot how I fixed it exactly but these are the packages I install now before activating my virtual environment. Also I created my virtual environment with 
`python3 -m venv --system-site-packages ~/path/venv`

`sudo apt install build-essential python3 python3-dev python3-setuptools python3-pip python3-venv libssl-dev libffi-dev -y`
&
`python3 -m pip install --user --upgrade setuptools`

I hope this helps.

pdp7, 2020-07-29:
@mucolon 
Thanks for the comment.  I've updated the readme instructions to use Python3.

If anyone runs into problems, then please let me know in the comments here.

closed 2020-07-29 by pdp7