adafruit/adafruit-beaglebone-io-python #341: Adafruit_BBIO.SPI not working with rt kernel

pwhudson11, 2020-08-27:
@RobertCNelson @pdp7 

I've been writing a python library for beaglebone devices to communicate with a ti chip. I was first testing it on a vanilla pocketbeagle, and we recently moved onto a Beaglebone Black. I didn't have any major issues besides the response time to GPIO interrupts on those, so I installed a different kernel: https://github.com/RobertCNelson/ti-linux-kernel-dev/tree/ti-linux-rt-4.19.y.

My Python code still runs without errors, but shows nothing coming from the SPI connection. After monitoring the SPI lines with an oscilloscope, it's clear that it never sends anything over the SPI lines -- everything is held low the whole time.

For wiring I have everything connected to SPI0 and have double checked this.

To set up the SPI connection I have:

```python
def __init__(self, device_num, DReady_Pin):
        self.spi = SPI.SPI(0,device_num) # initialize SPI
        GPIO.setup(DReady_Pin, GPIO.IN) # Set the DRDY pin specified by the user set to input
        GPIO.add_event_detect(DReady_Pin, GPIO.FALLING, self.__on_data_ready)   # Set an interrupt for when data is ready from the device
        self.drdy = DReady_Pin
        self.spi.msh = 1000000   # Set SPI speed
        self.spi.mode = 0b01    # set SPI mode
```

To give you  a clear understanding of how I'm reading and writing as well, here are my basic read/write functions:

```python
# Private method to send the bytes over SPI
    # takes a list of bytes
    def __send_command(self, byte_buffer):
        self.spi.xfer2(byte_buffer)

    # Private method to write a value to a register
    def __write_reg(self, address, values, n=0x01):
        # for the WREG commands, the byte command is 01rr rrrr 00nn nnnn xxxxxxxx...(bytes to be written)
        byte1 = 0x40|address    # = 01rr rrrr
        byte2 = n   # = 00nn nnnn
        byte_buffer = [byte1, byte2] + values   # make a list of the bytes
        self.__send_command(byte_buffer)

 # method to start continuous reading
    def __on_data_ready(self,e):        
        if GPIO.input(self.drdy):
            return
        bytes_in = self.spi.readbytes(27)
        self.__decode_recv_bytes(bytes_in)

```

pwhudson11, 2020-08-27:
Output of  sudo /opt/scripts/tools/version.sh:

git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT00C04319BBBK11C3]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-rt-r50]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   64.057428] remoteproc remoteproc0: wkup_m3 is available
[   64.080562] remoteproc remoteproc0: powering up wkup_m3
[   64.080598] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   64.080899] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   69.264505] remoteproc remoteproc1: 4a334000.pru is available
[   69.271936] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   69.264505] remoteproc remoteproc1: 4a334000.pru is available
[   69.264740] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   69.271936] remoteproc remoteproc2: 4a338000.pru is available
[   69.272126] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.687282] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.706489] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

RobertCNelson, 2020-08-27:
So the PocketBeagle defaults the spi pins to "spi" whereas the BeagleBone defaults everything to 'gpio' use config-pin to switch the pin's from gpio mode to spi mode..

Regards,

pwhudson11, 2020-08-27:
Thank you, that's solved the issue of not reading anything on the bus, but it still appears that the board isn't responding to gpio interrupts (they occur on 1ms increments) in a timely manner. Would you happen to know a way to solve this?

pdp7, 2020-08-30:
> Thank you, that's solved the issue of not reading anything on the bus, but it still appears that the board isn't responding to gpio interrupts (they occur on 1ms increments) in a timely manner. Would you happen to know a way to solve this?

Does this still occur when running the regular non-rt kernel?

pdp7, 2020-09-29:
Please re-open if still issue

closed 2020-09-29 by pdp7