adafruit/adafruit-beaglebone-io-python #342: BeagleBone Green - "RuntimeError: Unknown error" python PWM.start

deelerke, 2020-08-29:
Hi, I'm trying to get this working on a Beaglebone Green with Debian Buster (Linux beaglebone 5.4.38-ti-r8)

```
root@beaglebone:~# python pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 2, in <module>
    pwm.start("P9_14",50);
RuntimeError: Unknown error

root@beaglebone:~# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_14",50);
pwm.stop("P9_14");

root@beaglebone:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[ef16ffa8c27225005cc2b2c974a83da3dee6c1ef]
eeprom:[A335BNLTBBG1BBG217110074]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[BeagleBoard.org Debian Buster Console Image 2020-05-18]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[M-BB-BBG-00A0]
kernel:[5.4.38-ti-r8]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~buster+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev gpio pwm eqep remoteproc admin spi iio i2c docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    2.534070] remoteproc remoteproc0: wkup_m3 is available
[    2.757005] remoteproc remoteproc1: 4a334000.pru is available
[    2.758730] remoteproc remoteproc2: 4a338000.pru is available
[    2.763043] remoteproc remoteproc0: powering up wkup_m3
[    2.763202] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.767376] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    2.757005] remoteproc remoteproc1: 4a334000.pru is available
[    2.757173] pru-rproc 4a334000.pru: PRU rproc node /ocp/interconnect@4a000000/segment@0/target-module@300000/pruss@0/pru@34000 probed successfully
[    2.758730] remoteproc remoteproc2: 4a338000.pru is available
[    2.758899] pru-rproc 4a338000.pru: PRU rproc node /ocp/interconnect@4a000000/segment@0/target-module@300000/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    1.765188] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.779887] gpio-of-helper ocp:cape-universal: ready
END
```

and this is my /boot/uEnv.txt

```
root@beaglebone:~# cat /boot/uEnv.txt | grep -v '#'

uname_r=5.4.38-ti-r8

enable_uboot_overlays=1
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
enable_uboot_cape_universal=1

cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

```

thanks in advance for your suggestions

Dieter

pdp7, 2020-08-30:
@deelerke please trying running the script with python3:
`python3 pwmtest.py`

Also please paste the contenxt of pwmtest.py.

deelerke, 2020-08-31:
hi pdp7 ... 

```
root@beaglebone:~# python3 pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 2, in <module>
    pwm.start("P9_14",50);
RuntimeError: Unknown error
root@beaglebone:~# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_14",50);
pwm.stop("P9_14");
root@beaglebone:~#
```

thank you

pdp7, 2020-08-31:
@deelerke Please try running with strace so I can see what issues may be happening
```
sudo strace -f -o /tmp/strace-log.txt python3 ./pwmtest.py
```
and attach a text file with output.

If strace does not exist, install it with:
```
sudo apt-get install strace
```

deelerke, 2020-08-31:
hi ! 
I've attached the strace output.

[strace-log.txt](https://github.com/adafruit/adafruit-beaglebone-io-python/files/5149913/strace-log.txt)

thank you

pdp7, 2020-08-31:
@deelerke thank you, unfortunately,  I don't see anything that points to the issue in the strace.  Could you please try doing a manual installation from this GitHub repo and trying again?

These are the instructions:
https://github.com/adafruit/adafruit-beaglebone-io-python#installation-on-debian
```
sudo apt-get update
sudo apt-get install build-essential python3-dev python3-pip -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python3 setup.py install
```

deelerke, 2020-09-01:
hey pdp7, thanks for the instructions ... wish I had some good news, but unfortunately the results are the same

fyi, here's the output from the commands

```
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.6).
python3-pip is already the newest version (18.1-5).
python3-dev is already the newest version (3.7.3-1).
python3-dev set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@beaglebone:~# git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
Cloning into 'adafruit-beaglebone-io-python'...
remote: Enumerating objects: 69, done.
remote: Counting objects: 100% (69/69), done.
remote: Compressing objects: 100% (47/47), done.
remote: Total 2429 (delta 42), reused 36 (delta 22), pack-reused 2360
Receiving objects: 100% (2429/2429), 612.15 KiB | 1.46 MiB/s, done.
Resolving deltas: 100% (1567/1567), done.
root@beaglebone:~# cd adafruit-beaglebone-io-python/
root@beaglebone:~/adafruit-beaglebone-io-python# python3 setup.py install
running install
running bdist_egg
running egg_info
creating Adafruit_BBIO.egg-info
writing Adafruit_BBIO.egg-info/PKG-INFO
writing dependency_links to Adafruit_BBIO.egg-info/dependency_links.txt
writing top-level names to Adafruit_BBIO.egg-info/top_level.txt
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-3.7
copying Adafruit_I2C.py -> build/lib.linux-armv7l-3.7
creating build/lib.linux-armv7l-3.7/Adafruit_BBIO
copying Adafruit_BBIO/__init__.py -> build/lib.linux-armv7l-3.7/Adafruit_BBIO
copying Adafruit_BBIO/Encoder.py -> build/lib.linux-armv7l-3.7/Adafruit_BBIO
copying Adafruit_BBIO/sysfs.py -> build/lib.linux-armv7l-3.7/Adafruit_BBIO
creating build/lib.linux-armv7l-3.7/overlays
copying overlays/__init__.py -> build/lib.linux-armv7l-3.7/overlays
copying overlays/builder.py -> build/lib.linux-armv7l-3.7/overlays
running build_ext
building 'Adafruit_BBIO.GPIO' extension
creating build/temp.linux-armv7l-3.7
creating build/temp.linux-armv7l-3.7/source
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=forma                                                                                                  t-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c so                                                                                                  urce/py_gpio.c -o build/temp.linux-armv7l-3.7/source/py_gpio.o -Wall
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=forma                                                                                                  t-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c so                                                                                                  urce/event_gpio.c -o build/temp.linux-armv7l-3.7/source/event_gpio.o -Wall
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=forma                                                                                                  t-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c so                                                                                                  urce/c_pinmux.c -o build/temp.linux-armv7l-3.7/source/c_pinmux.o -Wall
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c source/constants.c -o build/temp.linux-armv7l-3.7/source/constants.o -Wall
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c source/common.c -o build/temp.linux-armv7l-3.7/source/common.o -Wall
arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.7/source/py_gpio.o build/temp.linux-armv7l-3.7/source/event_gpio.o build/temp.linux-armv7l-3.7/source/c_pinmux.o build/temp.linux-armv7l-3.7/source/constants.o build/temp.linux-armv7l-3.7/source/common.o -o build/lib.linux-armv7l-3.7/Adafruit_BBIO/GPIO.cpython-37m-arm-linux-gnueabihf.so
building 'Adafruit_BBIO.PWM' extension
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c source/py_pwm.c -o build/temp.linux-armv7l-3.7/source/py_pwm.o -Wall
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DBBBVERSION41 -Isource/include/ -I/usr/include/python3.7m -c sour [... cut ...]
Installed /usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg
Processing dependencies for Adafruit-BBIO==1.2.0
Finished processing dependencies for Adafruit-BBIO==1.2.0
root@beaglebone:~/adafruit-beaglebone-io-python# python3 pwmtest.py
python3: can't open file 'pwmtest.py': [Errno 2] No such file or directory
root@beaglebone:~/adafruit-beaglebone-io-python# cd ..
root@beaglebone:~# python3 pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 2, in <module>
    pwm.start("P9_14",50);
RuntimeError: Unknown error
root@beaglebone:~#
````

pdp7, 2020-09-01:
@deelerke Thank for doing that.

Would it possible for you start fresh with another microSD card?

If so, this is the most recent testing Debian image: [bone-debian-10.5-iot-armhf-2020-08-25-4gb.img.xz](https://elinux.org/Beagleboard:Latest-images-testing#Debian_10_.28Buster.29_IoT)

deelerke, 2020-09-01:
sure, downloading it now... it's not that far off from the one I've downloaded before: bone-debian-10.4-console-armhf-2020-05-18-1gb.img ... let's see,  keep you posted ....

deelerke, 2020-09-01:
that works !

both python3 and python (2.7) run the pwmtest.py without errors.

I'll stick to this image then !

thanks a lot

pdp7, 2020-09-01:
@deelerke great to hear!

deelerke, 2020-09-01:
thanks pdp7 ... 

The issue seems to return however with a different PWM pin.
First of all, let me demonstrate the older system (hostname:  "lighthub") it works on:

```
root@lighthub:~# uname -na
Linux lighthub 4.14.108-ti-r104 #1 SMP PREEMPT Tue Apr 9 18:16:29 UTC 2019 armv7l GNU/Linux
root@lighthub:~# more /etc/dogtag
rcn-ee.net console Debian Image 2019-04-10
root@lighthub:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[9df8deb6b2ae7570c9102ab58f795d3bed592194]
eeprom:[A335BNLTBBG1BBG217097038]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[rcn-ee.net console Debian Image 2019-04-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.01-00002-g6de8c74429]:[location: dd MBR]
kernel:[4.14.108-ti-r104]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~stretch+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~stretch+20200813]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep admin spi tisdk west                                                        on-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet c                                                        ape_universal=enable]
dmesg | grep remote
[    1.254581] remoteproc remoteproc0: wkup_m3 is available
[    1.343054] remoteproc remoteproc0: powering up wkup_m3
[    1.343172] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.347686] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.928665] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.941168] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

if I do run the pwmtest.py on this system, then it works:

```
root@lighthub:~# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_22",50);
pwm.start("P9_42",50);

root@lighthub:~# python pwmtest.py
root@lighthub:~#
```

so all good no errors for both p9_22 and p9_42

But.... running this on my brand new image: 

```
root@beaglebone:~# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_22",50);
pwm.start("P9_42",50);

root@beaglebone:~# python pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 3, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
root@beaglebone:~#
```

the dreaded error is back :(

I must be missing something here ...

deelerke, 2020-09-01:
so in a nutshell: 
on the new image with kernel 4.19.94-ti-r50   :  P9_14 seems ok , so does P9_22 ... but not P9_45
on the older image with kernel 4.14.108-ti-r104 :  all PWM pins work

pdp7, 2020-09-02:
@deelerke please paste `/boot/uEnv.txt` as it is possible that some of those pins are being occuppied by settings in device tree overlay (such as for HDMI or eMMC): https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Disable_on-board_devices

Also, please run this for those pins:
```
config-pin -q P9.22
config-pin -q P9.42
sudo config-pin P9.22
sudo config-pin P9.42
```

deelerke, 2020-09-07:
hi , sorry missed your reply there, hence the late reaction:

this is the uEnv.txt
```
root@beaglebone:~# cat /boot/uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.19.94-ti-r50
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
###Overide capes with eeprom
#uboot_overlay_addr0=<file0>.dtbo
#uboot_overlay_addr1=<file1>.dtbo
#uboot_overlay_addr2=<file2>.dtbo
#uboot_overlay_addr3=<file3>.dtbo
###
###Additional custom capes
#uboot_overlay_addr4=<file4>.dtbo
#uboot_overlay_addr5=<file5>.dtbo
#uboot_overlay_addr6=<file6>.dtbo
#uboot_overlay_addr7=<file7>.dtbo
###
###Custom Cape
#dtb_overlay=<file8>.dtbo
###
###Disable auto loading of virtual capes (emmc/video/wireless/adc)
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###
###PRUSS OPTIONS
###pru_rproc (4.14.x-ti kernel)
#uboot_overlay_pru=AM335X-PRU-RPROC-4-14-TI-00A0.dtbo
###pru_rproc (4.19.x-ti kernel)
uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
###pru_uio (4.14.x-ti, 4.19.x-ti & mainline/bone kernel)
#uboot_overlay_pru=AM335X-PRU-UIO-00A0.dtbo
###
###Cape Universal Enable
enable_uboot_cape_universal=1
###
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
###
###U-Boot fdt tweaks... (60000 = 384KB)
#uboot_fdt_buffer=0x60000
###U-Boot Overlays###

cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

#In the event of edid real failures, uncomment this next line:
#cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet video=HDMI-A-1:1024x768@60e

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

```

and the config commands:

```
root@beaglebone:~# config-pin -q P9.22

Current mode for P9_22 is:     default

root@beaglebone:~# config-pin -q P9.42

Current mode for P9_42 is:     default

root@beaglebone:~# python pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 3, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error

root@beaglebone:~# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_22",50);
pwm.start("P9_42",50);

```

pdp7, 2020-09-07:
Apologies, I truncated my commands.  The last two should have been:
```
sudo config-pin P9.22 pwm
sudo config-pin P9.42 pwm
```
This should put the pins into pwm mode.

Could you retry running the script after that?

Also, please paste the output of this command so I can see the state of the Linux pwm driver:
```
sudo find /sys/devices/ -type f | grep pwm 
```

deelerke, 2020-09-07:
no problem ! 
this is the output:

```
Last login: Mon Sep  7 07:39:24 2020 from 192.168.1.9
root@beaglebone:~# config-pin P9.22 pwm

Current mode for P9_22 is:     pwm

root@beaglebone:~# config-pin P9.42 pwm

Current mode for P9_42 is:     pwm

root@beaglebone:~# python pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 3, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
root@beaglebone:~# find /sys/devices/ -type f | grep pwm
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/period
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/runtime_active_kids
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/runtime_suspended_time
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/autosuspend_delay_ms
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/runtime_enabled
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/runtime_active_time
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/control
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/async
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/runtime_usage
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/power/runtime_status
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/duty_cycle
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/enable
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/capture
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/polarity
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/uevent
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/export
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/unexport
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/runtime_active_kids
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/runtime_suspended_time
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/autosuspend_delay_ms
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/runtime_enabled
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/runtime_active_time
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/control
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/async
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/runtime_usage
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/power/runtime_status
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/uevent
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/npwm
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/runtime_active_kids
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/runtime_suspended_time
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/autosuspend_delay_ms
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/runtime_enabled
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/runtime_active_time
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/control
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/async
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/runtime_usage
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/power/runtime_status
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/driver_override
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/modalias
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/uevent
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/export
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/unexport
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/runtime_active_kids
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/runtime_suspended_time
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/autosuspend_delay_ms
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/runtime_enabled
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/runtime_active_time
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/control
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/async
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/runtime_usage
/sys/devices/platform/ocp/48300000.epwmss/48300200.pwm/pwm/pwmchip1/power/runtime_status
[... 191 lines cut ...]
/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/power/runtime_usage
/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/power/runtime_status
/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/driver_override
/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/modalias
/sys/devices/platform/ocp/48302000.epwmss/48302200.pwm/uevent
/sys/devices/platform/ocp/48302000.epwmss/uevent
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/period
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/mode
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/enabled
[... cut ...]
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/power/async
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/power/runtime_usage
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/power/runtime_status
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/driver_override
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/modalias
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/position
/sys/devices/platform/ocp/48302000.epwmss/48302180.eqep/uevent
root@beaglebone:~#

```

RobertCNelson, 2020-09-08:
What version of "bb-customizations" do you have installed? that controls the udev rules..

```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.01-00002-g6de8c74429]:[location: dd MBR]
```

That might also be a problem.

deelerke, 2020-09-09:
hi Robert, thanks for your response ... this is the bb-customizations version:

```
root@beaglebone:~# dpkg -l bb-customizations
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name              Version                        Architecture Description
+++-=================-==============================-============-=================================
ii  bb-customizations 1.20200806.0-0~buster+20200806 all          beagleboard.org customizations
```

thanks

deelerke, 2020-09-09:
after this I also did a dist-upgrade, and there was a new version of bb-customizations available.

```
Preparing to unpack .../22-bb-customizations_1.20200826.0-0~buster+20200826_all.deb ...
Unpacking bb-customizations (1.20200826.0-0~buster+20200826) over (1.20200806.0-0~buster+20200806) ...
```

After installing, I rebooted the BBG and tested again:

```
root@beaglebone:~# python pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 3, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error

root@beaglebone:~# dpkg -l bb-customizations
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name              Version                        Architecture Description
+++-=================-==============================-============-=================================
ii  bb-customizations 1.20200826.0-0~buster+20200826 all          beagleboard.org customizations

```

pdp7, 2020-11-30:
@deelerke hello, is this still an issue?

deelerke, 2020-12-01:
hi! yes it is unfortunately. I haven't checked if there are newer debian builds by this time. Anything you can recommend me to try?

thanks!

deelerke, 2020-12-06:
I just tried with bone-debian-10.6-console-armhf-2020-11-30-1gb.img
Linux beaglebone 4.19.94-ti-r57 #1buster SMP PREEMPT Thu Nov 19 23:44:09 UTC 2020 armv7l GNU/Linux
and again

````
Traceback (most recent call last):
  File "lighthub3.py", line 384, in <module>
    for i in list(pwms.keys()): PWM.start(pwms[i]['pin'],0,1800,0)
RuntimeError: Unknown error
````

deelerke, 2021-01-13:
happy new year all :) 
tried with the latest new image: https://rcn-ee.com/rootfs/bb.org/testing/2021-01-11/buster-console/bone-debian-10.7-console-armhf-2021-01-11-1gb.img.xz
But same error .... 

I'm running out of options .... any ideas ? ? ? 

thanks

pdp7, 2021-01-13:
@deelerke sorry about the delay in addressing this.  I will use that new image and try to reproduce tonight.

deelerke, 2021-01-13:
no problem pdp7 👍 thanks
maybe this has some clues: https://e2e.ti.com/support/processors/f/791/t/664369?Linux-AM3358-PWM-signal-capture
They talking about "eCAP" and "control registers in the TRM"....  but to be honest I don't even know what that means :)

pdp7, 2021-01-14:
I just tried with a BeagleBone Black and it did not encounter an error for P9_14.

Version:
```
root@beaglebone:~# cat /etc/issue
Debian GNU/Linux 10 \n \l

BeagleBoard.org Debian Buster IoT Image 2021-01-11

Support: https://bbb.io/debian

default username:password is [debian:temppwd]

root@beaglebone:~# cat /etc/dogtag 
BeagleBoard.org Debian Buster IoT Image 2021-01-11
root@beaglebone:~# uname -a
Linux beaglebone 4.19.94-ti-r57 #1buster SMP PREEMPT Thu Nov 19 23:44:09 UTC 2020 armv7l GNU/Linux

debian@beaglebone:~$ cat /boot/uEnv.txt |grep -v ^#

uname_r=4.19.94-ti-r57

enable_uboot_overlays=1
uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
enable_uboot_cape_universal=1

cmdline=coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet

```

Install:
```
sudo connmantctl #connect to wifi
sudo apt update
sudo apt upgrade
sudo apt-get install build-essential python3-dev python3-pip -y
git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git
cd adafruit-beaglebone-io-python
sudo python3 setup.py install
```

Test:
```
debian@beaglebone:~$ cat pwm_test,py 
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_14",50);
pwm.stop("P9_14");
debian@beaglebone:~$ sudo python3 pwm_test,py
```

pdp7, 2021-01-14:
P_22 is ok as well:
```
debian@beaglebone:~$ cat pwm_test.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_22",50);
pwm.stop("P9_22");

debian@beaglebone:~$ sudo python3 ./pwm_test.py 
```
However, P9_42 fails:
```
debian@beaglebone:~$ cat pwm_test.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_42",50);
pwm.stop("P9_42");

debian@beaglebone:~$ sudo python3 ./pwm_test.py 
Traceback (most recent call last):
  File "./pwm_test.py", line 2, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
```

pdp7, 2021-01-14:
update - I can confirm that P9_42 is supposed to be in pwm mode, but still encounter the error:
```
debian@beaglebone:~$ config-pin P9.42 pwm

Current mode for P9_42 is:     pwm

debian@beaglebone:~$ sudo python3 ./pwm_test.py 
Traceback (most recent call last):
  File "./pwm_test.py", line 2, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
```

pdp7, 2021-01-14:
@deelerke which pins are you having trouble with?  Is it all or just some like P9_42?

deelerke, 2021-01-14:
good morning!
I tested with my home automation python program:

```
# definitions of the PWM outputs on the BeagleBone
pwms = {
        'D_living_deur':        {'pin':'P9_22','state':0},      # PWM dimmers pin=Beaglebone pin  state=lookup table (PWMmap)
        'D_living_midden':      {'pin':'P9_42','state':0},
        'D_living_vide':        {'pin':'P9_14','state':0},
        'D_keuken':             {'pin':'P9_16','state':0},
        'D_slpk3':              {'pin':'P8_46','state':0},
        'D_keuken_led':         {'pin':'P8_45','state':0}
}

# initialize GPIO & PWM
for i in list(pins.keys()): GPIO.setup(pins[i]['pin'],GPIO.OUT,delay=0)
for i in list(pwms.keys()): PWM.start(pwms[i]['pin'],0,1800,0)
```

so I'm not sure where it throws an error
I will reflash the image again (I messed it up trying to upgrade the kernel) and will test with your test program.
Update later today; fingers crossed

deelerke, 2021-01-14:
I'm getting the same as you, tested with and without uboot_overlays for emmc & hdmi

```
root@beaglebone:/home/debian# python3 pwmtest.py
root@beaglebone:/home/debian# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_14",50);
pwm.stop("P9_14");
root@beaglebone:/home/debian# vim pwmtest.py
root@beaglebone:/home/debian# python3 pwmtest.py
root@beaglebone:/home/debian# cat pwmtest.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_22",50);
pwm.stop("P9_22");
root@beaglebone:/home/debian# vim pwmtest.py
root@beaglebone:/home/debian# python3 pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 2, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
root@beaglebone:/home/debian# cat /boot/uEnv.txt | grep disable
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
#disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
root@beaglebone:/home/debian# vim /boot/uEnv.txt
root@beaglebone:/home/debian# reboot
root@beaglebone:/home/debian# Connection to 192.168.1.169 closed by remote host.
Connection to 192.168.1.169 closed.
root@svr:~# ssh debian@192.168.1.169
Debian GNU/Linux 10

BeagleBoard.org Debian Buster Console Image 2021-01-11

Support: https://bbb.io/debian

default username:password is [debian:temppwd]

debian@192.168.1.169's password:
X11 forwarding request failed on channel 0

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jan 14 07:25:26 2021 from 192.168.1.7
debian@beaglebone:~$ sudo su root
[sudo] password for debian:
root@beaglebone:/home/debian# cat /boot/uEnv.txt | grep disable
disable_uboot_overlay_emmc=1
disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###Debug: disable uboot autoload of Cape
#disable_uboot_overlay_addr0=1
#disable_uboot_overlay_addr1=1
#disable_uboot_overlay_addr2=1
#disable_uboot_overlay_addr3=1
root@beaglebone:/home/debian# python3 pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 2, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
root@beaglebone:/home/debian#
```

pdp7, 2021-01-15:
I believe one of the issues with P9_42 is that there are two signal on the BGA ball for it.  We refer to it as P9.42A and P9.42B in the header diagram.  The config-pin utility uses P9_42 and P9_92 respectively.

It does merit me figuring out some solution for pwm on P9.42, but is it possible for you to use a different pin?

deelerke, 2021-01-15:
hi pdp7 ... I'm not sure if I understand it. 
Do you mean that "ECAPPWM0" (P9_42) would also be available on another pin on the expansion header?  ( [https://beagleboard.org/static/images/cape-headers-pwm.png](url) )
On that picture it says 8 PWMs in the title , but showing 14 on the pins (I never fully grasped that concept, but with trial and error I just started using the PWM pins I was able to configure first)
I'm actually using PWM on P9_42 in my home automation (goes to a MOSFET driver and dims my 12V light bulbs)
Like I mentioned in ealier posts, this works with another BeagleBone Green that seems to be able to control that pin fine on an older debian image.
This is the working older image:

```
root@lighthub:~# uname -na
Linux lighthub 4.14.108-ti-r104 #1 SMP PREEMPT Tue Apr 9 18:16:29 UTC 2019 armv7l GNU/Linux
root@lighthub:~# more /etc/dogtag
rcn-ee.net console Debian Image 2019-04-10
root@lighthub:~# more /etc/debian_version
9.13
root@lighthub:~# /opt/scripts/tools/version.sh
git:/opt/scripts/:[9df8deb6b2ae7570c9102ab58f795d3bed592194]
eeprom:[A335BNLTBBG1BBG217097038]
model:[TI_AM335x_BeagleBone_Green]
dogtag:[rcn-ee.net console Debian Image 2019-04-10]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.01-00002-g6de8c74429]:[location: dd MBR]
kernel:[4.14.108-ti-r104]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~stretch+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~stretch+20200813]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 quiet cape_universal=enable]
dmesg | grep remote
[    1.254564] remoteproc remoteproc0: wkup_m3 is available
[    1.339044] remoteproc remoteproc0: powering up wkup_m3
[    1.339165] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.343681] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.928351] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.940684] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
````

RobertCNelson, 2021-01-15:
> Do you mean that "ECAPPWM0" (P9_42) would also be available on another pin on the expansion header?

No, there are two Ball's from the SOC attached to P9_42..

What's the state of P9_92 (P9.42b)?

```
debian@bbb-pwr02-ser10:~$ sudo /opt/scripts/device/bone/show-pins.pl  | grep P9.42 
P9.42a                            89 C18 fast rx down 7 gpio 0.07        ocp/P9_42_pinmux (pinmux_P9_42_default_pin)
P9.42b                           104 B12 fast rx down 7 gpio 3.18        ocp/P9_92_pinmux (pinmux_P9_92_default_pin)
``

deelerke, 2021-01-15:
I think I'm missing  a package to run that command
```
debian@beaglebone:~$ sudo /opt/scripts/device/bone/show-pins.pl  | grep P9.42
[sudo] password for debian:
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
Can't locate Inline/Files.pm in @INC (you may need to install the Inline::Files module) (@INC cont                                                                                             ains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/l                                                                                             ib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/sha                                                                                             re/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /opt/scripts/devi                                                                                             ce/bone/show-pins.pl line 11.
BEGIN failed--compilation aborted at /opt/scripts/device/bone/show-pins.pl line 11.
```

RobertCNelson, 2021-01-15:
Yeah i really need to push an upload an update:

```
sudo apt update ; sudo apt install libinline-files-perl
```

Will add the perl library..

Regards,

deelerke, 2021-01-15:
thank you ! 

```
debian@beaglebone:~$ sudo /opt/scripts/device/bone/show-pins.pl  | grep P9.42
P9.42a                            89 C18 fast rx down 0 eCAP 0           ocp/P9_42_pinmux (pinmux_P9_42_ecap_pwm_pin)
P9.42b                           104 B12 fast rx down 7 gpio 3.18        ocp/P9_92_pinmux (pinmux_P9_92_default_pin)
````

RobertCNelson, 2021-01-15:
So the P9.42b pd is enabled, try moving to gpio_input mode:

```
debian@bbb-pwr02-ser10:~$ config-pin P9.92 gpio_input

Current mode for P9_92 is:     gpio_input

debian@bbb-pwr02-ser10:~$ sudo /opt/scripts/device/bone/show-pins.pl  | grep P9.42 

P9.42a                            89 C18 fast rx down 7 gpio 0.07        ocp/P9_42_pinmux (pinmux_P9_42_default_pin)

P9.42b                           104 B12 fast rx      7 gpio 3.18        ocp/P9_92_pinmux (pinmux_P9_92_gpio_input_pin)
```

deelerke, 2021-01-15:
hmmm ,  doesn't seem to work

```
debian@beaglebone:~$ config-pin P9.92 gpio_input
ERROR: write() to /sys/devices/platform/ocp/ocp:P9_92_pinmux/state failed, No such device
debian@beaglebone:~$ sudo su root
root@beaglebone:/home/debian# config-pin P9.92 gpio_input
ERROR: write() to /sys/devices/platform/ocp/ocp:P9_92_pinmux/state failed, No such device
root@beaglebone:/home/debian# config-pin P9.92 gpio

Current mode for P9_92 is:     gpio

root@beaglebone:/home/debian# python3 pwmtest.py
Traceback (most recent call last):
  File "pwmtest.py", line 2, in <module>
    pwm.start("P9_42",50);
RuntimeError: Unknown error
```

pdp7, 2021-01-15:
@deelerke this does seem to be a tough one.  I don't think there is anything we are aware that would have broke it between 4.14 and 4.19.  @RobertCNelson is that right?

Is the MOSFET hard wired to P9.42?  If not, it might be more expedient to identify another pin that could be used.

deelerke, 2021-01-15:
yeah sure seems a though one !  :-)
The mosfet is hardwired. I'll probably stick to the 4.14 image then.

So I noticed my output of the command was different of @RobertCNelson  : P9.42a  89 C18 fast rx down 0 **eCAP** 0  ocp/P9_42_pinmux ux_P9_42_ecap_pwm_pin)
That "**eCAP**" was also mentioned in the texas intruments url I saw earlier [https://e2e.ti.com/support/processors/f/791/t/664369?Linux-AM3358-PWM-signal-capture](url)
there they also say that the problem is in the eCAP ? - I don't know about it , but maybe the key to this solution is there 

thanks ! ! !

RobertCNelson, 2021-01-15:
Take my comment lightly, i didn't touch p9.42a in my logs.. i was just helping config p9.42b to not conflict..

I'd assume the python script would properly setup p9.42a...

Regards,

deelerke, 2021-01-17:
I'm trying some more things and looking for clues:
I did a strace of the python test program
and this shows:
```
nanosleep({tv_sec=0, tv_nsec=100000000}, NULL) = 0
stat64("/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm0", 0xbeff8668) = -1 ENOENT (No such file or directory)
stat64("/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-p:0", 0xbeff8668) = -1 ENOENT (No such file or directory)
stat64("/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-p:0", 0xbeff8668) = -1 ENOENT (No such file or directory)
gettimeofday({tv_sec=1610893772, tv_usec=46964}, NULL) = 0
getpid()                                = 4352
send(3, "<139>Jan 17 14:29:32 python3[435"..., 90, MSG_NOSIGNAL) = 90
gettimeofday({tv_sec=1610893772, tv_usec=51928}, NULL) = 0
getpid()                                = 4352
send(3, "<139>Jan 17 14:29:32 python3[435"..., 87, MSG_NOSIGNAL) = 87
write(2, "Traceback (most recent call last"..., 35Traceback (most recent call last):
) = 35
write(2, "  File \"pwmtest.py\", line 2, in "..., 41  File "pwmtest.py", line 2, in <module>
) = 41
openat(AT_FDCWD, "pwmtest.py", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 4
fstat64(4, {st_mode=S_IFREG|0644, st_size=74, ...}) = 0
ioctl(4, TCGETS, 0xbeff80e0)            = -1 ENOTTY (Inappropriate ioctl for device)
_llseek(4, 0, [0], SEEK_CUR)            = 0
fcntl64(4, F_DUPFD_CLOEXEC, 0)          = 5
fcntl64(5, F_GETFL)                     = 0x20000 (flags O_RDONLY|O_LARGEFILE)
fstat64(5, {st_mode=S_IFREG|0644, st_size=74, ...}) = 0
read(5, "import Adafruit_BBIO.PWM as pwm\n"..., 4096) = 74
close(5)                                = 0
_llseek(4, 0, [0], SEEK_SET)            = 0
read(4, "import Adafruit_BBIO.PWM as pwm\n"..., 8192) = 74
close(4)                                = 0
write(2, "    pwm.start(\"P9_42\",50);\n", 27    pwm.start("P9_42",50);
) = 27
write(2, "RuntimeError: Unknown error\n", 28RuntimeError: Unknown error
) = 28
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6cdcca1}, {sa_handler=0x1621a9, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0xb6cdcca1}, 8) = 0
sigaltstack(NULL, {ss_sp=0xad1a40, ss_flags=0, ss_size=8192}) = 0
sigaltstack({ss_sp=NULL, ss_flags=SS_DISABLE, ss_size=0}, NULL) = 0
exit_group(1)                           = ?
+++ exited with 1 +++
```

from /sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/
it tries to open
- pwm0
- pwm-p:0
- pwm-p:0

all three fail (no such file or directory)
when I look , there's a directory **pwm-0:0**

```root@beaglebone:/home/debian# ls -la /sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/
total 0
drwxrwxr-x 4 root gpio    0 Jan 14 07:47 .
drwxrwxr-x 3 root gpio    0 Jan 14 07:44 ..
lrwxrwxrwx 1 root gpio    0 Jan 14 07:44 device -> ../../../48300100.ecap
--w--w---- 1 root gpio 4096 Jan 17 14:29 export
-r--r--r-- 1 root gpio 4096 Jan 14 07:44 npwm
drwxrwxr-x 2 root gpio    0 Jan 14 07:44 power
drwxrwxr-x 3 root gpio    0 Jan 14 07:47 pwm-0:0
lrwxrwxrwx 1 root gpio    0 Jan 14 07:44 subsystem -> ../../../../../../../class/pwm
-rw-rw-r-- 1 root gpio 4096 Jan 14 07:44 uevent
--w--w---- 1 root gpio 4096 Jan 14 07:44 unexport
```

Can it be that this just got renamed in kernel >4.14 ?
Another link op StackExchange confirm there were changes in the >4.14 kernel
[https://superuser.com/questions/1436916/system-directory-files-to-access-pwm-pins-are-missing-from-by-beaglebone-black](url)

_Effectively around version 4.14 of the linux kernel, the started changing the hardware overlays for the pwm pin; since then it's gone through minor revisions but has largely stayed the same. What I was using was I was looking at the pre-4.14 version._

I really hope this is useful!

pdp7, 2021-01-24:
@deelerke thanks, it does look like that is a problem, comparing P9_14 (ok) with P9_42 (error).  I'll look into it further.
[image]
[image]

deelerke, 2021-01-24:
thanks in advance! 
if there's anything I can help with or test, let me know

pdp7, 2021-01-24:
I've noticed there is the same issue with P9_28 as with P9_42.  These are both using PWM outputs from ECAP blocks. Here is Device Tree information from:
- https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti-overlays/include/dt-bindings/board/am335x-bone-pins.h
- https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x-ti-overlays/src/arm/am335x-bone-common-univ.dtsi

### P9_28 
```
	#define P9_28(mode) AM33XX_IOPAD(0x099c, mode)  /* C12: mcasp0_ahclkr */

	P9_28_default_pin: pinmux_P9_28_default_pin { pinctrl-single,pins = <
		P9_28( PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE7) >; };	/* mcasp0_ahclkr.gpio3_17 */
	P9_28_pwm_pin: pinmux_P9_28_pwm_pin { pinctrl-single,pins = <
		P9_28( PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE1) >; };	/* mcasp0_ahclkr.ehrpwm0_synci */
	P9_28_ecap_pwm_pin: pinmux_P9_28_ecap_pwm_pin { pinctrl-single,pins = <
		P9_28( PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE4) >; };	/* mcasp0_ahclkr.ecap2_in_pwm2_out */	

	/* P9_28 (ZCZ ball C12) audio */
	P9_28_pinmux {
		compatible = "bone-pinmux-helper";
		status = "okay";
		pinctrl-names = "default", "gpio", "gpio_pu", "gpio_pd", "spi_cs", "pwm", "pwm2", "pruout", "pruin";
		pinctrl-0 = <&P9_28_default_pin>;
		pinctrl-1 = <&P9_28_gpio_pin>;
		pinctrl-2 = <&P9_28_gpio_pu_pin>;
		pinctrl-3 = <&P9_28_gpio_pd_pin>;
		pinctrl-4 = <&P9_28_spi_cs_pin>;
		pinctrl-5 = <&P9_28_pwm_pin>;      /* state: "pwm" */
		pinctrl-6 = <&P9_28_ecap_pwm_pin>; /* state: "pwm2" */
		pinctrl-7 = <&P9_28_pruout_pin>;
		pinctrl-8 = <&P9_28_pruin_pin>;
		pinctrl-9 = <&P9_28_mcasp_pin>;
	};
```
There is already logic for the unusual state name of "pwm2" for P9_28:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c#L370
```

    // Do pinmuxing
    if(!strcmp(key, "P9_28")) {
        // ecap2 (P9_28) requires mode pwm2
        // based on bonescript commit 23bf443 by Matthew West
        strncpy(pin_mode, "pwm2", PIN_MODE_LEN);
    } else {
        strncpy(pin_mode, "pwm", PIN_MODE_LEN);
    }
    set_pin_mode(key, pin_mode);
```

### P9_42
```

	#define P9_42(mode) AM33XX_IOPAD(0x0964, mode) /* C18: P0_in_PWM0_out */
	#define P9_42A(mode) AM33XX_IOPAD(0x0964, mode) /* C18: P0_in_PWM0_out */
	#define P9_42B(mode) AM33XX_IOPAD(0x09a0, mode) /* B12: mcasp0_aclkr */
	#define P9_92(mode) AM33XX_IOPAD(0x09a0, mode) /* B12: mcasp0_aclkr */

	P9_42_default_pin: pinmux_P9_42_default_pin { pinctrl-single,pins = <
		P9_42( PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE7) >; };	/* eCAP0_in_PWM0_out.gpio0_7 */
	P9_42_ecap_pwm_pin: pinmux_P9_42_ecap_pwm_pin { pinctrl-single,pins = <
		P9_42( PIN_OUTPUT_PULLDOWN | INPUT_EN | MUX_MODE0) >; };	/* eCAP0_in_PWM0_out.ecap0_in_pwm0_out */
	
	/* P9_42 (ZCZ ball C18) */
	P9_42_pinmux {
		compatible = "bone-pinmux-helper";
		status = "okay";
		pinctrl-names = "default", "gpio", "gpio_pu", "gpio_pd", "spi_cs", "spi_sclk", "uart", "pwm", "pru_ecap_pwm";
		pinctrl-0 = <&P9_42_default_pin>;-
		pinctrl-1 = <&P9_42_gpio_pin>;
		pinctrl-2 = <&P9_42_gpio_pu_pin>;
		pinctrl-3 = <&P9_42_gpio_pd_pin>;
		pinctrl-4 = <&P9_42_spi_cs_pin>;
		pinctrl-5 = <&P9_42_spi_sclk_pin>;
		pinctrl-6 = <&P9_42_uart_pin>;
		pinctrl-7 = <&P9_42_ecap_pwm_pin>;     /* state: "pwm" */
		pinctrl-8 = <&P9_42_pru_ecap_pwm_pin>; /* state: "pru_ecap_pwm" */
	};
```
Here are the /sys enteries for the ecap pwm devices:
```
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/period
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/duty_cycle
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/enable
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/capture
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/polarity
```
```
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0/period
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0/duty_cycle
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0/enable
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0/capture
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0/polarity
```
Looking at the https://github.com/beagleboard/BeagleBoard-DeviceTrees/blob/v4.19.x/src/arm/am33xx-l4.dtsi

48300100.ecap is ecap0 based on it being at:
```
		target-module@0 {			/* 0x48300000, ap 66 48.0 */
```
and at 0x100 offset:
```
				ecap0: ecap@100 {
```
48304100.ecap is ecap2 based on it being at:
```
		target-module@4000 {			/* 0x48304000, ap 70 44.0 */
```
and at 0x100 offset:
```
				ecap2: ecap@100 {
```

deelerke, 2021-01-24:
thanks again for the update pdp7 !!

yes, i'm seeing those directories too:
/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0/pwm-0:0/
/sys/devices/platform/ocp/48304000.epwmss/48304100.ecap/pwm/pwmchip6/pwm-6:0/

Is that good news? The DTSI stuff is a bit out of my league 😲

pdp7, 2021-01-24:
@deelerke sorry, I should have provided the context that I just wanted to make sure I had that information preserved.  I think there is an issue with how the path is being constructed that I am looking at.

pdp7, 2021-01-25:
The trouble seems to be:
https://github.com/adafruit/adafruit-beaglebone-io-python/blob/11e9d2d82e7c584f2e10decf4d6328273cb47568/source/c_pwm.c#L422
where `pwm_path[66]` is `p` whereas it should be a number.
```
DEBUG pwm_chip_path=/sys/devices/platform/ocp/48300000.epwmss/48300100.ecap/pwm/pwmchip0 dmtimer_pin=0 pwm_path[47]=1 pwm_path[66]=p p->index=0Traceback (most recent call last):
```

pdp7, 2021-01-25:
Yes, that does fix it in my testing.
```diff
diff --git a/source/c_pwm.c b/source/c_pwm.c
index 1756e5f..a5ef5ec 100644
--- a/source/c_pwm.c
+++ b/source/c_pwm.c
@@ -419,7 +419,7 @@ BBIO_err pwm_setup(const char *key, __attribute__ ((unused)) float duty, __attri
     snprintf(pwm_path_udev, sizeof(pwm_path_udev), "%s/pwm-%c:%d", pwm_chip_path, dmtimer_pin ? pwm_path[47] : pwm_path[66], p->index);
     syslog(LOG_DEBUG, "Adafruit_BBIO: pwm_start: key: %s, pwm_path_udev: %s", key, pwm_path_udev);
     //ecap output with udev patch
-    snprintf(ecap_path_udev, sizeof(ecap_path_udev), "%s/pwm-%c:%d", pwm_chip_path, dmtimer_pin ? pwm_path[47] : pwm_path[66], p->index);
+    snprintf(ecap_path_udev, sizeof(ecap_path_udev), "%s/pwm-%c:%d", pwm_chip_path, dmtimer_pin ? pwm_path[47] : pwm_path[67], p->index);
     syslog(LOG_DEBUG, "Adafruit_BBIO: pwm_start: key: %s, ecap_path_udev: %s", key, ecap_path_udev);
 
     // Export PWM if hasn't already been
```
```
debian@beaglebone:~/adafruit-beaglebone-io-python$ sudo python3 test/start_all_pwm.py 
P9_16
P9_21
P9_22
P9_28
P9_29
P9_31
P9_42
P8_13
P8_19
P8_34
P8_36
P8_45
P8_46
```

pdp7, 2021-01-25:
@deelerke could you please either pull the latest commit or just edit that line and let me know if it works for you?

deelerke, 2021-01-25:
hi pdp7 , that looks really promising!  sorry for the delays, I'm in Belgian timezone

I did a fresh install with the Linux beaglebone 4.19.94-ti-r57 image, 
and did a git clone git://github.com/adafruit/adafruit-beaglebone-io-python.git and then edited that line pwm_path[66] -> pwm_path[67] and a "python3 setup.py install" and tried the pwm-test again

```
root@beaglebone:/home/debian/adafruit-beaglebone-io-python# python3 pwm-test.py
Traceback (most recent call last):
  File "pwm-test.py", line 1, in <module>
    import Adafruit_BBIO.PWM as pwm
ModuleNotFoundError: No module named 'Adafruit_BBIO.PWM'
```

```
root@beaglebone:/home/debian/adafruit-beaglebone-io-python# cat pwm-test.py
import Adafruit_BBIO.PWM as pwm
pwm.start("P9_42",50);
pwm.stop("P9_42");
```

I haven't installed anything yet via pip3, but it seems like I must be missing something very trivial and silly. Or does the module have a different name now? thanks !

pdp7, 2021-01-25:
@deelerke hmm, that is weird.  have you tried moving the script to your home the directory and running again?

You also might want to run "sudo python3 ./setup install" and then "sudo python3 ./pwm-test.py".  I don't normally run in a root shell.  I don't think it should be a problem but there might be something going on with the environment variables.

Side note: I forgot to actually merge my pull request with the fix so I have done that now.
https://github.com/adafruit/adafruit-beaglebone-io-python/commit/7361d9a997bce4ab79c83c4ef30442e8dd13f363

pdp7, 2021-01-25:
Here what it looks like for me:
```
debian@beaglebone:~$ cd adafruit-beaglebone-io-python/
debian@beaglebone:~/adafruit-beaglebone-io-python$ sudo python3 ./setup.py install
[sudo] password for debian: 
running install
running bdist_egg
running egg_info
writing Adafruit_BBIO.egg-info/PKG-INFO
writing dependency_links to Adafruit_BBIO.egg-info/dependency_links.txt
writing top-level names to Adafruit_BBIO.egg-info/top_level.txt
reading manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'Adafruit_BBIO.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-3.7/Adafruit_I2C.py -> build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/GPIO.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/ADC.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/Encoder.py -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/PWM.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/UART.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/__init__.py -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/SPI.cpython-37m-arm-linux-gnueabihf.so -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
copying build/lib.linux-armv7l-3.7/Adafruit_BBIO/sysfs.py -> build/bdist.linux-armv7l/egg/Adafruit_BBIO
creating build/bdist.linux-armv7l/egg/overlays
copying build/lib.linux-armv7l-3.7/overlays/builder.py -> build/bdist.linux-armv7l/egg/overlays
copying build/lib.linux-armv7l-3.7/overlays/__init__.py -> build/bdist.linux-armv7l/egg/overlays
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_I2C.py to Adafruit_I2C.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/Encoder.py to Encoder.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/sysfs.py to sysfs.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/overlays/builder.py to builder.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/overlays/__init__.py to __init__.cpython-37.pyc
creating stub loader for Adafruit_BBIO/GPIO.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/PWM.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/ADC.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/SPI.cpython-37m-arm-linux-gnueabihf.so
creating stub loader for Adafruit_BBIO/UART.cpython-37m-arm-linux-gnueabihf.so
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/GPIO.py to GPIO.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/PWM.py to PWM.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/ADC.py to ADC.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/SPI.py to SPI.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_BBIO/UART.py to UART.cpython-37.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_BBIO.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
writing build/bdist.linux-armv7l/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
Adafruit_BBIO.__pycache__.ADC.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.GPIO.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.PWM.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.SPI.cpython-37: module references __file__
Adafruit_BBIO.__pycache__.UART.cpython-37: module references __file__
creating 'dist/Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg
removing '/usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg' (and everything under it)
creating /usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg
Extracting Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg to /usr/local/lib/python3.7/dist-packages
Adafruit-BBIO 1.2.0 is already the active version in easy-install.pth

Installed /usr/local/lib/python3.7/dist-packages/Adafruit_BBIO-1.2.0-py3.7-linux-armv7l.egg
Processing dependencies for Adafruit-BBIO==1.2.0
Finished processing dependencies for Adafruit-BBIO==1.2.0
debian@beaglebone:~/adafruit-beaglebone-io-python$ cd ..
debian@beaglebone:~$ vi pwm-test.py
debian@beaglebone:~$ sudo python3 pwm-test.py
```
It completes with no error

deelerke, 2021-01-25:
allright!!!!! moving to the home dir fixed it and NO ERRORS !!! :) :)
I'll check tonight if I got an actual PWM signal on the pin. but no doubt about it.

fantastic work pdp7 !!!!!

deelerke, 2021-01-26:
yes! pwm signal confirmed! 

thanks again for your efforts on this!

pdp7, 2021-01-26:
@deelerke thanks for testing.  I'm glad we finally got this one solved.

closed 2021-01-26 by pdp7

stoller2122, 2023-09-23:
Same issue here on the latest BBB image and latest version (built from most recent git update) of the adafruit library.

debian@BeagleBone:~$ sudo python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Adafruit_BBIO.PWM as PWM
>>> PWM.start("P9_14",5,50)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Unknown error
>>>