adafruit/adafruit-beaglebone-io-python #343: Issue setting up remoteproc for Beaglebone black

frank2597, 2020-09-04:
Hello, I know this is not directly related but I've been stuck for weeks trying to get the pru pins working on my beaglebone black and i cant seem to find help anywhere else. I have tried reproducing many possible solutions from other forums but they are mostly dated and dont seem to work for me. I am running debian 10.3 with kernel 4.19.94-ti-r50
Currently i am trying to follow Derek Molloy's exploring beaglebone chapter 15 to set up remoteproc so i entered in the command: 

`debian@beaglebone:/sys/class/remoteproc/remoteproc1$  echo 'start' > state
`
However i receive this error: 

`-bash: echo: write error: Invalid argument`
i also receive the following error in dmesg:
```
[91017.448340] remoteproc remoteproc1: loading /lib/firmware/am335x-pru0-fw failed with error -22
[91017.448351] remoteproc remoteproc1: Direct firmware load for am335x-pru0-fw failed with error -22
[91017.448375] remoteproc remoteproc1: request_firmware failed: -22
[91017.466238] remoteproc remoteproc1: Boot failed: -22
```

```
sudo /opt/scripts/tools/version.sh
```

```
git:/opt/scripts/:[ef16ffa8c27225005cc2b2c974a83da3dee6c1ef]
eeprom:[A335BNLTBWA52010BBWG0394]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gc9b3922522]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBBW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r50]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~buster+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   63.819495] remoteproc remoteproc0: wkup_m3 is available
[   63.896480] remoteproc remoteproc0: powering up wkup_m3
[   63.896512] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   63.896764] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   65.768658] remoteproc remoteproc1: 4a334000.pru is available
[   65.788881] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   65.768658] remoteproc remoteproc1: 4a334000.pru is available
[   65.768879] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   65.788881] remoteproc remoteproc2: 4a338000.pru is available
[   65.789100] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.950857] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.964492] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

Thank you for your support!

pdp7, 2020-09-04:
@MarkAYoder @RobertCNelson any suggestions for this PRU issue?  thanks!

MarkAYoder, 2020-09-04:
@frank2597  Everything seems to match what I have on my Beagle.  Have you tried compiling your own code and putting it in /lib/firmware/am335x-pru0-fw?

frank2597, 2020-09-04:
@MarkAYoder Hi thank you for the reply, i am using the pru software support package from texas instruments and i was able to compile the PRU_gpioToggle.c example they provide and attain the PRU_gpioToggle.out file. I place it in /lib/firmware/am335x-pru0-fw. I then tried to follow the steps from [link](https://www.element14.com/community/community/designcenter/single-board-computers/next-genbeaglebone/blog/2019/05/14/coding-for-the-beaglebone-pru-with-c-in-2019) 

so i did:
`debian@beaglebone:/sys/class/remoteproc/remoteproc1$ echo 'PRU_gpioToggle.out' > firmware`
`debian@beaglebone:/sys/class/remoteproc/remoteproc1$ echo 'start' > state`
but i still receive the error:
`-bash: echo: write error: No such file or directory`

RobertCNelson, 2020-09-04:
@frank2597 after you echo the firwmare name, please run

```
dmesg | grep remoteproc
```

PS, this was never fully solved, but in v4.19.x-ti you might have to use the am335x-pru*-fw names..

https://github.com/beagleboard/Latest-Images/issues/23

frank2597, 2020-09-04:
@RobertCNelson Here is the output:
```
[   63.819495] remoteproc remoteproc0: wkup_m3 is available
[   63.896480] remoteproc remoteproc0: powering up wkup_m3
[   63.896512] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   63.896764] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   65.768658] remoteproc remoteproc1: 4a334000.pru is available
[   65.788881] remoteproc remoteproc2: 4a338000.pru is available
[25961.859645] remoteproc remoteproc1: powering up 4a334000.pru
[25961.861611] remoteproc remoteproc1: Direct firmware load for PRU_gpioToggle.out failed with error -2
[25961.861643] remoteproc remoteproc1: request_firmware failed: -2
[25961.876331] remoteproc remoteproc1: Boot failed: -2

```

RobertCNelson, 2020-09-04:
thanks @frank2597 is PRU_gpioToggle.out under /lib/firmware/

```
ls -lha /lib/firmware/
```

frank2597, 2020-09-04:
@RobertCNelson I originally place it in /lib/firmware/am335x-pru0-fw but after placing it in am335x-pru0-fw i was able to run 
```
debian@beaglebone:/sys/class/remoteproc/remoteproc1$ echo 'PRU_gpioToggle.out' > firmware
debian@beaglebone:/sys/class/remoteproc/remoteproc1$ echo 'start' > state
```
without error! Thank you, Robert. It seems to me that if nothing is loaded into the firmware initially then:
`echo 'start' > state` 
wont wont. My led is blinking too, Thanks again.

closed 2020-09-29 by pdp7